zephyrproject-rtos/zephyr #33857: atomic xtensa  build fail

hongshui3000, 2021-03-30:
**Describe the bug**
Not all xtensa architectures support the s32c1i and SCOMPARE1 registers

log：
```
atomic_xtensa.h:35: Error: unknown opcode or format name 's32c1i'
atomic_xtensa.h:35: Error: invalid register 'SCOMPARE1' for 'wsr' instruction
atomic_xtensa.h:35: Error: unknown opcode or format name 's32c1i'
xt-xcc.exe ERROR:  C:/usr/xtensa/XtDevTools/install/tools/RI-2020.5-win32/XtensaTools/bin/xt-as.exe returned non-zero status 1
```
At least the code needs to be judged as follows
```
if (XCHAL_HAVE_S32C1)
..............
```

andyross, 2021-03-31:
Does your platform really not have S32C1?  That **is** the Xtensa atomic operations primitive, without it you have no hardware support and need to fall back to a much slower irq_lock() implementation in ATOMIC_OPERATIONS_C.  ATOMIC_OPERATIONS_BUILTIN just falls back to the gcc implementations, which are essentially identical to our own and also based on S32C1I -- they won't work for you (the only reason we have our own is that older xcc toolchains don't have the gcc builtins).

As far as selecting this automatically based on core-isa.h constants, that would be nice but difficult, as kconfig sits upstream of the C compiler.  You need the kconfig output header defined before you can build a C file against core-isa.h.  The Zephyr model is that the platform layer decides on software configuration.

hongshui3000, 2021-03-31:
@andyross
The biggest problem I encounter at the moment is that I have to change the code of Zephyr in order to compile and pass.
If I change this, it will affect other platforms.
zephyr\arch\Kconfig：
```
config XTENSA
	bool
	select ARCH_IS_SET
	select HAS_DTS
	select USE_SWITCH
	select USE_SWITCH_SUPPORTED
        select ATOMIC_OPERATIONS_ARCH          -----------------------------------------------------------del here
	help
	  Xtensa architecture
```
as long as CONFIG_XTENSA is configured, select ATOMIC_OPERATIONS_ARCH will be selected. So my suggestion is not to choose here, but to give others the right to choose.
Otherwise, the unsupported S32C1 platform cannot be built through

closed 2021-04-02 by nashif

jcmvbkbc, 2021-04-02:
> Does your platform really not have S32C1? That **is** the Xtensa atomic operations primitive

To be more precise that is one of the possible xtensa atomic operations primitive. There's also exclusive access option with l32ex/s32ex/getex/clrex opcodes and XCHAL_HAVE_EXCLUSIVE core-isa.h macro.

PR #33964 (2021-04-02): kernel: prioritize C atomic functions if selected
Usually, GCC builtin or arch-specific atomic functions are being used. The corresponding kconfigs are "selected" by architecture
or SoC kconfigs. CONFIG_ATOMIC_OPERATIONS_C is usually used to override the GCC built-in or arch-specific atomic unctions when these two are not supported. So change the priority of #include so that C version is included first if selected, and skips the inline versions of the other two variants. Or else there will be two compiled versions of atomic functions: inline version and the compiled C version.

Also remove the [... cut]
commit 9f0188da (2021-04-02): kernel: prioritize C atomic functions if selected

Usually, GCC builtin or arch-specific atomic functions are being
used. The corresponding kconfigs are "selected" by architecture
or SoC kconfigs. CONFIG_ATOMIC_OPERATIONS_C is usually used to
override the GCC built-in or arch-specific atomic functions when
these two are not supported. So change the priority of #include
so that C version is included first if selected, and skips
the inline versions of the other two variants. Or else there
will be two compiled versions of atomic functions: inline version
and the compiled C version. Note that the  [... cut]