zephyrproject-rtos/zephyr #33873: arm_arch_timer: Too many clock announcements with CONFIG_TICKLESS_KERNEL=n on SMP

luisgerhorst, 2021-03-30:
https://github.com/zephyrproject-rtos/zephyr/blob/95cddd4df7e63375f27d6308dd9b31ffd4d8ab97/drivers/timer/arm_arch_timer.c#L46

If `CONFIG_TICKLESS_KERNEL` is not set this always announces a tick. With SMP only one core is expected to announce a tick even if all cores receive the IRQ, therefore this announces too many ticks. For example, this causes applications which use `k_msleep` to be woken up too early.

The code works correctly in a tickless kernel because `last_cycle` is shared and the cores that execute the critical section subsequently calculate `delta_tick=0`.

Unrelated: I believe `last_cycle` does not need to be `volatile` because it is protected by the lock.

carlocaione, 2021-03-31:
out of curiosity, why do you need to have a non-tickless kernel?

carlocaione, 2021-06-02:
@galak I think we can merge #33909 to close this

galak, 2021-06-02:
> @galak I think we can merge #33909 to close this

Looks like PR still needs to address or come to resolution with @andyross comments.

closed 2021-09-18 by github-actions[bot]