zephyrproject-rtos/zephyr #33876: Lora sender sample build error for esp32

houssine97, 2021-03-30:
**Describe the bug**
When building a project with the esp32 , I get build errors for example:
`-- Generating done
-- Build files have been written to: C:/Users/houssine/zephyrproject/zephyr/build
[70/139] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_common.c.obj
FAILED: zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_common.c.obj
C:\Users\houssine\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DKERNEL -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I../kernel/include -I../arch/xtensa/include -I../include -Izephyr/include/generated -I../soc/xtensa/esp32 -I../soc/xtensa/esp32/include -I../lib/libc/newlib/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/adapter/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/adapter/include/crypto -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/adapter/include/wifi -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_common/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_rom/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/xtensa/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/xtensa/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/soc/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/soc/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/soc/esp32/private_include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/src/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/driver/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/soc/esp32 -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_wifi/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/efuse/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/efuse/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_system/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_wifi/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_timer/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_timer/private_include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_netif/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_event/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/log/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/wpa_supplicant/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/wpa_supplicant/port/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/wpa_supplicant/src -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/wpa_supplicant/include/esp_supplicant -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/wpa_supplicant/src/crypto -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/spi_flash/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/spi_flash/private_include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/bootloader_support/include -Os -imacros C:/Users/houssine/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -imacros C:/Users/houssine/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=C:/Users/houssine/zephyrproject/zephyr/samples/drivers/lora/send=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/Users/houssine/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/Users/houssine/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_common.c.obj -MF zephyr\lib\posix\CMakeFiles\lib__posix.dir\pthread_common.c.obj.d -o zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_common.c.obj -c C:/Users/houssine/zephyrproject/zephyr/lib/posix/pthread_common.c
In file included from C:/Users/houssine/zephyrproject/zephyr/lib/posix/pthread_common.c:10:
../include/posix/time.h:90:15: error: static declaration of 'clock_gettime' follows non-static declaration
 __syscall int clock_gettime(clockid_t clock_id, struct timespec *ts);
               ^~~~~~~~~~~~~
In file included from ../include/posix/time.h:12,
                 from C:/Users/houssine/zephyrproject/zephyr/lib/posix/pthread_common.c:10:
c:\users\houssine\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\time.h:187:5: note: previous declaration of 'clock_gettime' was here
 int clock_gettime (clockid_t clock_id, struct timespec *tp);
     ^~~~~~~~~~~~~
In file included from C:/Users/houssine/zephyrproject/zephyr/lib/posix/pthread_common.c:10:
../include/posix/time.h:94:5: error: conflicting types for 'timer_create'
 int timer_create(clockid_t clockId, struct sigevent *evp, timer_t *timerid);
     ^~~~~~~~~~~~
In file included from ../include/posix/time.h:12,
                 from C:/Users/houssine/zephyrproject/zephyr/lib/posix/pthread_common.c:10:
c:\users\houssine\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\time.h:192:5: note: previous declaration of 'timer_create' was here
 int timer_create (clockid_t clock_id,
     ^~~~~~~~~~~~
[71/139] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/nanosleep.c.obj
FAILED: zephyr/lib/posix/CMakeFiles/lib__posix.dir/nanosleep.c.obj
C:\Users\houssine\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DKERNEL -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I../kernel/include -I../arch/xtensa/include -I../include -Izephyr/include/generated -I../soc/xtensa/esp32 -I../soc/xtensa/esp32/include -I../lib/libc/newlib/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/adapter/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/adapter/include/crypto -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/adapter/include/wifi -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_common/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_rom/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/xtensa/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/xtensa/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/soc/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/soc/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/soc/esp32/private_include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/src/esp32/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/driver/include -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/soc/soc/esp32 -IC:/Users/houssine/zephyrproject/modules/hal/espressif/zephyr/../components/esp_wifi/include -IC:/Users/houssine/zephyrproject/ [... cut ...]
c:\users\houssine\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\time.h:192:5: note: previous declaration of 'timer_create' was here
 int timer_create (clockid_t clock_id,
     ^~~~~~~~~~~~
[75/139] Building C object zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_esp32.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\houssine\zephyrproject\zephyr\build'
`
**To Reproduce**
Steps to reproduce the behavior:
1. Build using `west build -b esp32 samples\drivers\lora\send`

**Expected behavior**
send a message with lora.

sylvioalves, 2021-03-30:
@houssine97 Lora expects `NEWLIB_C` support which is not working for ESP32 yet.
Would you try adding `CONFIG_NATIVE_APPLICATION=y` into you `prj.conf` to see if it compiles properly?
This is not the solution but at least confirms the error.

sylvioalves, 2021-03-31:
Hi @houssine97. I guess you have been using a custom board with a lora radio.  
`C:/Users/houssine/zephyrproject/zephyr/include/toolchain/gcc.h:62:36: error: static assertion failed: "No default LoRa radio specified in DT"`

You need to add the radio configuration into device tree to have it mapped properly.

One way is adding an .overlay file within the project with the radio instance/pin configs.

houssine97, 2021-03-31:
hi @sylvioalves the board i am using is Heltec WiFi LoRa 32 V2 boards and i this is the overlay that i created for dding the radio configuration into device tree but the code is not building.
`/ {
aliases {
    lora0 = &lora;
 };
};
&spi3 {
     status = "okay";
     lora: sx1276@0 {
         compatible = "semtech,sx1276";
         reg = <0>;
         label = "sx1276";
         reset = <14>;
         miso-pin = <19>;
         mosi-pin = <27>;
         dio-pin = <26>;
         dio-pin2 = <34>;
         dio-pin3 = <35>;
         sclk-pin = <5>;
         cs-pin = <15>;
         spi-max-frequency = <1000000>;
         power-amplifier-output = "pa-boost";
     };
};`

houssine97, 2021-04-02:
@sylvioalves ??

sylvioalves, 2021-04-05:
Hi @houssine97, someone on the team is analyzing that.

houssine97, 2021-04-06:
@sylvioalves  thank you

houssine97, 2021-04-27:
@sylvioalves any update

sylvioalves, 2021-04-28:
@houssine97, not yet, supporting newlibc is tricky. We are working on it.

protobits, 2021-07-13:
If this is still an issue preventing lora to be used on ESP32 maybe we can keep this open to track the missing feature?

cdrx, 2022-01-26:
I've just tested this and get the same error. 

@sylvioalves could this be reopened?

cdrx, 2022-02-04:
@sylvioalves what needs to be worked on to make this work? I'm having a hard time understanding the various pieces here

adalava, 2022-02-07:
With the  samples/drivers/lora/send/ I get the following build errors:

> In file included from /home/alfredoj/zephyrproject/zephyr/lib/posix/pthread_common.c:10:
> /home/alfredoj/zephyrproject/zephyr/include/posix/time.h:90:15: error: static declaration of 'clock_gettime' follows non-static declaration
>  __syscall int clock_gettime(clockid_t clock_id, struct timespec *ts);
>                ^~~~~~~~~~~~~
> In file included from /home/alfredoj/zephyrproject/zephyr/include/posix/time.h:12,
>                  from /home/alfredoj/zephyrproject/zephyr/lib/posix/pthread_common.c:10:
> /home/alfredoj/.espressif/tools/zephyr/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/time.h:187:5: note: previous declaration of 'clock_gettime' was here
>  int clock_gettime (clockid_t clock_id, struct timespec *tp);

And:

>In file included from /home/alfredoj/zephyrproject/zephyr/lib/posix/pthread_common.c:10:
>/home/alfredoj/zephyrproject/zephyr/include/posix/time.h:94:5: error: conflicting types for 'timer_create'
 >int timer_create(clockid_t clockId, struct sigevent *evp, timer_t *timerid);
 >    ^~~~~~~~~~~~
>In file included from /home/alfredoj/zephyrproject/zephyr/include/posix/time.h:12,
>                from /home/alfredoj/zephyrproject/zephyr/lib/posix/pthread_common.c:10:
>/home/alfredoj/.espressif/tools/zephyr/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/time.h:192:5: note: previous declaration of >'timer_create' was here
> int timer_create (clockid_t clock_id,

closed 2022-04-23 by github-actions[bot]

cdrx, 2022-04-23:
I think this is still an issue.

@sylvioalves could this be reopened please?

jeronimoagullo, 2022-05-12:
I am facing the same issue when I try to build the example with TensorFlow Lite, which requires the use of newlibc too. 
I there any update? Thanks

sylvioalves, 2022-05-12:
Hi, please, update Zephyr to latest main, run `west update` then  `west espressif update`.  
Notice that now ESP32 uses Zephyr-SDK 0.14.1. Please, also run `export ZEPHYR_TOOLCHAIN_VARIANT="zephyr"` so that it used the SDK as expected.  
In case you have issues running `west update` regarding `hal_espressif`, try deleting `hal_espressif` folder and run `west update` again.

jeronimoagullo, 2022-05-12:
Hi @sylvioalves 

Thank you very much for your answer. 

I tried firstly to use the Zephyr toolchain instead of the ESP-IDF. I am trying to compile for esp32, thus I am building with `west build -b esp32 -p`. My computer OS is Windows 11.  However, when I get the following error.

```
(.venv) C:\Workspace\zephyrproject\zephyr\samples\hello_world>west build -b esp32 -p
-- west build: making build dir C:\Workspace\zephyrproject\zephyr\samples\hello_world\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Workspace/zephyrproject/zephyr/samples/hello_world
-- Found Python3: C:/Workspace/zephyrproject/.venv/Scripts/python.exe (found suitable exact version "3.10.4") found components: Interpreter
-- Cache files will be written to: C:/Workspace/zephyrproject/zephyr/.cache
-- Zephyr version: 3.0.99 (C:/Workspace/zephyrproject/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: esp32
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.14.1 (C:/Workspace/zephyr-sdk-0.14.1)
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.14.1 (C:/Workspace/zephyr-sdk-0.14.1)
-- Found BOARD.dts: C:/Workspace/zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts
-- Generated zephyr.dts: C:/Workspace/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Workspace/zephyrproject/zephyr/samples/hello_world/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: C:/Workspace/zephyrproject/zephyr/samples/hello_world/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: C:/Workspace/zephyrproject/zephyr/samples/hello_world/build/zephyr/dts.cmake
C:/Workspace/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr.dts:138.27-216.5: Warning (simple_bus_reg): /soc/pin-controller: missing or empty reg/ranges property
Parsing C:/Workspace/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Workspace/zephyrproject/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration 'C:/Workspace/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to 'C:/Workspace/zephyrproject/zephyr/samples/hello_world/build/zephyr/.config'
Kconfig header saved to 'C:/Workspace/zephyrproject/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h'
CMake Error at C:/Workspace/zephyrproject/zephyr/cmake/compiler/gcc/target.cmake:17 (message):
  C compiler
  C:/Workspace/zephyr-sdk-0.14.1/xtensa-_zephyr-elf/bin/xtensa-_zephyr-elf-gcc
  not found - Please check your toolchain installation
Call Stack (most recent call first):
  C:/Workspace/zephyrproject/zephyr/cmake/modules/target_toolchain.cmake:63 (include)
  C:/Workspace/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:121 (include)
  C:/Workspace/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:51 (include)
  C:/Workspace/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:76 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '-DWEST_PYTHON=C:\Workspace\zephyrproject\.venv\Scripts\python.exe' '-BC:\Workspace\zephyrproject\zephyr\samples\hello_world\build' '-SC:\Workspace\zephyrproject\zephyr\samples\hello_world' -GNinja -DBOARD=esp32
```

The issue is that the path where west is looking for doesn't exist. `C:/Workspace/zephyr-sdk-0.14.1/xtensa-_zephyr-elf/bin/xtensa-_zephyr-elf-gcc` doesn't exist actually. However, the folders within the zephyr-sdk-0.14.1 are the following: 

```
Directorio de C:\Workspace\zephyr-sdk-0.14.1

12/05/2022  16:01    <DIR>          .
12/05/2022  12:06    <DIR>          ..
11/04/2022  21:44    <DIR>          aarch64-zephyr-elf
11/04/2022  22:32    <DIR>          arc-zephyr-elf
11/04/2022  21:43    <DIR>          arc64-zephyr-elf
11/04/2022  22:48    <DIR>          arm-zephyr-eabi
05/05/2022  11:31    <DIR>          cmake
11/04/2022  21:50    <DIR>          mips-zephyr-elf
11/04/2022  21:57    <DIR>          nios2-zephyr-elf
11/04/2022  22:06    <DIR>          riscv64-zephyr-elf
12/04/2022  04:51                 7 sdk_version
12/04/2022  04:51             5.581 setup.cmd
11/04/2022  22:08    <DIR>          sparc-zephyr-elf
11/04/2022  21:48    <DIR>          x86_64-zephyr-elf
11/04/2022  21:43    <DIR>          xtensa-espressif_esp32s2_zephyr-elf
11/04/2022  21:45    <DIR>          xtensa-espressif_esp32_zephyr-elf
11/04/2022  21:51    <DIR>          xtensa-intel_apl_adsp_zephyr-elf
11/04/2022  21:47    <DIR>          xtensa-intel_bdw_adsp_zephyr-elf
11/04/2022  21:51    <DIR>          xtensa-intel_byt_adsp_zephyr-elf
11/04/2022  21:51    <DIR>          xtensa-intel_s1000_zephyr-elf
11/04/2022  21:52    <DIR>          xtensa-nxp_imx8m_adsp_zephyr-elf
11/04/2022  21:53    <DIR>          xtensa-nxp_imx_adsp_zephyr-elf
11/04/2022  21:55    <DIR>          xtensa-sample_controller_zephyr-elf
               2 archivos          5.588 bytes
              21 dirs  368.887.701.504 bytes libres
```

Thus, the compiler is there and the issue is just the path. Do you know how I can fix it to set the correct folder for building?

thank you in advance

jeronimoagullo, 2022-05-13:
Hi @sylvioalves 
It is working now. 
The issue was that I was working with an all version of Zephyr code. I notice that `west update` updates all the sub-repositories of zephyr project but doesn't update the main repository. Thus, I needed to do a `git pull` in the root of zephyr project and a new `west update` afterwards. Then, the modules were updated (specially the hal_espressif) and everyhting works now fine. 
I think that we can close the issue. Thank very much!

sylvioalves, 2022-05-13:
@jeronimoagullo Have you also tested the application and perform Lora send/received contents or only built the project?

mpiccolino-tealblue, 2022-05-22:
@sylvioalves @jeronimoagullo I'm about to test some LoRa receiving (and possibly sending) on Heltec WiFi LoRa 32 (V2), will report

mpiccolino-tealblue, 2022-05-22:
@sylvioalves I haven't been able to compile the `lora/send` sample for the Heltec WiFi LoRa 32 (V2) with a fresh Zephyr install on macOS and Zephyr SDK 0.14.2. Here is the output:

```
[1/15] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /Users/marco_piccolino/zephyr-sdk-0.14.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc -DCONFIG_SPI_FLASH_ROM_DRIVER_PATCH -DCONFIG_SPI_FLASH_USE_LEGACY_IMPL -DKERNEL -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/Users/marco_piccolino/zephyrproject/zephyr/include/zephyr -I/Users/marco_piccolino/zephyrproject/zephyr/include -I/Users/marco_piccolino/zephyrproject/zephyr-lora-send/build/zephyr/include/generated -I/Users/marco_piccolino/zephyrproject/zephyr/soc/xtensa/esp32 -I/Users/marco_piccolino/zephyrproject/zephyr/soc/xtensa/esp32/include -I/Users/marco_piccolino/zephyrproject/zephyr/lib/libc/newlib/include -I/Users/marco_piccolino/zephyrproject/modules/hal/atmel/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/include/bt -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/include/crypto -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../esp_shared/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../esp_shared/include/wifi -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/hal/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/hal/esp32/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_hw_support/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_hw_support/port/esp32 -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_hw_support/port/esp32/private_include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_common/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp32/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/esp32/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/xtensa/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/xtensa/esp32/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/include/soc -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/src/esp32/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/driver/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_wifi/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/efuse/esp32/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/efuse/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_system/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_wifi/esp32/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_timer/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_timer/private_include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_netif/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_event/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/log/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/port/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/src -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/include/esp_supplicant -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/wpa_supplicant/src/crypto -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/spi_flash/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/spi_flash/private_include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/bootloader_support/include -I/Users/marco_piccolino/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/bootloader_support/include_bootloader -Os -imacros /Users/marco_piccolino/zephyrproject/zephyr-lora-send/build/zephyr/include/generated/auto [... cut ...]
/Users/marco_piccolino/zephyrproject/zephyr/drivers/lora/sx127x.c:339:13: warning: 'sx127x_irq_callback' defined but not used [-Wunused-function]
  339 | static void sx127x_irq_callback(const struct device *dev,
      |             ^~~~~~~~~~~~~~~~~~~
/Users/marco_piccolino/zephyrproject/zephyr/drivers/lora/sx127x.c:332:13: warning: 'sx127x_dio_work_handle' defined but not used [-Wunused-function]
  332 | static void sx127x_dio_work_handle(struct k_work *work)
      |             ^~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.

```

jeronimoagullo, 2022-05-23:
Hi @sylvioalves and @mpiccolino-tealblue 

I have just tried to build the lora send example for esp32 and I got the same output that @mpiccolino-tealblue posted.

sylvioalves, 2022-05-23:
@mpiccolino-tealblue @jeronimoagullo, can you share the overlay file?  
I haven't tested but I think it needs to be something like  
```c
&spi3 {
    status = "okay";
    #address-cells = <1>;
    #size-cells = <0>;
    pinctrl-0 = <&spim3_miso_gpio19 &spim3_mosi_gpio27
                     &spim3_sclk_gpio5 &spim3_csel_gpio18>;
    pinctrl-names = "default";

    cs-gpios = <&gpio0 18 GPIO_ACTIVE_LOW>;
    lora: sx1276@0 {
        compatible = "semtech,sx1276";
        reg = <0>;
        label = "sx1276";
        reset-gpios = <&gpio0 14 (GPIO_PULL_UP | GPIO_ACTIVE_LOW)>;
        dio-gpios = <&gpio0 26 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
            <&gpio1 34 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
            <&gpio1 5 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
        spi-max-frequency = <1000000>;
        power-amplifier-output = "pa-boost";
    };
};
```

mpiccolino-tealblue, 2022-05-23:
Thanks for your feedback. The output I shared refers to a missing overlay file (I am new to Zephyr and didn't notice that the heltec dtc doesnt contain info about lora). I tried with the overlay posted earlier in this thread but I get a cmake error. I have written to Heltec support to see if they can provide one. @sylvioalves I'll check yours later when I get back to the pc. I seem to remember there was a bit-related issue with the reg value.

mpiccolino-tealblue, 2022-06-25:
@sylvioalves hello! Will you have any time in the near future to have a look at the LoRa support for the Heltec v2, either as overlay or direcly in the device tree? many thanks

gportay, 2022-10-18:
Hello all,

I am currently working on enabling the SX1276 LoRa chip on the Heltec WiFi LoRa32 V2.

I have made the following changes in the device tree:
- added the `pinctrl` for the `spi3` in `boards/xtensa/heltec_wifi_lora32_v2/heltec_wifi_lora32_v2-pinctrl.dtsi`
- enabled the `spi3` node for the board `heltec_wifi_lora32_v2` in `boards/xtensa/heltec_wifi_lora32_v2/heltec_wifi_lora32_v2.dts`
- added the `lora` node into the `spi3` node
- add the `lora0` alias

The sample `samples/subsys/shell/shell_module/` compiles fine after the enabling the config `CONFIG_SPI`, `CONFIG_LORA`, and `CONFIG_LORA_SX127X` (the lora shell is compiled automagically) :tada:!

But it does not work yet :cry:!

1. The lora driver requires to setup the frequency before doing any LoRa transfers (send or recv). Therefore, the command `lora config freq 868000000` as to be run (for 868MHz). 
1. The command `lora recv` loop forever without receiving any lora frames from another device.
1. The command `lora send "Hello, World!" can be run, but it ends with `EAGAIN`:
```
uart:~$ lora send "Hello, World!"
LoRa send failed: -11
[00:03:04.184,000] <err> sx12xx_common: Packet transmission failed!
```

The log on the uart shows the lora driver retrieved the `0x00` as the chip version; I am expected a value such as `0x12` (not sure).

I suspect SPI communication issues, and I need some help to figure out what is going on here.

Also, I have analyzed the signals:

[image]
[image]

I am not an SPI expert at all, so I cannot see yet what is wrong with the signals.

_Note_: The changes are available [here](https://github.com/idiot-prototypes/zephyr/tree/heltec_wifi_lora32_v2-add-lora-support).

gportay, 2022-10-18:
Maybe, can someone double check the device tree from the [pinout](https://resource.heltec.cn/download/WiFi_LoRa_32/WIFI_LoRa_32_V2.1.pdf) or [schematics](https://resource.heltec.cn/download/WiFi_LoRa_32/V2/WIFI_LoRa_32_V2(868-915).PDF) documents?

| PIN  | GPIO   |
| ---- | -------|
| CS#  | GPIO18 |
| CLK  | GPIO19 |
| MOSI | GPIO27 |
| MISO | GPIO19 |
| RST# | GPIO14 |
| DIO0 | GPIO26 |
| DIO1 | GPIO35 |
| DIO2 | GPIO34 |

```
	spim3_default: spim3_default {
		group1 {
			pinmux = <SPIM3_MISO_GPIO19>,
				 <SPIM3_SCLK_GPIO5>,
				 <SPIM3_CSEL_GPIO18>;
		};
		group3 {
			pinmux = <SPIM3_MOSI_GPIO27>;
			output-low;
		};
	};
```

```
&spi3 {
	#address-cells = <1>;
	#size-cells = <0>;
	status = "okay";
	pinctrl-0 = <&spim3_default>;
	pinctrl-names = "default";
	cs-gpios = <&gpio0 18 GPIO_ACTIVE_LOW>;
	lora0: lora@0 {
		compatible = "semtech,sx1276";
		reg = <0>;
		reset-gpios = <&gpio0 14 GPIO_ACTIVE_LOW>;
		dio-gpios =
			/* SX1276 D0 */
			<&gpio0 26 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			/* SX1276 D1 */
			<&gpio0 35 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			/* SX1276 D2 */
			<&gpio0 34 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
		spi-max-frequency = <1000000>;
		power-amplifier-output = "pa-boost";
	};
};
```

cdrx, 2022-10-18:
I struggled with this recently. Chip version `0x00` is exactly what I was getting, before I had the SPI correctly setup.

I don't know if its the cause of your issue, but I think your pin assignment for DIO D1 & D2 needs to change. On the ESP32, pins above `GPIO31` are on port `&gpio1` not `&gpio0`. [See here for more details](https://zephyrproject.org/how-to-set-up-esp32-gpio-pins-in-zephyr-rtos/).

cdrx, 2022-10-18:
Try putting all your SPI pins into a single group (not `group1` and `group3`).

This config works for me on an ESP32 based board (not a Heltec board, so the GPIO PIN numbers will need to be changed for you):

```
/* this is app.overlay */
#include <dt-bindings/spi/spi.h>
#include <dt-bindings/pinctrl/esp32-pinctrl.h>

&spi2 {
    status = "okay";
    pinctrl-0 = <&spim2_default>;
    pinctrl-names = "default";
    cs-gpios = <&gpio0 5 GPIO_ACTIVE_LOW>;

    rfm95: sx1276@0 {
        compatible = "semtech,sx1276";
        reg = < 0 >;
        label = "sx1276";
        reset-gpios = < &gpio0 27 GPIO_ACTIVE_LOW >;
        dio-gpios = < &gpio0 26 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH) >,
                    < &gpio0 25 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH) >,
                    < &gpio1 1 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH) >;
        spi-max-frequency = <1000000>;
        power-amplifier-output = "pa-boost";
    };
};

&pinctrl {
    spim2_default: spim2_default {
        group1 {
            pinmux = < SPIM2_MISO_GPIO34 >,
                     < SPIM2_MOSI_GPIO23 >,
                     < SPIM2_SCLK_GPIO18 >;
	   };
    };
};

/ {
    aliases {
        lora0 = &rfm95;
    };
};
```

gportay, 2022-10-18:
> 

Oh, thank you so much!

It works since I switched to `spi2`, and I fixed the `gpio1` bank for `GPIO35` and `GPIO34`, and I used a single group only!

All in all, all you told me to change.

```
&pinctrl {
    spim2_default: spim2_default {
        group1 {
            pinmux = < SPIM2_MISO_GPIO19 >,
                     < SPIM2_MOSI_GPIO27 >,
                     < SPIM2_SCLK_GPIO5 >;
	   };
    };
};
```

```
/ {
    aliases {
        lora0 = &rfm95;
    };
};

&spi2 {
    #address-cells = <1>;
    #size-cells = <0>;
    status = "okay";
    pinctrl-0 = <&spim2_default>;
    pinctrl-names = "default";
    cs-gpios = <&gpio0 18 GPIO_ACTIVE_LOW>;

    rfm95: sx1276@0 {
        compatible = "semtech,sx1276";
        reg = < 0 >;
        reset-gpios = < &gpio0 14 GPIO_ACTIVE_LOW >;
        dio-gpios = < &gpio0 26 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH) >,
                    < &gpio1 3 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH) >,
                    < &gpio1 2 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH) >;
        spi-max-frequency = <1000000>;
        power-amplifier-output = "pa-boost";
    };
};
```

mpiccolino-tealblue, 2022-10-30:
@cdrx @gportay thanks for you work on this. It compiles fine on my side. Will try receiving a few messages.
@gportay what are you using for issuing lora commands?

gportay, 2022-10-30:
> @cdrx @gportay thanks for you work on this. It compiles fine on my side. Will try receiving a few messages. @gportay what are you using for issuing lora commands?

@mpiccolino-tealblue I used the Lora shell from the Zephyr tree (see #51404).

gportay, 2022-10-30:
Also, I would be happy if someone can answer this https://github.com/zephyrproject-rtos/zephyr/pull/51404#discussion_r998527942 😁

mpiccolino-tealblue, 2022-11-03:
@gportay many thanks.
I assume it is CONFIG_LORA_SX12XX, not
[image] right?

gportay, 2022-11-08:
Indeed, I use the [config](https://github.com/idiot-prototypes/esplorative-idiot/blob/main/prj_sx12xx.conf) chunk:

```
CONFIG_SPI=y
CONFIG_LORA=y
CONFIG_LORA_SX12XX=y
```