zephyrproject-rtos/zephyr #33895: Add I2C2 node to STM32L412 and STM32L422 in device tree

GunZi200, 2021-03-30:
The device tree for stm32l412xx and stm32l422xx does not include I2C2. 

We should add it the file `dts/arm/st/l4/stm32l412.dtsi`

This would enable using I2C2 with the following chips:
* stm32l412x8
* stm32l412xb
* stm32l422xb

erwango, 2021-03-31:
@GunZi200 Thanks for reporting. Don't hesitate to push the changes if you have some time.

Though, please note that OPAMP and COMP1 are not currently supported.
This also needs to be checked for LPTIM2 (I don't think it would work as is)

erwango, 2021-06-04:
@gudnimg I propose to close this point as we don't claim that STM32 soc description are complete, but they are done on need basis. So that's ok to open a point is if there's an action to close it, but I don't want to keep these for long as I don't plan to track them.
Feel free to re-open if you have plans to fix

closed 2021-06-04 by erwango