zephyrproject-rtos/zephyr #33897: Extended Advertising Can't Restart After Connection

legoabram, 2021-03-31:
**Describe the bug**
A peripheral using an extended advertising set is unable to restart that advertising set to allow a second connection to a central while remaining connected to the first.

The error first occurs in `ull_adv.c` at line 994 when `node_rx_adv_term = ll_rx_alloc()` returns null.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Zephyr application using the provided `main.c` and `prj.conf` [prj.zip](https://github.com/zephyrproject-rtos/zephyr/files/6232986/prj.zip)
2. mkdir build; cd build
3. cmake -DBOARD=board\nrf52840dk_nrf52840
4. make
5. See error

**Expected behavior**
An extended advertising set should be able to be restarted after obtaining a connection to allow a second connection.

**Impact**
We cannot ship product unless we can obtain multiple connections to central devices, and we must use extended advertising sets to do so as we have a secondary, non-connectable advertisement that is broadcast simultaneously with the connectable advertising.

**Logs and console output**
```
00> [00:00:12.129,608] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem taken
00> 
00> [00:00:12.129,638] <dbg> bt_ctlr_hci_driver.prio_recv_thread: RX node enqueue
00> [00:00:12.129,638] <dbg> bt_ctlr_hci_driver.prio_recv_thread: RX node enqueue
00> [00:00:12.129,669] <dbg> bt_ctlr_hci_driver.prio_recv_thread: RX node enqueue
00> [00:00:12.129,669] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem take...
00> [00:00:12.129,699] <dbg> bt_ctlr_hci_driver.recv_thread: unblocked
00> [00:00:12.129,730] <dbg> bt_ctlr_hci_driver.recv_thread: Packet in: type:1 len:33
00> [00:00:12.129,730] <dbg> bt_hci_core.bt_recv: buf 0x2000d6a0 len 33
00> [00:00:12.129,730] <dbg> bt_hci_core.hci_event: event 0x3e
00> [00:00:12.129,730] <dbg> bt_hci_core.hci_le_meta_event: subevent 0x0a
00> [00:00:12.129,974] <dbg> bt_hci_core.enh_conn_complete: status 0x00 handle 0 role 1 peer 4F:53:10:9A:48:41 (random) peer RPA 00:00:00:00:00:00
00> [00:00:12.130,096] <dbg> bt_hci_core.enh_conn_complete: local RPA 00:00:00:00:00:00
00> [00:00:12.130,096] <dbg> bt_conn.bt_conn_ref: handle 0 ref 1 -> 2
00> [00:00:12.130,096] <dbg> bt_conn.bt_conn_unref: handle 0 ref 2 -> 1
00> [00:00:12.130,126] <dbg> bt_conn.bt_conn_ref: handle 0 ref 1 -> 2
00> [00:00:12.130,126] <dbg> bt_adv.bt_le_adv_resume: No valid legacy adv
00> [00:00:12.130,126] <dbg> bt_conn.bt_conn_set_state: connect-adv -> connected
00> [00:00:12.130,157] <dbg> bt_att.bt_att_accept: conn 0x20000980 handle 0
00> [00:00:12.130,187] <dbg> bt_l2cap.bt_l2cap_chan_add: conn 0x20000980 chan 0x20005348
00> [00:00:12.130,187] <dbg> bt_att.bt_att_connected: chan 0x20005348 cid 0x0004
00> [00:00:12.130,187] <dbg> bt_att.att_chan_attach: att 0x2000d84c chan 0x20005340 flags 0
00> [00:00:12.130,218] <dbg> bt_att.bt_att_status: chan 0x20005348 status 0x20005388
00> [00:00:12.130,218] <dbg> bt_l2cap.l2cap_accept: conn 0x20000980 handle 0
00> [00:00:12.130,218] <dbg> bt_l2cap.bt_l2cap_chan_add: conn 0x20000980 chan 0x20000b00
00> [00:00:12.130,218] <dbg> bt_l2cap.l2cap_connected: ch 0x20000b00 cid 0x0005
00> [00:00:12.130,218] <dbg> bt_smp.bt_smp_accept: conn 0x20000980 handle 0
00> [00:00:12.130,249] <dbg> bt_l2cap.bt_l2cap_chan_add: conn 0x20000980 chan 0x20000d08
00> [00:00:12.130,279] <dbg> bt_conn.bt_conn_set_state: disconnected -> connect-adv
00> [00:00:12.130,279] <dbg> bt_conn.bt_conn_ref: handle 0 ref 1 -> 2
00> [00:00:12.130,279] <dbg> bt_hci_core.bt_hci_cmd_create: opcode 0x2039 param_len 6
00> [00:00:12.130,310] <dbg> bt_hci_core.bt_hci_cmd_create: buf 0x2000d748
00> [00:00:12.130,310] <dbg> bt_hci_core.bt_hci_cmd_send_sync: buf 0x2000d748 opcode 0x2039 len 9
00> [00:00:12.130,340] <dbg> bt_hci_core.process_events: count 2
00> [00:00:12.130,340] <dbg> bt_hci_core.process_events: ev->state 4
00> [00:00:12.130,340] <dbg> bt_hci_core.send_cmd: calling net_buf_get
00> [00:00:12.130,371] <dbg> bt_hci_core.send_cmd: calling sem_take_wait
00> [00:00:12.130,371] <dbg> bt_hci_core.send_cmd: Sending command 0x2039 (buf 0x2000d748) to driver
00> [00:00:12.130,371] <dbg> bt_hci_core.bt_send: buf 0x2000d748 len 9 type 0
00> [00:00:12.130,371] <dbg> bt_ctlr_hci_driver.hci_driver_send: enter
00> [00:00:12.130,432] <dbg> bt_buf.bt_buf_get_cmd_complete: sent_cmd 0x2000d748
00> [00:00:12.130,432] <dbg> bt_ctlr_hci_driver.cmd_handle: Replying with event of 6 bytes
00> [00:00:12.130,432] <dbg> bt_hci_core.hci_cmd_complete: opcode 0x2039
00> [00:00:12.130,462] <dbg> bt_hci_core.hci_cmd_done: opcode 0x2039 status 0x07 buf 0x2000d748[0m
00> [00:00:12.130,462] <dbg> bt_ctlr_hci_driver.hci_driver_send: exit: 0
00> [00:00:12.130,462] <dbg> bt_hci_core.process_events: ev->state 1
00> [00:00:12.130,493] <dbg> bt_conn.bt_conn_prepare_events: 
00> [00:00:12.130,493] <dbg> bt_conn.conn_prepare_events: Adding conn 0x20000980 to poll list
00> [00:00:12.130,493] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 3 events
00> [00:00:12.130,523] <wrn> bt_hci_core: opcode 0x2039 status 0x07
00> [00:00:12.130,523] <err> bt_adv: Failed to start advertiser
00> [00:00:12.130,523] <dbg> bt_conn.bt_conn_set_state: connect-adv -> disconnected
00> [00:00:12.130,554] <dbg> bt_conn.bt_conn_unref: handle 0 ref 2 -> 1
00> [00:00:12.130,554] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1 -> 0
00> [00:00:12.130,554] <dbg> bt_adv.bt_le_adv_resume: No valid legacy adv
00> [00:00:12.130,554] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x20000880
00> [00:00:12.130,584] <err> os: r3/a4:  0x00000036 r12/ip:  0x00000000 r14/lr:  0x000029b3
00> [00:00:12.130,584] <err> os:  xpsr:  0x61000000
00> [00:00:12.130,584] <err> os: Faulting instruction address (r15/pc): 0x00001c4e
00> [00:00:12.130,584] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
00> [00:00:12.130,584] <err> os: Current thread: 0x20000f08 (unknown)
00> [00:00:13.856,811] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: GNU Arm Embedded 10 2020-q4-major
 - Commit SHA or Version used: f282c93b9a71baf411b20c33cb8e4f4de2570156

cvinayak, 2021-03-31:
I will take a look on the `status 0x07` 

but,

> We cannot ship product unless we can obtain multiple connections to central devices

Extended advertising is experimental and under development. If you are looking for immediate production solution, take a look at nRF Connect SDK's SoftDevice Controller alternative.

cvinayak, 2021-04-01:
As  the connection complete event is processed ahead of the advertising terminate event, the advertising terminate event memory has not yet been processed and released, hence, immediately starting the advertiser again in the connected callback leads to the failure.

@joerchan Should the host permit restarting advertisement before host has processed the advertising terminated event?

jhedberg, 2021-04-01:
Is this a race caused by the `bt_recv()` and `bt_recv_prio()` distinction? Perhaps the advertising terminate event should be reclassified as a priority event to guarantee correct ordering?

cvinayak, 2021-04-01:
There isn't any race here, the advertising terminate event (with connection handle) comes after the enhanced connection complete event. 

The user is calling `bt_le_ext_adv_start` in the `connected` callback, any reason `bt_le_ext_adv_start` does not test `BT_ADV_ENABLED ` flag (which is cleared later in the handling of advertising terminate event)?

joerchan, 2021-04-06:
@cvinayak @jhedberg We might just have to document that the advertiser needs to be started in the advertising connected callback instead of the connection connected callback.
This is also related to this issue: https://github.com/zephyrproject-rtos/zephyr/issues/30699, which is to have the host manage this automatically.

cvinayak, 2021-04-06:
> We might just have to document that the advertiser needs to be started in the advertising connected callback instead of the connection connected callback.

Agree.

@joerchan But what about the missing check on `BT_ADV_ENABLED` in `bt_le_ext_adv_start`? Do you want me to send a PR?

joerchan, 2021-04-06:
> > We might just have to document that the advertiser needs to be started in the advertising connected callback instead of the connection connected callback.
> 
> Agree.
> 
> @joerchan But what about the missing check on `BT_ADV_ENABLED` in `bt_le_ext_adv_start`? Do you want me to send a PR?

I can fix it while I update the documentation.

PR #35534 (2021-05-25): Bluetooth: host: Add check for already enabled in advertising set start
Add check for the advertising already being enabled when attempting to
start the advertising set.
Document that the advertising set cannot be started from the connection
connected callback, and instead has to be started from the advertising
set connected callback.

Fixes: #33897
closed 2021-05-25 by carlescufi