zephyrproject-rtos/zephyr #33901: tests: interrupt: irq_enable() and irq_disable() do not work with direct and regular interrupt on x86

enjiamai, 2021-03-31:
**Describe the bug**
When trying to add test cases of the direct interrupt by IRQ_DIRECT_CONNECT() and the regular interrupt by IRQ_DIRECT_CONNECT() on x86,  irq_enable() and irq_disable() do not work well.
Note: I am not sure if this is really a bug or just some limitation on the usage, so file an issue for it.

**To Reproduce**
Steps to reproduce the behavior:
1.  Got patch or cherry this unmerged PR #33763 
2.  Uncomment line 128~132 in tests/kernel/interrupt 
3.  Run command : 
     twister -T tests/kernel/interrupt -p qemu_x86 
     or
     west build tests/kernel/interrupt/ -b qemu_x86 -p auto -DCONFIG_COMPILER_OPT='"-mgeneral-regs-only"' -t run
4. See error message of test case.

**Expected behavior**
The test case is passed as expected.

**Impact**
New issue. Need to pay attention to it. 

**Logs and console output**
```
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.4.0-5940-gfb9b6770a5cf  ***
Running test suite interrupt_feature
===================================================================
START - test_isr_dynamic
vector(48)
 PASS - test_isr_dynamic in 0.1 seconds
===================================================================
START - test_nested_isr
 SKIP - test_nested_isr in 0.1 seconds
===================================================================
START - test_prevent_interruption
locking interrupts
unlocking interrupts
timer fired
 PASS - test_prevent_interruption in 2.1 seconds
===================================================================
START - test_direct_interrupt
irq(3)=vector(34)
irq(4)=vector(33)

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/interrupt/src/direct_isr.c:128: test_direct_interrupt: (direct_int_executed[0] == 1 && direct_int_executed[1] == 1 is false)
Both ISR should not execute again
 FAIL - test_direct_interrupt in 0.2 seconds
===================================================================
Test suite interrupt_feature failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.12.3
 - Commit: https://github.com/zephyrproject-rtos/zephyr/pull/33763/commits/fb9b6770a5cf0069ed443eadcb723f9c8993f8c7  

**Additional context**
N/A

jhedberg, 2021-04-07:
@andyross I remember seeing some internal discussion wrt to this. IIRC there was some speculation about whether the feature should be completely removed? Is someone already looking into this?

enjiamai, 2021-05-11:
Hi, @nashif ,  no, I'll try to fix it before the 2.6 release.

enjiamai, 2021-05-21:
After studying, I think this is not a bug. In this testcase we use INT instruction to trigger an interrupt.

But the INT instruction does not cause a CPU interrupt. It's a synchronous trap, handled as an exception by the hardware. That's why the irq_disable() and even irq_lock() don't work here. (for X86)

The direct interrupt, regular interrupt, and dynamic interrupt cannot be masked by irq_disable(), because it needs a really IRQ interrupt such as timer. This was already verified workable in tests/kernel/context, test_kernel_timer_interrupts().  

Now we using APIC to send IPI to CPU itself to test our interrupt, irq_lock() works, it's more close to the really interrupt of IRQ but is still not.

So I tried to submit PR#35548 to update the incorrect usage of the testcases. I think this issue can be closed. 
@nashif  @andyross , do you think that make sense?  thank you!

enjiamai, 2021-08-17:
I would like to close this issue because I reported this issue with incorrect understanding.

The CPU exception cannot be masked by INT instruction by irq_disable() or irq_lock(). 
The IPI can be masked by irq_lock() but irq_disable().
The external interrupt can be masked by both irq_disable() or irq_lock().

Sorry, this is an incorrect issue reported by me, so I close it. If there is anything needed to be correct, please re-open this issue, thanks!

closed 2021-08-17 by enjiamai