zephyrproject-rtos/zephyr #33911: test:twr_ke18f: tests/kernel/sched/schedule_api - kernel_threads_sched_userspace cases meet out our space

hakehuang, 2021-03-31:
**Describe the bug**
kernel_threads_sched_userspace cases meet out our space

**To Reproduce**
Steps to reproduce the behavior:
tests/kernel/sched/schedule_api
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1010_evk ..
3. make
4. See error

**Expected behavior**
in former build this cases are PASS,but now the code size is too large

**Impact**
unknown

**Logs and console output**

```
+ docker exec confident_sinoussi build_zephyr_elf.sh twr_ke18f_kernel3_master tests/kernel/sched/schedule_api twr_ke18f build_3ebcce0 -DCONFIG_TIMESLICING=y kernel.scheduler tests/kernel/sched/schedule_api -DCONFIG_TIMESLICING=y

/build/src/workspace/twr_ke18f_kernel3_master

Including boilerplate (Zephyr base): /build/src/workspace/twr_ke18f_kernel3_master/zephyr/cmake/app/boilerplate.cmake

-- Application: /build/src/workspace/twr_ke18f_kernel3_master/zephyr/tests/kernel/sched/schedule_api

-- Zephyr version: 2.5.99 (/build/src/workspace/twr_ke18f_kernel3_master/zephyr)

-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 

-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")

-- Board: twr_ke18f

-- Cache files will be written to: /root/.cache/zephyr

-- Using toolchain: zephyr 0.12.2 (/opt/zephyr-sdk)

-- Found dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")

-- Found BOARD.dts: /build/src/workspace/twr_ke18f_kernel3_master/zephyr/boards/arm/twr_ke18f/twr_ke18f.dts

-- Generated zephyr.dts: /build/src/workspace/twr_ke18f_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_3ebcce0/zephyr/zephyr.dts

-- Generated devicetree_unfixed.h: /build/src/workspace/twr_ke18f_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_3ebcce0/zephyr/include/generated/devicetree_unfixed.h

-- Generated device_extern.h: /build/src/workspace/twr_ke18f_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_3ebcce0/zephyr/include/generated/device_extern.h

Parsing /build/src/workspace/twr_ke18f_kernel3_master/zephyr/Kconfig

Loaded configuration '/build/src/workspace/twr_ke18f_kernel3_master/zephyr/boards/arm/twr_ke18f/twr_ke18f_defconfig'

Merged configuration '/build/src/workspace/twr_ke18f_kernel3_master/zephyr/tests/kernel/sched/schedule_api/prj.conf'

Merged configuration '/build/src/workspace/twr_ke18f_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_3ebcce0/zephyr/misc/generated/extra_kconfig_options.conf'

Configuration saved to '/build/src/workspace/twr_ke18f_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_3ebcce0/zephyr/.config'

Kconfig header saved to '/build/src/workspace/twr_ke18f_kernel3_master/zephyr/tests/kernel/sched/schedule_api/build_3ebcce0/zephyr/include/generated/autoconf.h'

-- The C compiler identification is GNU 10.2.0

-- The CXX compiler identification is GNU 10.2.0

-- The ASM compiler identification is GNU

-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc

[... 481 lines cut ...]

Memory region         Used Size  Region Size  %age Used

           FLASH:       95204 B       512 KB     18.16%

            SRAM:       33880 B        32 KB    103.39%

        IDT_LIST:          0 GB         2 KB      0.00%

collect2: error: ld returned 1 exit status

make[2]: *** [zephyr/CMakeFiles/zephyr_final.dir/build.make:145: zephyr/zephyr.elf] Error 1

make[1]: *** [CMakeFiles/Makefile2:2603: zephyr/CMakeFiles/zephyr_final.dir/all] Error 2

make: *** [Makefile:84: all] Error 2

script returned exit code 2

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, )
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.5.0-1787-g59a3def756

hakehuang, 2021-03-31:
@MaureenHelm , twr_ke18f meets same issue as #33726

galak, 2021-03-31:
When did this build correct?  If I try building this on v2.4.0 and v2.5.0 I see similar size issues.

I don't think this is a bug.  A twister run for this platform doesn't actually generate any binaries and we've been treating out of space issues as not errors from a twister point of view for some time now.

hakehuang, 2021-04-01:
@galak yes it is skipped in former testing, I will add filter to this cases.

closed 2021-04-01 by hakehuang