zephyrproject-rtos/zephyr #33929: Sample app doesn't build if using Werror, logging with latest SDK

albertofloyd, 2021-03-31:
**Describe the bug**
Multiple errors in compilation (warning treated as errors) in Zephyr-based app with latest zephyr SDK zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
Able to repro

Any of these WAs allow to build
1) Use older SDK toolchain zephyr-sdk-0.11.2
2) Disable logging. 
3) Remove zephyr_compile_options(-Werror)

**To Reproduce**
Steps to reproduce the behavior:
Modify samples/helloworld as follows:

CMakelist.txt 
    zephyr_compile_options(-Werror)
prj.conf
   CONFIG_LOG=y

1. cd samples/helloworld
2. west build -c -p auto -b mec1501modular_assy6885 -- -G'Unix Makefiles' -DCONFIG_LOG=y -DCMAKE_VERBOSE_MAKEFILE=YES > build.log 2>&1" 
3. make
4. See error

**Expected behavior**
No errors as resulted of warnings in logging subsystem

**Impact**
Cannot enable warning detection as errors compilation option and use logging at app with latest toolchain.
Hence valid warnings cannot be caught.

**Logs and console output**
cd /home/mchp/sandbox/zephyr_snapshot/samples/hello_world/build/zephyr && ccache /home/mchp/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/mchp/sandbox/zephyr_snapshot/include -I/home/mchp/sandbox/zephyr_snapshot/samples/hello_world/build/zephyr/include/generated -I/home/mchp/sandbox/zephyr_snapshot/soc/arm/microchip_mec/mec1501 -I/home/mchp/sandbox/modules/hal/cmsis/CMSIS/Core/Include -I/home/mchp/sandbox/modules/hal/microchip/mec/common -I/home/mchp/sandbox/modules/hal/microchip/mec/mec1501 -isystem /home/mchp/sandbox/zephyr_snapshot/lib/libc/minimal/include -isystem /home/mchp/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/include -isystem /home/mchp/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/include-fixed -Os -imacros/home/mchp/sandbox/zephyr_snapshot/samples/hello_world/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros/home/mchp/sandbox/zephyr_snapshot/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/mchp/sandbox/zephyr_snapshot/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/mchp/sandbox/zephyr_snapshot=ZEPHYR_BASE -fmacro-prefix-map=/home/mchp/sandbox=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -Werror -MD -MT zephyr/CMakeFiles/zephyr.dir/subsys/logging/log_core.c.obj -MF CMakeFiles/zephyr.dir/subsys/logging/log_core.c.obj.d -o CMakeFiles/zephyr.dir/subsys/logging/log_core.c.obj -c /home/mchp/sandbox/zephyr_snapshot/subsys/logging/log_core.c
In file included from /home/mchp/sandbox/zephyr_snapshot/subsys/logging/log_core.c:9:
/home/mchp/sandbox/zephyr_snapshot/subsys/logging/log_core.c: In function 'log_hexdump_sync':
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:219:9: error: array subscript <unknown> is outside array bounds of 'const struct log_backend[0]' [-Werror=array-bounds]
  219 |  return &__log_backends_start[idx];
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:65:33: note: while referencing '__log_backends_start'
   65 | extern const struct log_backend __log_backends_start[0];
      |                                 ^~~~~~~~~~~~~~~~~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:219:30: error: array subscript <unknown> is outside array bounds of 'const struct log_backend[0]' [-Werror=array-bounds]
  219 |  return &__log_backends_start[idx];
      |          ~~~~~~~~~~~~~~~~~~~~^~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:65:33: note: while referencing '__log_backends_start'
   65 | extern const struct log_backend __log_backends_start[0];
      |                                 ^~~~~~~~~~~~~~~~~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:219:30: error: array subscript <unknown> is outside array bounds of 'const struct log_backend[0]' [-Werror=array-bounds]
  219 |  return &__log_backends_start[idx];
      |          ~~~~~~~~~~~~~~~~~~~~^~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:65:33: note: while referencing '__log_backends_start'
   65 | extern const struct log_backend __log_backends_start[0];
      |                                 ^~~~~~~~~~~~~~~~~~~~
/home/mchp/sandbox/zephyr_snapshot/subsys/logging/log_core.c: In function 'dropped_notify':
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:219:9: error: array subscript <unknown> is outside array bounds of 'const struct log_backend[0]' [-Werror=array-bounds]
  219 |  return &__log_backends_start[idx];
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:65:33: note: while referencing '__log_backends_start'
   65 | extern const struct log_backend __log_backends_start[0];
      |                                 ^~~~~~~~~~~~~~~~~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:219:30: error: array subscript <unknown> is outside array bounds of 'const struct log_backend[0]' [-Werror=array-bounds]
  219 |  return &__log_backends_start[idx];
      |          ~~~~~~~~~~~~~~~~~~~~^~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:65:33: note: while referencing '__log_backends_start'
   65 | extern const struct log_backend __log_backends_start[0];
      |                                 ^~~~~~~~~~~~~~~~~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:219:30: error: array subscript <unknown> is outside array bounds of 'const struct log_backend[0]' [-Werror=array-bounds]
  219 |  return &__log_backends_start[idx];
      |          ~~~~~~~~~~~~~~~~~~~~^~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:65:33: note: while referencing '__log_backends_start'
   65 | extern const struct log_backend __log_backends_start[0];
[... 73 lines cut ...]
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:219:9: error: array subscript <unknown> is outside array bounds of 'const struct log_backend[0]' [-Werror=array-bounds]
  219 |  return &__log_backends_start[idx];
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:65:33: note: while referencing '__log_backends_start'
   65 | extern const struct log_backend __log_backends_start[0];
      |                                 ^~~~~~~~~~~~~~~~~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:219:30: error: array subscript <unknown> is outside array bounds of 'const struct log_backend[0]' [-Werror=array-bounds]
  219 |  return &__log_backends_start[idx];
      |          ~~~~~~~~~~~~~~~~~~~~^~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:65:33: note: while referencing '__log_backends_start'
   65 | extern const struct log_backend __log_backends_start[0];
      |                                 ^~~~~~~~~~~~~~~~~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:219:30: error: array subscript <unknown> is outside array bounds of 'const struct log_backend[0]' [-Werror=array-bounds]
  219 |  return &__log_backends_start[idx];
      |          ~~~~~~~~~~~~~~~~~~~~^~~~~
/home/mchp/sandbox/zephyr_snapshot/include/logging/log_backend.h:65:33: note: while referencing '__log_backends_start'
   65 | extern const struct log_backend __log_backends_start[0];
      |                                 ^~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
zephyr/CMakeFiles/zephyr.dir/build.make:400: recipe for target 'zephyr/CMakeFiles/zephyr.dir/subsys/logging/log_core.c.obj' failed

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu)
 - Toolchain: Zephyr SDK
 - Zephyr master

nordic-krch, 2021-04-08:
Tried the latest one `zephyr-sdk-0.12.4` and don't see those warnings.

albertofloyd, 2021-04-12:
@nordic-krch let us try upgrading to that SDK version

albertofloyd, 2021-04-18:
Unable to repro with master branch and latest SDK, hence closing

closed 2021-04-18 by albertofloyd