zephyrproject-rtos/zephyr #33950: periodic_adv not working with nRF5340 DK

NaderSameh, 2021-04-01:
**Describe the bug**

periodic_adv sample is not working with nrf5340dk_nrf5340_cpuapp

it is working for nrf52840dk_nrf52840 after adding CONFIG_BT_LL_SW_SPLIT=y to proj.conf

**To Reproduce**
Steps to reproduce the behavior:
Open the sample from the SDK. build for nrf5340dk_nrf5340_cpuapp 

**Expected behavior**
To work

**Impact**
showstopper

**Logs and console output**

> Starting Periodic Advertising Demo
> Failed to set periodic advertising parameters (err -5)
> [00:00:00.442,626] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
> [00:00:00.442,657] <inf> bt_hci_core: HW Variant: nRF53x (0x0003)
> [00:00:00.442,657] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.5 Build 99
> [00:00:00.444,641] <inf> bt_hci_core: Identity: CC:E6:FA:E1:A6:E6 (random)
> [00:00:00.444,641] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1
> [00:00:00.444,641] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff
> [00:00:00.447,082] <wrn> bt_hci_core: opcode 0x203e status 0x01

**Environment (please complete the following information):**
 - Zephyr SDK
 - SES

Would appreciate some help

cvinayak, 2021-04-01:
If you are using nRF connect SDK, please raise the issue at https://devzone.nordicsemi.com/

Otherwise, if using latest Zephyr, please provide the commit hash you are using, also please provide detailed build and flashing commands etc as was requested in `To Reproduce` template when creating the issue.

cvinayak, 2021-05-02:
Closing the issue as periodic advertising feature is experiment and work is still in progress.

closed 2021-05-02 by cvinayak