zephyrproject-rtos/zephyr #33962: example-application module: add subsystem zxa_module_hello

gregshue, 2021-04-01:
## Introduction

The `example-application` repository is to demonstrate how an external module may be organized and configured so that module features may be configured, built, and verified by the Zephyr build/CI system. This proposal is to add the simple subsystem `zxa_module_hello` as an example demonstrating:
  - subsystem configuration
  - subsystem interface visibility
  - subsystem CI integration
  - subsystem tying in to the Zephyr initialization structure
  - subsystem using the Zephyr infrastructure (e.g. logging system)
  - where to use a module abbreviation (Zephyr eXample Application) to reduce conflicts in global namespaces.

### Problem description

The scope of the `example-application` module includes demonstrating a way to configure modules so that subsystems in the module will be recognized by the Zephyr build system.

### Proposed change

This subsystem requires:
- adding the following sub-trees:
   - zephyr/include/zxa/
   - zephyr/subsys/zxa_module_hello/
   - zephyr/tests/subsys/zxa_module_hello/
- adding the necessary Kconfig, build, and test files for high coverage
- extend `module.yml` so that the build and CI system will build and run the subsystem tests

The subsystem itself uses the Zephyr INIT and logging infrastructure to provide a testable behavior for CI. It also provides one interface function to demonstrate that the build system is detecting the interface file and that the interface is callable by the tests.

Additionally, this example establishes naming patterns for module-specific symbols in global namespaces to minimize collisions as when multiple modules co-exist.

## Detailed RFC

In this section of the document the target audience is the dev team. Upon
reading this section each engineer should have a rather clear picture of what
needs to be done in order to implement the described feature.

### Proposed change (Detailed)

Since the implementation is quite simple and short, it will be declared below. Some key points:

- The module abbreviation is used as a prefix for all symbols in global namespaces.  These include symbols in Kconfig, CMake, C preprocessor, linker symbols, include paths, filenames, subsys names, test case identifiers, and usage should be extended to all symbols originating from this module.
- module-provided external headers are placed beneath the folder `zephyr/include/<module_abbrev>/` to provide clarity and predictability in the include file pathname.
- the module ties into the kernel init process so that it can be included in a trivial project: `void main(void) { }`.
- configuration, build, and CI files flow from the interface and implementation, following the existing pattern demonstrated in the `zephyr` repository.

Add `zephyr/include/zxa/zxa_module_hello.h`, containing:

```
/** @file
 *  @brief ZXA Module Hello API
 */

/*
 * Copyright (c) 2020 Legrand North America, LLC.
 *
 * SPDX-License-Identifier: Apache-2.0
 */
#ifndef ZXA_INCLUDE_ZXA_ZXA_MODULE_HELLO_H_
#define ZXA_INCLUDE_ZXA_ZXA_MODULE_HELLO_H_

/**
 * @brief ZXA Module Hello
 * @defgroup zxa_module_hello_apis Module Hello sAPIs
 * @{
 */

#include <stdint.h>

#ifdef __cplusplus
extern "C" {
#endif

/**
 * @brief Return the number of times module_hello said this power-up.
 *
 * @retval >= 0 If successful.
 * @retval Errno Negative errno code if failure.
 */
int32_t zxa_module_hello_said(void);

#ifdef __cplusplus
}
#endif

/**
 * @}
 */

#endif /* ZXA_INCLUDE_ZXA_ZXA_MODULE_HELLO_H_ */
```

Add `zephyr/subsys/zxa_module_hello/hello.c`, containing:

```
/*
 * Copyright (c) 2020 Legrand North America, LLC.
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr.h>
#include <init.h>
#include <sys/printk.h>
#include <zxa/zxa_module_hello.h>

#include <logging/log.h>
LOG_MODULE_REGISTER(zxa_module_hello);

int32_t zxa_module_hello_said(void)
{
        return 1;
}

static int init(const struct device *dev)
{
        ARG_UNUSED(dev);

        printk("Hello from " MODULE_NAME "!\n");
        LOG_INF("Hello from " STRINGIFY(MODULE_NAME));

        return 0;
}

SYS_INIT(init, APPLICATION, CONFIG_APPLICATION_INIT_PRIORITY);
```

Add `zephyr/subsys/zxa_module_hello/Kconfig`, containing:

```
# Copyright (c) 2020 Legrand North America, LLC.
# SPDX-License-Identifier: Apache-2.0

# Kconfig -

config ZXA_MODULE_HELLO
    bool "ZXA Module \"Hello\""
    select LOG
    help
      This option enables ZXA Module Hello service

if ZXA_MODULE_HELLO

module = ZXA_MODULE_HELLO
module-str = Log level for ZXA Module Hello
source "subsys/logging/Kconfig.template.log_config"

endif # ZXA_MODULE_HELLO
```

Add `zephyr/tests/subsys/zxa_module_hello/prj.conf`, containing:

```
CONFIG_ZTEST=y

CONFIG_LOG=y
CONFIG_ZXA=y
CONFIG_ZXA_MODULE_HELLO=y
```

### Dependencies

This only depends on the existing Zephyr config/build/CI system and a minimal set of stable kernel/infrastructure features.

### Concerns and Unresolved Questions

None.  The module was specified to follow the Zephyr directory tree and fit within the Zephyr config/build/CI systems.
Similarly, the namespace management follows the patterns established within the `zephyr` repository.

## Alternatives

The following trivial subsystems were considered, but were considered as unnecessarily complicated or impractically trivial for guiding new users.

1. just a function stub.
2. just an init sequenced call.
3. a thread running from a OS primitive (message queue, semaphore, condition variable).

gregshue, 2021-05-13:
Seeing that example-application has begun using `custom_*` for some naming patterns, the new subsystem should follow the same pattern.

gregshue, 2021-07-07:
An implementation in https://github.com/zephyrproject-rtos/example-application/pull/6 has been under review since May 13, 2021.

nashif, 2024-02-08:
This seems to be out of scope for the customer application, mostly showing module integration.

closed 2024-02-08 by nashif