zephyrproject-rtos/zephyr #33967: The printed total size differs from calculated from .json

wangnuannuan, 2021-04-02:
**Describe the bug**
I want to get the rom and ram size by reading the JSON file, so I use the script https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/footprint/size_report to parse the elf file. 

Take rom size as example, the script will print the total size in the terminal and generate a file `rom.json`. But when I use the following code to calculate the rom size in the JSON file, the printed total size differs from calculated from JSON file.
```
from anytree.importer import JsonImporter

importer = JsonImporter()
size = 0
with open("./rom.json", "r") as fp:
    root = importer.read(fp)

    for leave in root.leaves:
        size += leave.size
print(size)
```
@dcpleung @nashif

dcpleung, 2021-04-06:
The sizes of individual items are of the items themselves, while the total size takes into account of alignments and padding. That's why the total size is larger.

And one important thing to remember, not all symbols are itemized. There are symbols in the image (for example, strings) that do not appear in the symbol table, and thus not in the list but counted towards total size.

wangnuannuan, 2021-05-12:
https://github.com/zephyrproject-rtos/zephyr/pull/33112

closed 2021-05-12 by wangnuannuan