zephyrproject-rtos/zephyr #33968: ESP32 Porting GSM Module Compile Error

maweicoltsmart, 2021-04-02:
[1/8] Linking C executable zephyr\zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map
cmd.exe /C "cd . && C:\Users\Colt\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr\zephyr_prebuilt.elf  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=D:/ZephyrProject/code/zephyrproject/zephyr/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ppp/libsubsys__net__l2__ppp.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/conn_mgr/libsubsys__net__lib__conn_mgr.a  zephyr/subsys/net/lib/dns/libsubsys__net__lib__dns.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -L"c:/users/colt/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0"  -LD:/ZephyrProject/code/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lhal  -LD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/../components/xtensa/esp32 && cmd.exe /C "cd /D D:\ZephyrProject\code\zephyrproject\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E echo ""
c:/users/colt/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: zephyr/subsys/net/l2/ppp/libsubsys__net__l2__ppp.a(ppp_l2.c.obj):(.literal.ppp_recv+0x0): undefined reference to `_ppp_protocol_handler_list_start'
c:/users/colt/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: zephyr/subsys/net/l2/ppp/libsubsys__net__l2__ppp.a(ppp_l2.c.obj):(.literal.ppp_recv+0x4): undefined reference to `_ppp_protocol_handler_list_end'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'D:\ZephyrProject\code\zephyrproject\zephyr\build'

who can help me？

sylvioalves, 2021-04-06:
HI @maweicoltsmart, Please, can you try this:

Add this into `zephyr/soc/xtensa/esp32/linker.ld`:

L310:
```c
#if defined(CONFIG_NET_L2_PPP)
    . = ALIGN(4);
    Z_LINK_ITERABLE(ppp_protocol_handler)
    . = ALIGN(4);
#endif
```

maweicoltsmart, 2021-04-06:
The problem is solved, thank you for your help!

aep, 2021-05-11:
same issue.

@sylvioalves the line has changed, this won't apply anymore. does this go between shell and .rodate ?

sylvioalves, 2021-05-11:
@aep, yes, you can add this in L280 after 
 ```c
 #if defined(CONFIG_SETTINGS)
    . = ALIGN(4);
    Z_LINK_ITERABLE(settings_handler_static);
    . = ALIGN(4);
#endif
```
I missed pushing this. Doing it right now.

maweicoltsmart, 2021-05-28:
west build -b esp32 samples\net\gsm_modem

compile error.

c:/users/colt/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: zephyr/subsys/net/l2/ppp/libsubsys__net__l2__ppp.a(ppp_l2.c.obj):(.literal.ppp_recv+0x0): undefined reference to `_ppp_protocol_handler_list_start'
c:/users/colt/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: zephyr/subsys/net/l2/ppp/libsubsys__net__l2__ppp.a(ppp_l2.c.obj):(.literal.ppp_recv+0x4): undefined reference to `_ppp_protocol_handler_list_end'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'D:\ZephyrProject\code\zephyrproject\zephyr\build'

sylvioalves, 2021-05-28:
@maweicoltsmart Have you pulled latest main and run `west update`? It works in here. Please, let me know.

maweicoltsmart, 2021-05-28:
Sorry, it is indeed because I have not synced, thank you!

closed 2021-05-28 by maweicoltsmart