zephyrproject-rtos/zephyr #33974: The stm32wb55rc MCU does not operate on zephyr

wngur300, 2021-04-02:
**Describe the bug**
I changed the mcu to **stm32wb55rc** by configuring the circuit of the **nucleo_wb55rg** development board similarly.
We reduced the sram size and flash size in the device tree and tested it.
I tried to build a sample source of samples/basic/blinky, but the OSC_IN, OSC_OUT pin does not produce a clock of 32MHZ and does not work.
However, both work well when constructed via Keil compiler with either **wb55rg** or **wb55rc** via STM cubemx (mcu is wb55rc, but the configuration of the **wb55rg** board also works).)
And the clock configuration of **wb55rg** and **wb55rc** was the same.
Eventually, with the **nucleo_wb55rg** board setting, we believe that a function like gpio should work when **wb55rc**,
but it does not work well.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nucleo_wb55rg ..
3. make

Environment (please complete the following information)
OS: Linux ubuntu 18.04
Toolchain: zephyr-sdk-0.11.4
Commit SHA or Version used: 7a3b253

**Environment (please complete the following information):**
 - OS: 
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
The circuit on the nucleo_wb55rg development board is the same as configured with SRAM and wb55rc with a smaller FLASH capacity.
When built with the nucleo_wb55rg board, the base of the sram and flash memory map are the same, so I think I just need to reduce the size of them.
I tested the stm32wb55Xg.dtsi file by modifying it, but the default gpio on/off behavior is not working.
However, when configured with stm32wb55rg mcu in cubemx and compiled, the wb55rc board acts as gpio.
Why is the board not normal with wb55rc?

GunZi200, 2021-04-02:
What pin are you using for the blink?

The nucleo_wb55rg expect one of the pins: PB5, PB0, or PB1. 

Though I expect only PB0 will work with the blinky example since it is given the alias `led0`

https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nucleo_wb55rg/nucleo_wb55rg.dts

GunZi200, 2021-04-02:
> the OSC_IN, OSC_OUT pin does not produce a clock of 32MHZ and does not work.

It seems to me this is your main issue. HSE is the system clock, without it nothing happens.

1. Does this problem occur with `nucleo_wb55rg` or with the `rc` variant?
2. Do you have an oscilloscope to measure the clock frequency?
3. Can you confirm the HSE clock is actually turned on and is 32MHz?

The GPIO should absolutely work, I agree!

wngur300, 2021-04-06:
> > the OSC_IN, OSC_OUT pin does not produce a clock of 32MHZ and does not work.
> 
> It seems to me this is your main issue. HSE is the system clock, without it nothing happens (at least as far as I understand the default Zephyr settings).
> 
> 1. Does this problem occur with `nucleo_wb55rg` or with the `rc` variant?
> 2. Do you have an oscilloscope to measure the clock frequency?
> 3. Can you confirm the HSE clock is actually turned on and is 32MHz?
> 
> The GPIO should absolutely work, I agree!

 I tried attaching wb55rc to the nucleo_wb55rg board.
[image]

**LED on/off works well on the nucleo_wb55rg board with wb55rc MCU. (built with nucleo_wb55rg board setting).**
**Checking the waveforms of the OSC_IN and OSC_OUT pins also produces a good clock (32 MHz).**

By the way, my circuit diagram, which was produced referring to the circuits on the nucleo_wb55rg board, does not produce clocks on the OSC_IN and OSC_OUT pins.
The LED (PE4) is not controlled on/off (built with the nucleo_wb55rg board setting).

This is a circuit diagram that is constructed.
[image]

However, this circuit diagram does not work when it is built from zephyr, and when it is built using a keil compiler, both boards operate.
After all, it seems that my circuit doesn't work when compiled in a zephyr environment.

I tried changing the clock setting in the nucleo_wb55rg_defconfig file, but it does not work.

It cannot be considered a circuit problem because the gpio operates normally with a keil compiler using cubemx.
However, on the nucleo_wb55rg board with wb55rc attached, it works fine with zephyr, so it's an ambiguous situation that's not even a problem with zephyr...

Why doesn't my circuit work normally on zephyr..?

erwango, 2021-04-06:
@wngur300 This is indeed strange. I suspect a clock configuration issue.
Can you check that the clock configuration used with CubeMx is the exact same one as used on Zephyr ?

wngur300, 2021-04-06:
> @wngur300 This is indeed strange. I suspect a clock configuration issue.
> Can you check that the clock configuration used with CubeMx is the exact same one as used on Zephyr ?

Yes, this is the information I set as cubemx. (Please click on the screenshot)
[image]

**Changed to CONFIG_CLOCK_STM32_CPU2_PRESCALER=2 as per CubeMax setting:**
[image]

**And I set up the board as follows:**

[image]

[image]

The nucleo_wb55rg board with wb55rg mcu is functioning normally in the zephyr. 
However, it doesn't work with the circuit I configured.
However, when compile with cubemx, both work well...

FRASTM, 2021-04-06:
I confirm : no pb seen sofar with this sample config. of the nucleo wb55rg 
```
*** Booting Zephyr OS build zephyr-v2.5.0-1956-g5cc20faa2f8e  ***
Hello World! nucleo_wb55rg
```

erwango, 2021-04-13:
Moving this to 'question' as this is reported on a SoC that is not yet supported with zephyr, w/o board supported.

@wngur300 Do you confirm the issue is till present ?

wngur300, 2021-04-26:
R1 (1M Ohm) of the circuit diagram captured above was removed and operated normally.
```
config_enable_default_clocks() :
        while (!LL_RCC_LSE_IsReady()) {
                /* Wait for LSE ready */
       }
```
The LSE was not ready and was unable to exit the repeat.
And even if the R1 resistance was removed from the circuit diagram, the C5 capacitor had the same symptoms with 10 pF.

erwango, 2021-04-26:
@wngur300 Thanks for this feedback!

So from what I understand, this is not linked with Zephyr implementation, as you would face the same issue on CubeMx if using LSE. If you agree, can you close the point ?

wngur300, 2021-04-28:
> @wngur300 Thanks for this feedback!
> 
> So from what I understand, this is not linked with Zephyr implementation, as you would face the same issue on CubeMx if using LSE. If you agree, can you close the point ?

**The following configuration in cubemx also booted (LSE settings):**
[image]

It is assumed that the code generated by cubemx was booted because it did not check registers to verify that the LSE was ready, such as the LL_RCC_LSE_IsReady() function.
Currently, PC14 and PC15 cannot be tested because the correct resistance and capacitors are soldered.

wngur300, 2021-05-25:
It was a hardware problem, so I'll close it.

closed 2021-05-25 by wngur300