zephyrproject-rtos/zephyr #33977: Question: How best to contribute drivers upstream?

adamfowleruk, 2021-04-02:
Hello all,

On the Herald Project we've been asked to put together some opensource hardware designs. (To be published soon). These will use some external ICs that may not have drivers in Zephyr RTOS already.

I have a holding place for drivers that we may create for these devices, but really I'd prefer to contribute them back upstream to this project.

Do you have any past best practice for people that create drivers for their own needs/products immediately, and contribute them back upstream to Zephyr for appearance in a later release?

The main issue I see is that of time lag vs. supportability. We need those taking on our opensource hardware and software to grab the latest release of the Nordic SDK, add in our code project, and then build for the target platform. The consumers are Public Health Authorities and so can't really work against the bleeding edge develop branches of Zephyr, or to patch NCS. So we need a way to give them any missing drivers immediately, but then make sure we contribute them back upstream to your project to ensure everyone else can benefit.

Any advice greatly appreciated. Hardware and device drivers not yet uploaded (should be in the next 1-2 weeks), but they will live here:-
https://github.com/theheraldproject/herald-hardware
https://github.com/theheraldproject/zephyr-devices

With our C++ based API that uses the data from these things living here:
https://github.com/theheraldproject/herald-for-cpp

Thanks in advance.

carlescufi, 2021-04-06:
As long as you are willing to contribute and then support the platform and driver support, then the actual board does not need to be available commercially at the moment of the contribution. That said, you will need to document very carefully what hardware the drivers and boards are targeting.

adamfowleruk, 2021-04-21:
OK great. The hardware designs (still in development, initial run in the coming weeks) are now live on the above repo. I've also created 1 custom driver so far for a generally available and common sensor. Once I have the set all working I'll look to issue a PR for the hardware and drivers.

carlescufi, 2021-04-28:
Thanks! Closing the question for now, feel free to reopen if you have more questions.

closed 2021-04-28 by carlescufi