zephyrproject-rtos/zephyr #33978: MCP2515 wrong BRP value

ustinov-dev, 2021-04-02:
**Describe the bug**
There is a wrong BRP register calculation for **mcp2515** after  [8b6c1bd4b7a0](https://github.com/zephyrproject-rtos/zephyr/commit/8b6c1bd4b7a0ae4e1d1b3819d5db728120218c6b) commit (**zephyr-v2.5.0**).

In previous versions(like  [zephyr-v2.4.0](https://github.com/zephyrproject-rtos/zephyr/blob/7a3b253ced7333f5c0269387a7f3ed1dee69739d/drivers/can/can_mcp2515.c#L312))  for this calculation was used the formula from datasheet: `uint8_t brp = (dev_cfg->osc_freq / (bit_length * bitrate * 2)) - 1;` It's equal to `uint8_t brp = prescaler - 1;`

But after  [8b6c1bd4b7a0](https://github.com/zephyrproject-rtos/zephyr/commit/8b6c1bd4b7a0ae4e1d1b3819d5db728120218c6b)  we have:  `uint8_t brp = timing->prescaler;`. And it's a wrong value for MCP2515 BRP register.

When I use `uint8_t brp = timing->prescaler;`( [line 336](https://github.com/zephyrproject-rtos/zephyr/blob/fe7c2efca800a0cf1bccf23aefe08b3c4beb88bf/drivers/can/can_mcp2515.c#L336) can_mcp2515.c) I can't work with CAN.
And if I change it to `uint8_t brp = timing->prescaler - 1;`, everything works fine for me.
But in my case `timing->prescaler` is always bigger than 1.

carlescufi, 2021-04-06:
@ustinov-dev perhaps you could send a PR with a fix?

ustinov-dev, 2021-04-09:
I have done PR #34161

commit 73db4433 (2021-04-15): drivers: can: can_mcp2515: fix BRP bits value

Fixes #33978.
8b6c1bd commit set 'presacaler' in BRP(Baud Rate Prescaler) bits.
It's wrong, because these bits from CNF1 register use 'prescaler - 1'.
MCP2515(DS20001801J) datasheet 'REGISTER 5-1: CNF1: CONFIGURATION
REGISTER 1 (ADDRESS: 2Ah)'.
closed 2021-04-15 by carlescufi