zephyrproject-rtos/zephyr #33986: TCP stack doesn't handle data received in FIN_WAIT_1

jimparis, 2021-04-02:
The TCP stack doesn't appear to handle data received in `FIN_WAIT_1`.  What I'm seeing is this, whenever HTTPS connection closes:

[image]

It looks like what happens is the client (192.x) and the server (75.x) are both trying to close the connection at the same time.  I'm guessing, from the client's point of view, things happen in a slightly different order:

- client sends packet 374 (FIN+ACK, ack=3229)
- client enters `FIN_WAIT_1`
- client receives packet 372 (23 bytes of data.  **The bug is that this length is ignored**).
- client receives packet 373 (FIN+ACK, seq=3252) but is still expecting seq=3229, so it never answers
- server retransmits until everyone gives up

This seems to fix the problem:
```diff
diff --git a/subsys/net/ip/tcp2.c b/subsys/net/ip/tcp2.c
index c6cf60c9a8fe..b8029f8e510b 100644
--- a/subsys/net/ip/tcp2.c
+++ b/subsys/net/ip/tcp2.c
@@ -1777,6 +1777,9 @@ next_state:
                do_close = true;
                break;
        case TCP_FIN_WAIT_1:
+               /* Acknowledge but drop any data */
+               conn_ack(conn, + len);
+
                if (th && FL(&fl, ==, (FIN | ACK), th_seq(th) == conn->ack)) {
                        tcp_send_timer_cancel(conn);
                        conn_ack(conn, + 1);
```

With that, the packet capture looks great:

[image]

jukkar, 2021-04-06:
Your fix looks good, can you send a PR for it?

jukkar, 2021-05-14:
@jimparis I submitted your patch proposal in order to get this merged to 2.6, please +1 it if ok.

PR #35289 (2021-05-25): net: tcp2: Ack any data received in FIN_WAIT_1 state
If we receive any data in FIN_WAIT_1, then ack it even if we
are discarding it.

Fixes #33986
commit 0f691238 (2021-05-25): net: tcp2: Ack any data received in FIN_WAIT_1 state

If we receive any data in FIN_WAIT_1, then ack it even if we
are discarding it.

Fixes #33986
closed 2021-05-25 by nashif