zephyrproject-rtos/zephyr #33994: kscan_ft5336 doesn't provide proper up/down information when polling, and hogs resources in interrupt mode

dhoove, 2021-04-03:
The 'pressed' value that is provided by the callback is only 1 if the touchpad was being pressed at the moment it was being sampled. If you press quickly between sampling periods, 'pressed' will always remain zero. One way to solve this problem is by activating interrupt mode. But in this case, the interrupts are generated at high frequency (approximatly 40Hz), causing significant I2C traffic and hogging resources.

To reproduce:
- set `KSCAN_FT5336_PERIOD` to 1000
- set `KSCAN_LOG_LEVEL_INF` to `Debug` so that you can observe the touch events in the log
- create a devicetree alias to `touch_controller` named `kscan0`
- make sure the touch_controller has a int-gpios line defining the IRQ line.
- make sure that touch_controller's i2c interface is up and running
- create the device with `device_get_binding(DT_LABEL(DT_ALIAS(kscan0)))`
- attach a callback with `kscan_config()`, and enable the callback with `kscan_enable_callback()`.
- observe the 'pressed' column of the log, while quickly touching and releasing the display. Most of the time, the `pressed` value will never change to 1.

This problem is disturbing because accurate press detection results in high CPU use, both in polling and interrupt mode.

MaureenHelm, 2021-04-06:
Which board are you using?

dhoove, 2021-04-08:
Hi,I just added a board file to the PR for the STM32L476 nucleo board.Cheers,David

gmarull, 2021-04-08:
related PR: https://github.com/zephyrproject-rtos/zephyr/pull/33996

closed 2021-08-04 by github-actions[bot]