zephyrproject-rtos/zephyr #33998: gsm_ppp: Adding power key sequence for Quectel and possibly other modems

ycsin, 2021-04-03:
**Is your enhancement proposal related to a problem? Please describe.**
Some modems like Quectel has a power key pin that requires a special sequence to power on or off the modem. The current `gsm_ppp` doesn't take care of that. With this change the `gsm_ppp` driver could increase its coverage to at least the Quectel EC2X series.

**Describe the solution you'd like**
I'm suggesting that this power on sequence to be added to `gsm_ppp_start` & power off sequence to be added to `gsm_ppp_stop` such that the on or off of the driver can be controlled by the driver itself instead of application.

**Describe alternatives you've considered**
Creating another driver based on the current `gsm_ppp` but for device that requires power on/off sequence is also possible, but I'm not sure if this is the right way going forward as that might end up having different PPP drivers with only minimal differences.

**Additional context**
I tested ARM's mbed-os for a short while before I decided to move on to Zephyr. I'm using Quectel EC21 and in mbed-os the EC21 is basically plug and play, the EC2X driver only contains some AT commands and power on/off sequence.

jukkar, 2021-04-03:
All enhancements are welcome so if you have a solution for this issue, just send a PR. People can then review this and comment.

ycsin, 2021-04-14:
I'd love to try it, but how should I use the pin from dts without dts binding file?