zephyrproject-rtos/zephyr #34007: [Coverity CID: 220534] Out-of-bounds access in tests/arch/arm/arm_no_multithreading/src/main.c

zephyrbot, 2021-04-03:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b86f7addae05add0db45d9b528854235fbb93a48/tests/arch/arm/arm_no_multithreading/src/main.c#L104

Category: Memory - corruptions
Function: `test_main`
Component: Tests
CID: [220534](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=220534)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b86f7addae05add0db45d9b528854235fbb93a48/tests/arch/arm/arm_no_multithreading/src/main.c#L104

```
98    
99         __ASSERT(i >= 0,
100             "No available IRQ line to use in the test\n");
101    
102         printk("Available IRQ line: %u\n", i);
103    
>>>     CID 220534:  Memory - corruptions  (OVERRUN)
>>>     Overrunning callee's array of size 128 by passing argument "i" (which evaluates to 4294967295) in call to "arch_irq_connect_dynamic".
104         arch_irq_connect_dynamic(i, 0 /* highest priority */,
105             arm_isr_handler,
106             NULL,
107             0);
108    
109         NVIC_EnableIRQ(i);
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

ioannisg, 2021-04-15:
That's a false positive - as the ASSERT statement above will catch the case if i becomes negative. Somehow we need to tell coverity that ASSERTS are ON, on tests. I'll "fix" this in the code, anyways...

ioannisg, 2021-04-20:
Need to close this manually, since I had forgotten to properly add the "Fixes" comment in the PR

closed 2021-04-20 by ioannisg