zephyrproject-rtos/zephyr #34011: NRF52840dongle DTS bug

RenierVDWBluerain, 2021-04-04:
When I tried to implement PWM for the onboard rgb led on an nrf52840 dongle an error occurs in the devicetree_unfixed.h file zephyr/include/generated/devicetree_unfixed.h:1306:36: error: 'DT_N_S_pwmleds_S_pwm_led_0' undeclared

the same error occurs for ...led_1, 2 etc.

**Code**
#define RED_NODE DT_ALIAS(red_pwm_led)
#if DT_NODE_HAS_STATUS(RED_NODE, okay)
#define RED_CTLR_NODE	DT_PWMS_CTLR(RED_NODE)
#define RED_CHANNEL	DT_PWMS_CHANNEL(RED_NODE)
#define RED_FLAGS	DT_PWMS_FLAGS(RED_NODE)
enum { RED, GREEN, BLUE };
const struct device *pwm_dev[3];
pwm_dev[RED] = DEVICE_DT_GET(RED_CTLR_NODE); //This line invokes the error

**Logs and console output**
zephyr/include/generated/devicetree_unfixed.h:1306:36: error: 'DT_N_S_pwmleds_S_pwm_led_0' undeclared

**Environment (please complete the following information):**
 - Windows 10
 - Segger nrf connect 1.5.0 sdk Zephyr

RenierVDWBluerain, 2021-04-04:
#define DT_N_ALIAS_red_pwm_led     DT_N_S_pwmleds_S_pwm_led_0 //this is where the actuall error occurs devicetree_unfixed.h

mbolivar-nordic, 2021-04-06:
> pwm_dev[RED] = DEVICE_DT_GET(RED_CTLR_NODE); //This line invokes the error

Are you missing an include? I'm guessing the full compiler output looks like this:

```
/home/mbolivar/zp/zephyr/samples/hello_world/src/main.c:19:17: warning: implicit declaration of function 'DEVICE_DT_GET' [-Wimplicit-function-declaration]
   19 |  pwm_dev[RED] = DEVICE_DT_GET(RED_CTLR_NODE); //This line invokes the error
      |                 ^~~~~~~~~~~~~
/home/mbolivar/zp/zephyr/samples/hello_world/src/main.c:19:31: error: 'DT_N_S_soc_S_pwm_4001c000' undeclared (first use in this function)
   19 |  pwm_dev[RED] = DEVICE_DT_GET(RED_CTLR_NODE); //This line invokes the error
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~
```

At any rate, I can't reproduce this as this C file builds fine for me:

```c
#include <zephyr.h>
#include <devicetree.h>
#include <device.h>

#define RED_NODE DT_ALIAS(red_pwm_led)
#define RED_CTLR_NODE DT_PWMS_CTLR(RED_NODE)
#define RED_CHANNEL DT_PWMS_CHANNEL(RED_NODE)
#define RED_FLAGS DT_PWMS_FLAGS(RED_NODE)
enum { RED, GREEN, BLUE };

void main(void)
{
	const struct device *pwm_dev[3];
	pwm_dev[RED] = DEVICE_DT_GET(RED_CTLR_NODE);
}
```

Changing from bug to question.

shachar-oren, 2021-05-03:
I have the same problem when compiling for the nRF52840dk. The following code produces the same error:

```
#include <zephyr.h>
#include <devicetree.h>
#include <device.h>

#define PWM_NODE DT_ALIAS(pwm_led0)
#define PWM_CTLR_NODE DT_PWMS_CTLR(PWM_NODE)
#define PWM_CHANNEL DT_PWMS_CHANNEL(PWM_NODE)
#define PWM_FLAGS DT_PWMS_FLAGS(PWM_NODE)

void main(void)
{
	const struct device *pwm_dev;
	pwm_dev = DEVICE_DT_GET(PWM_CTLR_NODE);
}
```

Relevant compiler output:
```
1> C:/ncs/v1.5.1/zephyr/include/device.h:46:39: warning: implicit declaration of function '__device_DT_PWMS_CTLR' [-Wimplicit-function-declaration]
1> C:/ncs/v1.5.1/zephyr/include/toolchain/common.h:124:26: note: in definition of macro '_DO_CONCAT'
1> C:/ncs/v1.5.1/zephyr/include/device.h:46:31: note: in expansion of macro '_CONCAT'
1> C:/ncs/v1.5.1/zephyr/include/device.h:205:37: note: in expansion of macro 'DEVICE_NAME_GET'
1> C:/ncs/v1.5.1/zephyr/include/device.h:222:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
1> ../src/main.c:31:12: note: in expansion of macro 'DEVICE_DT_GET'
1> zephyr/include/generated/devicetree_unfixed.h:1480:33: error: 'DT_N_S_pwmleds_S_pwm_led_0' undeclared (first use in this function); did you mean 'DT_N_S_pwmleds_S_pwm_led_0_ORD'?
1> C:/ncs/v1.5.1/zephyr/include/toolchain/common.h:124:31: note: in definition of macro '_DO_CONCAT'
1> C:/ncs/v1.5.1/zephyr/include/device.h:46:31: note: in expansion of macro '_CONCAT'
1> C:/ncs/v1.5.1/zephyr/include/device.h:205:37: note: in expansion of macro 'DEVICE_NAME_GET'
1> C:/ncs/v1.5.1/zephyr/include/device.h:222:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
1> ../src/main.c:31:12: note: in expansion of macro 'DEVICE_DT_GET'
1> C:/ncs/v1.5.1/zephyr/include/devicetree.h:2176:24: note: in expansion of macro 'DT_N_ALIAS_pwm_led0'
1> C:/ncs/v1.5.1/zephyr/include/devicetree.h:208:25: note: in expansion of macro 'DT_CAT'
1> ../src/main.c:23:18: note: in expansion of macro 'DT_ALIAS'
1> ../src/main.c:24:36: note: in expansion of macro 'PWM_NODE'
1> ../src/main.c:31:26: note: in expansion of macro 'PWM_CTLR_NODE'
1> zephyr/include/generated/devicetree_unfixed.h:1480:33: note: each undeclared identifier is reported only once for each function it appears in
1> C:/ncs/v1.5.1/zephyr/include/toolchain/common.h:124:31: note: in definition of macro '_DO_CONCAT'
1> C:/ncs/v1.5.1/zephyr/include/device.h:46:31: note: in expansion of macro '_CONCAT'
1> C:/ncs/v1.5.1/zephyr/include/device.h:205:37: note: in expansion of macro 'DEVICE_NAME_GET'
1> C:/ncs/v1.5.1/zephyr/include/device.h:222:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
1> ../src/main.c:31:12: note: in expansion of macro 'DEVICE_DT_GET'
1> C:/ncs/v1.5.1/zephyr/include/devicetree.h:2176:24: note: in expansion of macro 'DT_N_ALIAS_pwm_led0'
1> C:/ncs/v1.5.1/zephyr/include/devicetree.h:208:25: note: in expansion of macro 'DT_CAT'
1> ../src/main.c:23:18: note: in expansion of macro 'DT_ALIAS'
1> ../src/main.c:24:36: note: in expansion of macro 'PWM_NODE'
1> ../src/main.c:31:26: note: in expansion of macro 'PWM_CTLR_NODE'
```

mbolivar-nordic, 2021-05-04:
> 1> C:/ncs/v1.5.1/zephyr/include/device.h:46:39: warning: implicit declaration of function '__device_DT_PWMS_CTLR' [-Wimplicit-function-declaration]

This means the device doesn't exist, most likely because the driver is not enabled.

You probably need to add these to your prj.conf:

```
CONFIG_LED=y
CONFIG_LED_PWM=y
CONFIG_PWM=y
```

> I have the same problem when compiling for the nRF52840dk. The following code produces the same error:

With those enabled, that builds for me.

mbolivar-nordic, 2021-05-04:
Closing since these appear to be user errors

closed 2021-05-04 by mbolivar-nordic