zephyrproject-rtos/zephyr #34045: samples: subsys: mgmt: smp_srv: UDP sample does not build for frdm_k64f

aunsbjerg, 2021-04-06:
**Describe the bug**
When following the [instructions](https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html) on building the UDP sample for the frdm_k64f board, the board never boots up the actual application after initial flashing. 

I have also seen boot loops, but cannot consistently reproduce them.

The bug is seen on latest master (64995e2a54). I've tracked the breaking commit cc29dbde4f37923b7ec9f59febe309afac21fd30 which is part of #28519. Before that commit, the sample works as expected.

I have not yet looked into possible causes.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b frdm_k64f -d build_mcuboot bootloader/mcuboot/boot/zephyr
2. west flash -d build_mcuboot --erase -r jlink
3. west build -b frdm_k64f zephyr/samples/subsys/mgmt/mcumgr/smp_svr -- -DOVERLAY_CONFIG=overlay-udp.conf
4. west sign -t imgtool -- --key bootloader/mcuboot/root-rsa-2048.pem
5. west flash --bin-file build/zephyr/zephyr.signed.bin -r jlink
4. See error

**Logs and console output**

Log output for master (64995e2a54):

```
*** Booting Zephyr OS build zephyr-v2.5.0-1955-g64995e2a54a2  ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: primary slot
I: Swap type: none
I: Bootloader chainload address offset: 0x20000
I: Jumping to the first image slot
```

And log output for cc29dbde4f37923b7ec9f59febe309afac21fd30 for reference:

```
*** Booting Zephyr OS build zephyr-v2.4.0-1116-gcc29dbde4f37  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, i3
[00:00:00.006,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_o3
[00:00:00.006,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.008,000] <inf> mcuboot: Swap type: none
[00:00:00.000,000] <err> os: ***** BUS FAULT *****
[00:00:00.000,000] <err> os: r0/a1:  0x0002a009  r1/a2:  0x00000000  r2/a3:  0x4000d0b0
[00:00:00.000,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x0002d10d r14/lr:  0x000231f7
[00:00:00.000,000] <err> os:  xpsr:  0x61000000
[00:00:00.000,000] <err> os: Faulting instruction address (r15/pc): 0x00022904
[00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.000,000] <err> os: Current thread: 0x200031f0 (unknown)
[00:00:00.000,000] <err> os: Halting system
```

de-nordic, 2021-04-07:
@aunsbjerg Can you check if any other sample works with the same bootloader?
For example hello world?

de-nordic, 2021-04-07:
The problem isn't with the build as title suggest, because the app seems to build. It does not boot.

aunsbjerg, 2021-04-07:
> @aunsbjerg Can you check if any other sample works with the same bootloader?
> For example hello world?

I can, but not until later this week. I didn't do exhaustive testing yesterday because of time, but wanted to document the issue in any case. I'll update the issue when I know more.

aunsbjerg, 2021-04-09:
The hello world sample with `CONFIG_BOOTLOADER_MCUBOOT=y` works just fine on the `frdm_k64f` board. It boots up and runs the hello world code as expected.

The issue seem to be isolated to the SMP UDP sample. When `CONFIG_MCUMGR_SMP_UDP=y` is set AND `CONFIG_BOOTLOADER_MCUBOOT=y` is set at the same time, the application does not boot. Disable either of the options, and the applications boots.

de-nordic, 2021-04-21:
Similar problem reported here https://github.com/zephyrproject-rtos/zephyr/issues/34101#issuecomment-823706258

nandojve, 2021-04-28:
The #34680 fix this issue?

aunsbjerg, 2021-04-29:
> The #34680 fix this issue?

No, as far as I can tell, that fix doesn't make any difference.

Building mcuboot with `CONFIG_MCUBOOT_CLEANUP_ARM_CORE=y` does make the application image boot just fine though, so that's progress. I'm not sure if that setting should be added to the sample on a permanent basis or if the mentioned PR should make that unnecessary for cortex-m SoC's.

carlescufi, 2021-05-11:
Assigning to @MaureenHelm and @mmahadevan108 since this seems to be an NXP MPU issue.

henrikbrixandersen, 2021-05-11:
> I am not sure, maybe @ioannisg, @henrikbrixandersen, @nandojve or @mmahadevan108 can clarify whether, once #34680 is merged then `CONFIG_MCUBOOT_CLEANUP_ARM_CORE` will not be necessary anymore.

It should not be needed once that PR is merged.

aunsbjerg, 2021-05-11:
> > I am not sure, maybe @ioannisg, @henrikbrixandersen, @nandojve or @mmahadevan108 can clarify whether, once #34680 is merged then `CONFIG_MCUBOOT_CLEANUP_ARM_CORE` will not be necessary anymore.
> 
> It should not be needed once that PR is merged.

I can test #34680 in relation to this issue tomorrow when I'm in the office

aunsbjerg, 2021-05-12:
> I can test #34680 in relation to this issue tomorrow when I'm in the office

Can confirm that #34680 fixes this issue

PR #34680 (2021-05-26): arch: arm: cortex-m: add support for clearing NXP MPU regions at boot
Clear NXP MPU regions at boot if CONFIG_INIT_ARCH_HW_AT_BOOT is enabled. Also remove MPU support from the K2x SoC since this leads to compilation issues as the K22F does not contain an MPU.

Fixes: #34045
commit 2b0a4812 (2021-05-26): arch: arm: cortex-m: add support for clearing NXP MPU regions at boot

Clear NXP MPU regions at boot if CONFIG_INIT_ARCH_HW_AT_BOOT is
enabled.

Fixes: #34045
closed 2021-05-26 by galak