zephyrproject-rtos/zephyr #34046: Failed to build arm64 architecture related board

ngboonkhai, 2021-04-06:
**Describe the bug**
I was trying to build the zephyr on qemu_cortex_a53, bcm958402m2_a72, xenvm using command eq:$west build -b qemu_cortex_a53 samples/hello_world and it failed with error attached in the log section. it seems like environment issue, but i reinstall zephyr followed the getting started guide but no luck.

I have tried 
1)	Switch back to the changes https://github.com/zephyrproject-rtos/zephyr/commit/3539c2fbb35f3e85f2b6348b577fa3b3378a97b6 Minus 1 => Able to build

2)	Try to find on internet, what environment variable might possibly causing the issue, and play around with setting the variable => failed with the same error
		a.	ZEPHYR_BASE
		b.	ZEPHYR_TOOLCHAIN_VARIANT
		c.	ZEPHYR_SDK_INSTALL_DIR
		d.	CMAKE_C_COMPILER

3)	Updated everything according to the getting started guide => failed with the same errror

4)	Reinstall zephyr and follow every steps in getting started guide on another machine => failed with the same error

**To Reproduce**
Steps to reproduce the behavior:
$west build -b qemu_cortex_a53 samples/hello_world

**Expected behavior**
Build pass.

**Impact**
Showstopper

**Logs and console output**
-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/boonkhai/zephyrtesting/zephyrproject/zephyr/build -S/home/boonkhai/zephyrtesting/zephyrproject/zephyr/samples/hello_world -GNinja -DBOARD=qemu_cortex_a5
boonkhai@pg-centralnode-1:~/zephyrtesting/zephyrproject/zephyr$ west build -b qemu_cortex_a53 samples/hello_world                                              -- west build: generating a build system
Including boilerplate (Zephyr base (cached)): /home/boonkhai/zephyrtesting/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/boonkhai/zephyrtesting/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.5.99 (/home/boonkhai/zephyrtesting/zephyrproject/zephyr)
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: qemu_cortex_a53
-- Cache files will be written to: /home/boonkhai/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.12.3 (/usr/zephyr-sdk-0.12.3)
-- Found dtc: /usr/zephyr-sdk-0.12.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/boonkhai/zephyrtesting/zephyrproject/zephyr/boards/arm64/qemu_cortex_a53/qemu_cortex_a53.dts
-- Generated zephyr.dts: /home/boonkhai/zephyrtesting/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/boonkhai/zephyrtesting/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/boonkhai/zephyrtesting/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /home/boonkhai/zephyrtesting/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/boonkhai/zephyrtesting/zephyrproject/zephyr/boards/arm64/qemu_cortex_a53/qemu_cortex_a53_defconfig'
Merged configuration '/home/boonkhai/zephyrtesting/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/boonkhai/zephyrtesting/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/boonkhai/zephyrtesting/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
CMake Error at /home/boonkhai/zephyrtesting/zephyrproject/zephyr/cmake/compiler/gcc/target.cmake:10 (message):
  **C compiler /usr/zephyr-sdk-0.12.3//bin/-gcc not found - Please check your
  toolchain installation**
Call Stack (most recent call first):
  /home/boonkhai/zephyrtesting/zephyrproject/zephyr/cmake/target_toolchain.cmake:51 (include)
  /home/boonkhai/zephyrtesting/zephyrproject/zephyr/cmake/app/boilerplate.cmake:592 (include)
  /home/boonkhai/zephyrtesting/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/boonkhai/zephyrtesting/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/boonkhai/zephyrtesting/zephyrproject/zephyr/build -S/home/boonkhai/zephyrtesting/zephyrproject/zephyr/samples/hello_world -GNinja -DBOARD=qemu_cortex_a53

**Environment (please complete the following information):**
 - OS: (Ubuntu 20)
 - Toolchain Zephyr SDK
 - mainline (c4b35c1d36a77f71ecee1086118cff38b5813154)

galak, 2021-04-06:
Have you updated to SDK 0.12.4, you need that version for the recent arm/arm64 split.

carlocaione, 2021-04-06:
@galak is there a way to print a warning if the latest SDK is not installed?

carlescufi, 2021-04-06:
> @galak is there a way to print a warning if the latest SDK is not installed / used?

I don't think so, no. @tejlmand?

ngboonkhai, 2021-04-07:
> Have you updated to SDK 0.12.4, you need that version for the recent arm/arm64 split.

Nope, I didn't. But my teammates just got it debugged yesterday from our side, there is a cmake file need to be updated in the zephyr sdk (/usr/zephyr-sdk-0.11.4/cmake/zephyr/target.cmake)

I will try on 0.12.4 before closing this issue.
Thanks

ngboonkhai, 2021-04-07:
> @galak is there a way to print a warning if the latest SDK is not installed / used?

That will be great if there is a warning message.

The getting started guide must get updated as well, as for now, the getting started guide still asking user to install 0.12.3.

And for future, if there are any changes that require user to update to the latest zephyr (specific version) probably include that information in the commit message.

ngboonkhai, 2021-04-07:
> Have you updated to SDK 0.12.4, you need that version for the recent arm/arm64 split.
Hi galak,
Tested with the latest 0.12.14 sdk, it's working now, Thanks

tejlmand, 2021-04-07:
> > @galak is there a way to print a warning if the latest SDK is not installed / used?
> 
> I don't think so, no. @tejlmand?

We have:
https://github.com/zephyrproject-rtos/zephyr/blob/103fc5e43aacc5baa5b15e789fbe6f61c22d28dc/cmake/verify-toolchain.cmake#L19
which can be set to `0.12.4` when `arm64` is used.

but to make it fail if the toolchain is not new enough we would also need to add `REQUIRED` here:
https://github.com/zephyrproject-rtos/zephyr/blob/103fc5e43aacc5baa5b15e789fbe6f61c22d28dc/cmake/verify-toolchain.cmake#L70
which isn't currently the case as we still support the use of zephyr sdk 0.11.1 and 0.11.2.

PR #34064 (2021-05-26): cmake: arm64: Bail out if the old SDK is used
After the ARM/ARM64 split the SDK 0.12.4 is needed to be able to compile
the ARM64 arch code. Bail out if an old SDK is detected at compile time.

Closes: #34046
closed 2021-05-26 by galak