zephyrproject-rtos/zephyr #34049: Nordic nrf9160 switching between drivers and peripherals

e-devs, 2021-04-06:
Nordic nRF9160 has a limitation for a number of simultaneously enabled peripherals with Shared ID.

E.g. UARTE0 and SPIM0 cannot be enabled at the same time.

This limitation is applicable to UARTE, SPIM, SPIS, TWIM, and TWIS peripherals. Only totally 4 peripherals from the list above can be enabled simultaneously. This can achieved by changing of peripheral's ID value, e.g. UARTE0, UARTE1, SPIM2, SPIM3 is completely acceptable configuration.

Some applications may require different peripheral configurations depending on the functionality this application implements. E.g. an application may need 3 UART interfaces and 1 SPI, or 2 UART and 2 SPI, depending on the current state of the application.

nRF9160 Product specification has a recommendation on how-to switch between peripherals:

3.3.2 Peripherals with shared ID

In general (with the exception of ID 0), peripherals sharing an ID and base address may not be used
simultaneously. The user can only enable one peripheral at the time on this specific ID.
When switching between two peripherals sharing an ID, the user should do the following to prevent
unwanted behavior:
• Disable the previously used peripheral.
• Disable any publish/subscribe connection to the DPPI system for the peripheral that is being disabled.
• Clear all bits in the INTEN register, i.e. INTENCLR = 0xFFFFFFFF.
• Explicitly configure the peripheral that you are about to enable, and do not rely on configuration values
that may be inherited from the peripheral that was disabled.
• Enable the now configured peripheral.

This recommendation greatly works for bare-metal applications. The situation becomes more complex for Zephyr RTOS driven application. Zephyr has its own Device Drive model. Drivers are initialized at different initialization levels (PRE_KERNEL_1, PRE_KERNEL_2, etc.) at Zephyr startup. Driver connects to choosen peripheral and it looks impossible (or very complex) to switch between peripherals and their drivers. 

A driver of the peripheral to be switched should be stopped, its resources should be released and new driver should be initialized and connected to the peripheral. E.g. UARTE0 driver to be replaced with SPIM0 driver.

There is a possible solution for this problem: application can choose (based on its internal configuration) what drivers should be initialized at Zephyr start-up. In order to change set of active drivers - application may reboot nrf9160 SoC and provide list of drivers to be initiliazed via special memory section or SoC registers. This approach greatly simplifies switching between peripherals for some applications.

Current Zephyr build for nrf9160 procedure has to be changed to support this feature. Currently Peripheral Shared ID allocation is enforced during build and irq assignment procedure. Zephyr Driver initialization routine also has to be patched.]

Would you please your feedback on the proposed solution (pros and cons) and may be give any other clues on how to switch between peripherals and drivers?

ioannisg, 2021-04-07:
I believe @anangl could answer this?

e-devs, 2021-04-29:
Hello @anangl 

Did you have a chance to check my question?

closed 2021-07-27 by github-actions[bot]

zafersn, 2022-11-13:
Hi folks, is there any solution to this issue?