zephyrproject-rtos/zephyr #34068: DOC BUILD FAIL

hongshui3000, 2021-04-07:
**Describe the bug**
LOG:
```
E:zephyr\doc\_build\rst\doc\reference\bluetooth\gatt.rst:112: WARNING: Error in declarator
If declarator-id with parameters (e.g., 'void f(int arg)'):
  Invalid C declaration: Expected identifier in nested name. [error at 14]
    ATOMIC_DEFINE (flags, BT_GATT_SUBSCRIBE_NUM_FLAGS)
    --------------^
If parenthesis in noptr-declarator (e.g., 'void (*f(int arg))(double)'):
  Error in declarator or parameters
  Invalid C declaration: Expecting "(" in parameters. [error at 20]
    ATOMIC_DEFINE (flags, BT_GATT_SUBSCRIBE_NUM_FLAGS)
    --------------------^
```
FILE: zephyr\include\bluetooth\gatt.h
```
/** @brief GATT Subscribe parameters */
struct bt_gatt_subscribe_params {
	/** Notification value callback */
	bt_gatt_notify_func_t notify;
	/** Subscribe CCC write request response callback */
	bt_gatt_write_func_t write;
	/** Subscribe value handle */
	uint16_t value_handle;
	/** Subscribe CCC handle */
	uint16_t ccc_handle;
#if defined(CONFIG_BT_GATT_AUTO_DISCOVER_CCC)
	/** Subscribe End handle (for automatic discovery) */
	uint16_t end_handle;
	/** Discover parameters used when ccc_handle = 0 */
	struct bt_gatt_discover_params *disc_params;
#endif /* CONFIG_BT_GATT_AUTO_DISCOVER_CCC */
	/** Subscribe value */
	uint16_t value;
	/** Subscription flags */
	ATOMIC_DEFINE(flags, BT_GATT_SUBSCRIBE_NUM_FLAGS); -------------------------Here

	sys_snode_t node;
};
```

gmarull, 2021-04-07:
`ATOMIC_DEFINE(flags, BT_GATT_SUBSCRIBE_NUM_FLAGS);` is a known issue, and it is actually filtered out, has no relation with the reported problem. Looks like you are running out of date dependencies, make sure to run first `pip install -U scripts/requirements-doc.txt`. If the problem persists, you may try to set `SPHINXOPTS=` to see if default parallelization is the culprit.

hongshui3000, 2021-04-08:
@gmarull 
Thanks
It can work after updating the version.

hongshui3000, 2021-04-08:
This issue has been repeated with another issue   close it

closed 2021-04-08 by hongshui3000