zephyrproject-rtos/zephyr #34076: Unrecognized characters generated during document construction

hongshui3000, 2021-04-07:
**Describe the bug**
```
W眉rth Elektronik GmbH. (we)
===========================

.. rst-class:: rst-columns

- :ref:`dtbinding_we_wsen_itds`

.. _dt_vendor_xlnx:
```
file:zephyr/dts/bindings/vendor-prefixes.txt
```
Würth Elektronik GmbH
```
Open this file in my system, the text is correct

**Environment (please complete the following information):**
 -  Windows(Chinese)

hongshui3000, 2021-04-08:
Similarly, I can’t write Chinese characters in it, and I will report an error directly.
zephyr/dts/bindings/vendor-prefixes.txt
LOG:
```
FAILED: CMakeFiles/devicetree
cmd.exe /C "cd /D E:\work\zephyr\doc && "C:\Program Files\CMake\bin\cmake.exe" -E env PYTHONPATH=E:/work/zephyr/scripts/dts/python-devicetree/src; ZEPHYR_BASE=E:/work/ZC3827/ZCAN-IOT/zephyr GEN_DEVICETREE_REST_ZEPHYR_DOCSET= C:/Python39/python.exe E:/work/ZC3827/ZCAN-IOT/zephyr/doc/_scripts/gen_devicetree_rest.py --vendor-prefixes E:/work/ZC3827/ZCAN-IOT/zephyr/dts/bindings/vendor-prefixes.txt --dts-root E:/work/ZC3827/ZCAN-IOT/zephyr E:/work/ZC3827/ZCAN-IOT/zephyr/doc/_build/rst/doc/reference/devicetree"
ninja: build stopped: subcommand failed.
```

mbolivar-nordic, 2021-05-11:
That's weird. The issue doesn't reproduce on Linux. At first I thought it was just a missing PYTHONIOENCODING environment variable setting, but I see that doc/CMakeLists.txt includes cmake/python.cmake, which sets PYTHONIOENCODING to "utf-8" on Windows.

@gmarull any idea?

mbolivar-nordic, 2021-05-11:
@hongshui3000 could you please apply this patch and retest, to see if the change to vendor-prefixes.txt is working?

```patch
diff --git a/doc/CMakeLists.txt b/doc/CMakeLists.txt
index 3cc9928381..562ad52098 100644
--- a/doc/CMakeLists.txt
+++ b/doc/CMakeLists.txt
@@ -111,6 +111,7 @@ configure_file(${DOXYFILE_IN} ${DOXYFILE_OUT} @ONLY)
 set(EXTRACT_CONTENT_COMMAND
   ${CMAKE_COMMAND} -E env
   ZEPHYR_BASE=${ZEPHYR_BASE}
+  PYTHONIOENCODING=utf-8
   ${PYTHON_EXECUTABLE} _scripts/extract_content.py
   # Ignore any files in the output directory.
   --ignore ${CMAKE_CURRENT_BINARY_DIR}
```

gmarull, 2021-05-11:
We may need to explicitly open file with encoding, ie `open("file", "r", encoding="utf-8")`. I faced similar encoding problems on Windows and solved them this way (see https://github.com/nrfconnect/sdk-nrf/pull/4449/files#diff-87607595cbd4c2220ec7e0a03979db859c93fc90c6e060b326f406afd08ebb45R68)

mbolivar-nordic, 2021-05-12:
> It looks like there is an `-X utf8` we can use instead:
> 
> [](https://docs.python.org/3/library/locale.html#locale.getpreferredencoding)

@hongshui3000 could you please test with this patch, which uses that option: https://github.com/zephyrproject-rtos/zephyr/pull/35234

hongshui3000, 2021-05-13:
> > It looks like there is an `-X utf8` we can use instead:
> 
> @hongshui3000 could you please test with this patch, which uses that option: #35234

hongshui3000, 2021-05-13:
> > It looks like there is an `-X utf8` we can use instead:
> 
> @hongshui3000 could you please test with this patch, which uses that option: #35234

[image]
it works

mbolivar-nordic, 2021-05-13:
Thank you very much for testing!

mbolivar-nordic, 2021-05-18:
I've removed my assignment since I've posted a fix that @gmarull rejected. Will leave it up to him to decide on a proper fix.

PR #35470 (2021-05-20): doc: scripts: gen_devicetree_rest: use utf-8 encoding
Use utf-8 encoding when opening vendor prefix files. Note that utf-8 is enforced in other scripts, e.g. gen_kconfig_rest, gen_helpers, etc.

Fixes #34076

Alternative to #35234
closed 2021-05-20 by galak