zephyrproject-rtos/zephyr #34102: Convert drivers/serial/uart_sam0.c to new kwork API

galak, 2021-04-07:
Convert drivers/serial/uart_sam0.c to use new kwork API.

See the following issue for details:

#33104

galak, 2021-04-07:
@nandojve @mnkp can one of you take a look at this?

galak, 2021-04-13:
@nandojve @mnkp ping.

mnkp, 2021-04-13:
I don't have SAM0 hardware to test it, SAM and SAM0 families are quite different. I looked at the `uart_sam0.c` driver code however and it seems like it should be one of those easy cases where the change could be done with #33924 from @pabigot.

nandojve, 2021-04-20:
I can take a look and test on SAMR21.

galak, 2021-04-20:
> I can take a look and test on SAMR21.

thanks

nandojve, 2021-04-21:
@galak, @mnkp 

There are building problems with DT when using async for sam0.
Can someone help identify what is missing?

To reproduce can be (at master) any one from above, don't need board:

```
west build -b atsamr21_xpro tests/drivers/uart/uart_async_api
west build -b atsamr21_xpro tests/drivers/spi/spi_loopback/
```

galak, 2021-04-21:
@nandojve if I remember you need to disable all uarts except the one with dma as when you enable async uart its assumed to be supported on ALL uarts.

nandojve, 2021-04-23:
The sam0 DMA broke after 7967aa6af0. It was just before the mass const-ify on Zephyr. My guess is that some structure was supposed to not be const because `device_get_binding` returns NULL and then a hard fault happen.

My proposal is convert uart to use new kwork API but driver will continue with a bug. It may take long to understand what is going on with DMA driver.

Is it reasonable?

galak, 2021-04-24:
@nandojve sounds good

PR #34549 (2021-04-26): serial: sam0: Conversion of k_work API
Replace all existing deprecated API with the recommended alternative.

Fixes #34102
commit d364062b (2021-04-26): serial: sam0: Conversion of k_work API

Replace all existing deprecated API with the recommended alternative.

Fixes #34102
closed 2021-04-26 by nashif