zephyrproject-rtos/zephyr #34116: mec15xxevb_assy6853: tests/kernel/mutex/sys_mutex/

yerabolu, 2021-04-07:
**Describe the bug**
test_mutex_multithread_competition in tests/kernel/mutex/sys_mutex/ failed

**To Reproduce**
Steps to reproduce the behavior:
1.  twister --hardware-map ~/mec1.file --device-testing -s tests/kernel/mutex/sys_mutex/test_mutex_multithread_competition/
2. See error

**Expected behavior**
Test should pass

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.5.0-2011-gecac194448da  ***
Running test suite mutex_complex
===================================================================
START - test_mutex
START - Test kernel Mutex API
===================================================================
Done LOCKING!  Current priority = 5
*** Booting Zephyr OS build zephyr-v2.5.0-2011-gecac194448da  ***
Running test suite mutex_complex
===================================================================
START - test_mutex
START - Test kernel Mutex API
===================================================================
Done LOCKING!  Current priority = 5
Testing recursive locking
Recursive locking tests successful
PASS - test_mutex in 7.519 seconds
===================================================================
START - test_user_access
SKIP - test_user_access in 0.1 seconds
===================================================================
START - test_supervisor_access
PASS - test_supervisor_access in 0.1 seconds
===================================================================
START - test_mutex_multithread_competition
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mutex/sys_mutex/src/thread_competition.c:119: test_mutex_multithread_competition: (flag == LOW_PRO is false)
The low priority thread failed to take the mutex.
FAIL - test_mutex_multithread_competition in 0.75 seconds
===================================================================
Test suite mutex_complex failed.
===================================================================
PROJECT EXECUTION FAILED 
```

**Environment (please complete the following information):**
 - OS: Fedora 
 - Toolchain: zephyr-sdk-0.12.3
 - Commit ID: zephyr-v2.5.0-2011-gecac194448da

nashif, 2021-04-08:
This is a new test that was introduced recently in 7e91223ae0e, please verify if the issue is in the test as the first thing.

PerMac, 2021-04-08:
I see failures in the same place on all Nordic boards we deliver results for, Ubuntu 20.04, zephyr-sdk 0.12.2
Hovewer, I see `The highest priority thread failed to take the mutex.` on most of them

Zhaoningx, 2021-04-08:
Hi, yes, there are some platforms failed to run this case, I'm working on it now. And this case's PR is : https://github.com/zephyrproject-rtos/zephyr/pull/33492 , and I designed this case based on the zephyr doc:
[image]

KangJianX, 2021-04-09:
Same behavior can be reproduce on ADSP.

PR #34157 (2021-04-12): tests: mutex: fixed a testcase assert failure
Modify the testcase design to solve some threads
can't lock mutex. Using array index to detect the order
of threads getting mutex instead of delaying.

Fixed #34116
commit 05e5f484 (2021-04-12): tests: mutex: fixed a testcase assert failure

Modify the testcase design to solve some threads
can't lock mutex. Using array index to detect the order
of threads getting mutex instead of delaying.

Fixed #34116
closed 2021-04-12 by nashif