zephyrproject-rtos/zephyr #34117: ehl_crb: tests/kernel/context tests failed

yerabolu, 2021-04-07:
**Describe the bug**
kernel_cpu_idle_atomic and kernel_cpu_idle tests failed

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W -p ehl_crb --device-testing --device-serial /dev/ttyUSB0 -T tests/kernel/context/ --west flash="/home/user/EHL_X86_PXE.sh" 
2. Test results

**Expected behavior**
Tests should pass

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v2.5.99-ww13.4  ***
Running test suite context
===================================================================
START - test_kernel_interrupts
PASS - test_kernel_interrupts in 0.1 seconds
===================================================================
START - test_kernel_ctx_thread
Testing k_current_get() from an ISR and thread
Testing k_is_in_isr() from an ISR
Testing k_is_in_isr() from a preemptible thread
PASS - test_kernel_ctx_thread in 0.13 seconds
===================================================================
START - test_busy_wait
Thread busy waiting for 20000 usecs
Thread busy waiting completed
Thread busy waiting for 20000 usecs (irqs locked)
Thread busy waiting completed (irqs locked)
PASS - test_busy_wait in 0.56 seconds
===================================================================
START - test_k_sleep
thread sleeping for 50 milliseconds
thread back from sleep
Testing k_thread_create() without cancellation
thread (q order: 2, t/o: 500) is running
got thread (q order: 2, t/o: 500) as expected
thread (q order: 3, t/o: 750) is running
got thread (q order: 3, t/o: 750) as expected
thread (q order: 0, t/o: 1000) is running
got thread (q order: 0, t/o: 1000) as expected
thread (q order: 6, t/o: 1250) is running
got thread (q order: 6, t/o: 1250) as expected
thread (q order: 1, t/o: 1500) is running
got thread (q order: 1, t/o: 1500) as expected
thread (q order: 4, t/o: 1750) is running
got thread (q order: 4, t/o: 1750) as expected
thread (q order: 5, t/o: 2000) is running
got thread (q order: 5, t/o: 2000) as expected
Testing k_thread_create() with cancellations
cancelling [q order: 0, t/o: 1000, t/o order: 0]
thread (q order: 3, t/o: 750) is running
got (q order: 3, t/o: 750, t/o order 1) as expected
thread (q order: 0, t/o: 1000) is running
got (q order: 0, t/o: 1000, t/o order 2) as expected
cancelling [q order: 3, t/o: 750, t/o order: 3]
cancelling [q order: 4, t/o: 1750, t/o order: 4]
thread (q order: 4, t/o: 1750) is running
got (q order: 4, t/o: 1750, t/o order 5) as expected
cancelling [q order: 6, t/o: 1250, t/o order: 6]
PASS - test_k_sleep in 5.340 seconds
===================================================================
START - test_kernel_cpu_idle_atomic
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/context/src/main.c:350: _test_kernel_cpu_idle: (tms2 < tms is true)
Bad ms value computed,got 5482 which is less than 5483
FAIL - test_kernel_cpu_idle_atomic in 0.20 seconds
===================================================================
START - test_kernel_cpu_idle
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/context/src/main.c:350: _test_kernel_cpu_idle: (tms2 < tms is true)
Bad ms value computed,got 5516 which is less than 5517
FAIL - test_kernel_cpu_idle in 0.20 seconds
===================================================================
START - test_k_yield
PASS - test_k_yield in 0.1 seconds
===================================================================
START - test_kernel_thread
PASS - test_kernel_thread in 0.1 seconds
===================================================================
START - test_kernel_timer_interrupts
PASS - test_kernel_timer_interrupts in 0.4 seconds
===================================================================
Test suite context failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
- OS: Fedora33
- Toolchain: zephyr-sdk-0.12.3
 - Commit ID: ecac194448da

jenmwms, 2021-04-09:
Ok, I see this is one of those tests sensitive to timing (off by 1ms). Let's see what we find in debug. Thanks!

jenmwms, 2021-04-29:
FWIW This/similar issue was recently resolved with patch 148769c "sched: timeout: Do not miss slice timeouts".  I can also use that as a last-known-working for git bisect to find where/why this broke again.

jenmwms, 2021-04-29:
Perhaps this bug is not reproduceable on all EHL platforms? I'll keep trying on a different board next, because first device passed. Then debug once it's isolated.

yerabolu, 2021-04-29:
@jenmwms : I retested this on EHL_CRB now and tests are passing. So closing this issue.

closed 2021-04-29 by yerabolu

jenmwms, 2021-04-29:
Thank you @yerabolu !