zephyrproject-rtos/zephyr #34121: PDF document generation is not supported under windows

hongshui3000, 2021-04-08:
**Describe the bug**
From the documentation, it supports pdf generation, on the windows system. But it does not actually support it.
```
ninja: error: unknown target 'pdfdocs'
```
[image]

hongshui3000, 2021-04-08:
When I replaced this picture with a jpg picture, another problem appeared
LOG:
```
FAILED: latex/zephyr.pdf 
cmd.exe /C "cd /D E:\work\zephyr\doc\_build\latex && "C:\Program Files\CMake\bin\cmake.exe" -E env LATEXOPTS="-halt-on-error -no-shell-escape" "C:/Program Files/MiKTeX/miktex/bin/x64/latexmk.exe" -quiet -pdf -dvi- -ps-"
Latexmk: Run number 1 of rule 'pdflatex'
This is pdfTeX, Version 3.141592653-2.6-1.40.22 (MiKTeX 21.3)
entering extended mode
pdflatex: major issue: So far, no MiKTeX administrator has checked for updates.
Latexmk: Summary of warnings from last run of *latex:
  Latex failed to resolve 45 citation(s)
Collected error summary (may duplicate other messages):
  pdflatex: Command for 'pdflatex' gave return code 1
      Refer to 'zephyr.log' for details
Latexmk: Use the -f option to force complete processing,
 unless error was exceeding maximum runs, or warnings treated as errors.
latexmk: major issue: So far, no MiKTeX administrator has checked for updates.
ninja: build stopped: subcommand failed.
```
zephyr.log
```
[2510] [2511]

LaTeX Warning: Hyper reference `boards/arm/nrf52dk_nrf52832/doc/index:nrf52dk-n
rf52832' on page 2512 undefined on input line 288843.

[2512] [2513]
Overfull \hbox (24.48854pt too wide) in paragraph at lines 1--288952
[][][]$\T1/ptm/m/n/10 https : / / eu . industrial . panasonic . com / products 
/ sensors-[]optical-[]devices / sensors-[]automotive-[]and-[]industrial-[]appli
cations /
 []

[2514]

LaTeX Warning: Hyper reference `reference/peripherals/sensor:sensor-api' on pag
e 2515 undefined on input line 289028.

! Illegal parameter number in definition of \Hy@tempa.
<to be read again> 
                   1
l.289029 ...me280/\textasciigrave{}}{Bosch BME280}
                                                   environmental sensor.
 
Here is how much of TeX's memory you used:
 35832 strings out of 479321
 607628 string characters out of 2866508
 990018 words of memory out of 3000000
 33283 multiletter control sequences out of 15000+600000
 477387 words of font info for 128 fonts, out of 8000000 for 9000
 1142 hyphenation exceptions out of 8191
 83i,16n,88p,1409b,751s stack positions out of 5000i,500n,10000p,200000b,80000s

!  ==> Fatal error occurred, no output PDF file produced!
```

hongshui3000, 2021-04-08:
When I deleted the points, another problem appeared
zephyr.log
```

[2804]

LaTeX Warning: Hyper reference `samples/hello_world/README:hello-world' on page
 2805 undefined on input line 312210.

<blackpill-v3.jpg, id=32778, 3854.4pt x 3854.4pt>
File: blackpill-v3.jpg Graphic file (type jpg)
<use blackpill-v3.jpg>
Package pdftex.def Info: blackpill-v3.jpg  used on input line 312242.
(pdftex.def)             Requested size: 469.7421pt x 469.7421pt.
[2805] [2806 <./blackpill-v3.jpg>]
<Blackpill_Pinout.png, id=32796, 1175.14032pt x 886.81313pt>
File: Blackpill_Pinout.png Graphic file (type png)
<use Blackpill_Pinout.png>
Package pdftex.def Info: Blackpill_Pinout.png  used on input line 312411.
(pdftex.def)             Requested size: 469.74297pt x 354.50246pt.
[2807] [2808 <./Blackpill_Pinout.png>] [2809]
<blackpill-v2.jpg, id=32842, 3854.4pt x 3854.4pt>
File: blackpill-v2.jpg Graphic file (type jpg)
<use blackpill-v2.jpg>
Package pdftex.def Info: blackpill-v2.jpg  used on input line 312538.
(pdftex.def)             Requested size: 469.7421pt x 469.7421pt.
[2810 <./blackpill-v2.jpg>] [2811]
<Blackpill_Pinout1.png, id=32861, 1175.14032pt x 886.81313pt>
File: Blackpill_Pinout1.png Graphic file (type png)
<use Blackpill_Pinout1.png>
Package pdftex.def Info: Blackpill_Pinout1.png  used on input line 312707.
(pdftex.def)             Requested size: 469.74297pt x 354.50246pt.
[2812 <./Blackpill_Pinout1.png>] [2813]
!pdfTeX error: pdflatex (file ./bmd_345_eval.png): libpng: internal error
 ==> Fatal error occurred, no output PDF file produced!
```

hongshui3000, 2021-04-08:
When I replaced it with the correct picture, a new problem appeared.
It seems that the utf-8 character "μ" cannot be recognized

LOG:
```
Latexmk: Run number 1 of rule 'pdflatex'
This is pdfTeX, Version 3.141592653-2.6-1.40.22 (MiKTeX 21.3)
entering extended mode
pdflatex: major issue: So far, no MiKTeX administrator has checked for updates.
Latexmk: Summary of warnings from last run of *latex:
  =====Latex reported missing or unavailable character(s).
=====See log file for details.
  Latex failed to resolve 45 citation(s)
Collected error summary (may duplicate other messages):
  pdflatex: Command for 'pdflatex' gave return code 1
      Refer to 'zephyr.log' for details
Latexmk: Use the -f option to force complete processing,
 unless error was exceeding maximum runs, or warnings treated as errors.
latexmk: major issue: So far, no MiKTeX administrator has checked for updates.
ninja: build stopped: subcommand failed.
```
zephyr.log
```
[2928]

! Package inputenc Error: Unicode character μ (U+03BC)
(inputenc)                not set up for use with LaTeX.

See the inputenc package documentation for explanation.
Type  H <return>  for immediate help.
 ...                                              
                                                  
l.324434 using Servo V2, μ
                           Servo, or Servo V4 (CCD). See the 
Here is how much of TeX's memory you used:
 57302 strings out of 479321
 1787612 string characters out of 2866508
 1861978 words of memory out of 3000000
 52158 multiletter control sequences out of 15000+600000
 478012 words of font info for 130 fonts, out of 8000000 for 9000
 1142 hyphenation exceptions out of 8191
 83i,16n,88p,1409b,949s stack positions out of 5000i,500n,10000p,200000b,80000s

!  ==> Fatal error occurred, no output PDF file produced!
```

hongshui3000, 2021-04-08:
Same problem as above. The picture “lpcxpresso11u68.png” is also unrecognizable

hongshui3000, 2021-04-09:
summary:
1.The steps to generate the document in the document are incomplete
2.Picture problem：ReelBoard-Blinky.gif bmd_345_eval.png lpcxpresso11u68.png mimx8mm_evk.jpg stm32f103_mini_yellow.jpg
3.Character problem:   ？ （  ） μ
4.Extra character problem :  https://www.bosch-sensortec.com/products/environmental-sensors/humidity-sensors-bme280/`

gmarull, 2021-04-30:
I think PDF documentation is currently not well maintained, so I'd not be surprised if it is broken or the output is not properly structured. I think that the addition of Kconfig/DTS bindings made things worse, the number of pages has likely exploded. I'd personally remove the PDF option since it adds no value as of today. Opinions @carlescufi @nashif ?

hongshui3000, 2021-05-01:
For a user who is trying to use zephyrrtos in a real product, it is useful to be able to generate pdf. This is why I am trying to generate the pdf

carlescufi, 2021-05-06:
> I think PDF documentation is currently not well maintained, so I'd not be surprised if it is broken or the output is not properly structured. I think that the addition of Kconfig/DTS bindings made [...]

The .pdf was added in part due to archival purposes related to Safety certification, so I think we should keep it even if it is not complete. It was good enough that it could be considered usable.

hongshui3000, 2021-05-13:
> The .pdf was added in part due to archival purposes related to Safety certification, so I think we should keep it even if it is not complete. It was good enough that it could be considered usable.

Is there a plan to clean up the characters and pictures of the document? This is not only for generating pdf?

gmarull, 2021-05-21:
I've tried building removing gifs (and running in fast mode, ie no Kconfig), but build still fails:

```
41776 If you really absolutely need more capacity,                                                                                                            
41777 you can ask a wizard to enlarge me.                                                                                                                     
41778                                                                                                                                                         
41779                                                                                                                                                         
41780 Here is how much of TeX's memory you used:                                                                                                              
41781  38350 strings out of 481239                                                                                                                            
41782  653456 string characters out of 5920376                                                                                                                
41783  2212319 words of memory out of 5000000                                                                                                                 
41784  31874 multiletter control sequences out of 15000+600000                                                                                                
41785  605682 words of font info for 123 fonts, out of 8000000 for 9000                                                                                       
41786  1142 hyphenation exceptions out of 8191                                                                                                                
41787  5000i,16n,5003p,1407b,756s stack positions out of 5000i,500n,10000p,200000b,80000s                                                                     
41788 !  ==> Fatal error occurred, no output PDF file produced!  
```

It looks like we need to [increase latex capacity](https://tex.stackexchange.com/questions/26205/increase-latex-capacity).

As mentioned earlier, I think it doesn't make sense to support PDF due to the size of the current docs. The content is not optimized for PDF viewing either.

gmarull, 2021-07-15:
Documentation PDF build is now back to operational state, so closing. Thanks @nashif!

closed 2021-07-15 by gmarull