zephyrproject-rtos/zephyr #34131: TFTP client ignores incoming data packets

Delsian, 2021-04-08:
TFTP client implementation (/subsus/net/lib/tftp/tftp_client.c) doesn't work with tftpd ver.0.17-22
After creating socket (line 235) it sends request to server port 69 (line 196). Server responds with data packet, but the data transfer ports are chosen independently by the sender and receiver during the transfer initialization.
As a result, /subsus/net/ip/connection.c drops incoming data packet from server as it has source port differ than expected port 69 (checking at line 695):
```
[00:01:23.478,000] <dbg> net_conn.conn_register_debug: (sysworkq): [0x20004fcc/17/2/0x3f] remote fe80::5c00:82ff:fe96:9800/69
[00:01:23.485,000] <dbg> net_conn.conn_register_debug: (sysworkq):   local ::/38166 cb 0x803cd91 ud (nil)
[00:01:23.599,000] <dbg> net_conn.net_conn_input: (rx_q[0]): Check UDP listener for pkt 0x2001497c src port 36610 dst port 38166 family 2
[00:01:23.607,000] <dbg> net_conn.net_conn_input: (rx_q[0]): No match found.
[00:01:25.995,000] <dbg> net_conn.net_conn_unregister: (sysworkq): Connection handler 0x20004fcc removed
[00:01:26.001,000] <dbg> net_conn.conn_register_debug: (sysworkq): [0x20004fcc/17/2/0x3f] remote fe80::5c00:82ff:fe96:9800/69
[00:01:26.008,000] <dbg> net_conn.conn_register_debug: (sysworkq):   local ::/38166 cb 0x803cd91 ud (nil)
[00:01:26.110,000] <dbg> net_conn.net_conn_input: (rx_q[0]): Check UDP listener for pkt 0x2001497c src port 45641 dst port 38166 family 2
[00:01:26.118,000] <dbg> net_conn.net_conn_input: (rx_q[0]): No match found.
[00:01:28.518,000] <dbg> net_conn.net_conn_unregister: (sysworkq): Connection handler 0x20004fcc removed
[00:01:28.524,000] <dbg> net_conn.conn_register_debug: (sysworkq): [0x20004fcc/17/2/0x3f] remote fe80::5c00:82ff:fe96:9800/69
[00:01:28.531,000] <dbg> net_conn.conn_register_debug: (sysworkq):   local ::/38166 cb 0x803cd91 ud (nil)
[00:01:28.599,000] <dbg> net_conn.net_conn_input: (rx_q[0]): Check UDP listener for pkt 0x2001497c src port 36610 dst port 38166 family 2
[00:01:28.606,000] <dbg> net_conn.net_conn_input: (rx_q[0]): No match found.
```

https://snipboard.io/GhUDEF.jpg

**Temporary workaround:**

Comment out 'continue' on line 698 file connection.c, it disables source port checking; use recvfrom() and sendto() to keep server's port.

Delsian, 2021-04-08:
Is it possible to use 'setsockopt' to modify socket behaviour in part of source port checking?
Or any suggestion, how to change conn->remote_addr->sin_port value to 0 after sending the first packet?
Or better to use 'bind' to random incoming port instead 'connect' to port 69?

bwasim, 2021-04-08:
The TFTP client polls on the socket (see https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/tftp/tftp_client.c#L108), And so any data received on this socket should make it to the client even if its sent by a different source port. So the client is doing the correct thing.

I'm not aware of the latest changes within the networking stack which may have caused this. @jukkar would know if its a networking stack issue.

jukkar, 2021-04-09:
Noticed following issues with tftp:

* There is no sample app for the tftp functionality. Dunno how this went unnoticed but this needs to be fixed, every feature needs to have a sample so that it is easy to test things and also make sure the code does not bitrot. @bwasim would you be able to send a PR that adds simple tftp client app? The app does not need to implement anything useful, just to make sure it calls relevant bits and pieces that are needed for the tftp.
* Noticed resource leak in `tftp_get()`, the sock is not closed in the case of error or timeout. Also tftp_lock is not relased in the case of error. We should goto to req_done: label in the case of errors (except the socket() failing)
* I setup a tftp_server in Linux and tried this from Zephyr side (after adding `tftp_get()` call to existing sample). In Zephyr side the socket is created like this which creates this listener in `connection.c`
```
[ 6] 0x808dff8  2      4DU      0.0.0.0:37362       192.0.2.2:69
```
So in Zephyr side we are only accepting data from `192.0.2.2:69` but now the server is sending data from some random port (instead of 69). Then zephyr just drops the packet because there is no listener for that external port. After quick browse of the code, Zephyr should be able to accept the tftp packet if the port is set to 0 in `connect()` and then set the destination port when using `sendto()`. Atm the tftp lib is using `send()` so the code would need some tweaking. I did not test this but it looks like it might work.

Delsian, 2021-04-12:
@jukkar Approve. My code works fine with connection port set to 0. Thanks.
Additionally, it's good to have streaming mode for TFTP client (for example, to write data to Flash)

bwasim, 2021-04-12:
I will do a pull request firstly to fix the issues described in this issue, and we can think about enhancements later.

@jukkar , I was supposed to send out a pull request for the app, but I moved away from this project fairly quickly and forgot to do so. Will also do that with this issue.

PR #36579 (2021-06-29): net: refactor tftp library
- bugfix: Accept initial TFTP server reply from a port different than
  the one used to establish the connection (typically 69) as mandated
  by RFC 1350. Previous implementation was not standard compliant.
- bugfix: close socket in case of error or timeout.
- bugfix: Reset retransmit counter after receipt of a good packet.
- bugfix: Use CONFIG_TFTP_LOG_LEVEL to set log level.
- api: upon successful receipt of the file set `client.user_buf_size`
  to the size of the file received.
- Restructure the code, comments.
- Limit usage of global variables.
- Limit usage of ` [... cut]
closed 2021-06-29 by nashif