zephyrproject-rtos/zephyr #34134: USB do not works if bootloader badly use the device before

wysman, 2021-04-08:
**Describe the bug**
From the version 2.5 of zephyr-os, the USB-ACM of mcuboot is not working anymore, but the chain loading of the application slot is working. My application is using USB too. I think the previous failed init of USB from MCUBoot prevent the zephyr-os usb init to works. 

**Work around**
- Use mcuboot provided and compiled for Zephyr-os 2.4
- Use Application compiled  for Zephyr-os 2.5.

**Related issue (mcuboot)**
https://github.com/mcu-tools/mcuboot/issues/972

**To Reproduce**
Get a STM3F4 dev board
Use MCUBoot as bootloader with USB-ACM enabled.
Use `samples/subsys/usb/hid` as application

**Expected behavior**
Zephyr must do a more complete cleanup of the USB device to ensure a correct working state.

**Impact**
We can not use USB in the zephyr-os based application.

**Logs and console output (Board)**
usb_enable() works fine, no error

**Logs and console output (Linux host)**
avril 08 13:33:40 cortex kernel: usb 1-3: new full-speed USB device number 90 using xhci_hcd
avril 08 13:33:40 cortex kernel: usb 1-3: device descriptor read/64, error -71
avril 08 13:33:40 cortex kernel: usb 1-3: device descriptor read/64, error -71
avril 08 13:33:40 cortex kernel: usb 1-3: new full-speed USB device number 91 using xhci_hcd
avril 08 13:33:41 cortex kernel: usb 1-3: device descriptor read/64, error -71
avril 08 13:33:41 cortex kernel: usb 1-3: device descriptor read/64, error -71
avril 08 13:33:41 cortex kernel: usb usb1-port3: attempt power cycle
avril 08 13:33:42 cortex kernel: usb 1-3: new full-speed USB device number 92 using xhci_hcd
avril 08 13:33:42 cortex kernel: usb 1-3: Device not responding to setup address.
avril 08 13:33:42 cortex kernel: usb 1-3: Device not responding to setup address.
avril 08 13:33:42 cortex kernel: usb 1-3: device not accepting address 92, error -71
avril 08 13:33:42 cortex kernel: usb 1-3: new full-speed USB device number 93 using xhci_hcd
avril 08 13:33:42 cortex kernel: usb 1-3: Device not responding to setup address.
avril 08 13:33:42 cortex kernel: usb 1-3: Device not responding to setup address.
avril 08 13:33:43 cortex kernel: usb 1-3: device not accepting address 93, error -71
avril 08 13:33:43 cortex kernel: usb usb1-port3: unable to enumerate USB device

**Environment (please complete the following information):**
 - OS: Linux
 - SDK: v0.12.4
 - Zephyr-os: v2.5.0

carlescufi, 2021-04-08:
Are you enabling `CONFIG_MCUBOOT_CLEANUP_ARM_CORE` in MCUboot?

jfischer-no, 2021-04-08:
> USB-ACM of mcuboot is not working anymore

I guess it is Zephyr OS with USB device stack, CDC-ACM class support and MCUboot as application?
Can you please test samples/subsys/usb/cdc_acm without MCUboot and any modifications?

wysman, 2021-04-08:
@carlescufi I have made a test with the `CONFIG_MCUBOOT_CLEANUP_ARM_CORE`. It's do not works better.

wysman, 2021-04-08:
@jfischer-no I have test the `samples/subsys/usb/cdc_acm` application without mcuboot. It's do not works.

**Logs and console output (Board)**
Booting Zephyr OS build zephyr-v2.5.0
[00:00:00.054,000] <inf> cdc_acm_echo: Wait for DTR

**Logs and console output (Linux host)**
avril 08 17:35:07 cortex kernel: usb 1-2.1: new full-speed USB device number 76 using xhci_hcd
avril 08 17:35:07 cortex kernel: usb 1-2.1: device descriptor read/64, error -32
avril 08 17:35:07 cortex kernel: usb 1-2.1: device descriptor read/64, error -32
avril 08 17:35:07 cortex kernel: usb 1-2.1: new full-speed USB device number 77 using xhci_hcd
avril 08 17:35:07 cortex kernel: usb 1-2.1: device descriptor read/64, error -32
avril 08 17:35:07 cortex kernel: usb 1-2.1: device descriptor read/64, error -32
avril 08 17:35:08 cortex kernel: usb 1-2-port1: attempt power cycle
avril 08 17:35:08 cortex kernel: usb 1-2.1: new full-speed USB device number 78 using xhci_hcd
avril 08 17:35:08 cortex kernel: usb 1-2.1: Device not responding to setup address.
avril 08 17:35:08 cortex kernel: usb 1-2.1: Device not responding to setup address.
avril 08 17:35:09 cortex kernel: usb 1-2.1: device not accepting address 78, error -71
avril 08 17:35:09 cortex kernel: usb 1-2.1: new full-speed USB device number 79 using xhci_hcd
avril 08 17:35:09 cortex kernel: usb 1-2.1: Device not responding to setup address.
avril 08 17:35:09 cortex kernel: usb 1-2.1: Device not responding to setup address.
avril 08 17:35:09 cortex kernel: usb 1-2.1: device not accepting address 79, error -71
avril 08 17:35:09 cortex kernel: usb 1-2-port1: unable to enumerate USB device

jfischer-no, 2021-04-09:
@wysman can you please check STM32 relevant [release-nodes-2.5](https://docs.zephyrproject.org/latest/releases/release-notes-2.5.html). Maybe your issue is related to pinmux changes, "_STM32 pinmux driver has been reworked to allow pin configuration using devicetree definitions. The previous C macros are now deprecated._"

jfischer-no, 2021-04-09:
> **To Reproduce**
> Get a STM3F4 dev board

Exactly which board supported by Zephyr OS can be used to reproduce this problem?

wysman, 2021-04-09:
@jfischer-no I have test the sample on our **custom board** with your custom DTS. We have migrated the pinmux stuff during our 2.4 to 2.5 migration, and USB is working with the older mcuboot.

Our custom board are based on STM32F412RG.

erwango, 2021-04-09:
I've just had a try using `nucleo_f429zi` on `samples/subsys/usb/cdc_acm` and it seems to work fine, on latest master (`zephyr-v2.5.0-2093-ga074b335db`):
```
[74920.149870] usb 1-1: USB disconnect, device number 20
[74920.641092] usb 1-1: new full-speed USB device number 21 using xhci_hcd
[74920.791475] usb 1-1: New USB device found, idVendor=2fe3, idProduct=0001, bcdDevice= 2.05
[74920.791481] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[74920.791485] usb 1-1: Product: Zephyr CDC ACM sample
[74920.791488] usb 1-1: Manufacturer: ZEPHYR
[74920.791491] usb 1-1: SerialNumber: 373038323236510B
[74920.793837] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
```

carlescufi, 2021-04-14:
> I've just had a try using `nucleo_f429zi` on `samples/subsys/usb/cdc_acm` and it seems to work fine, on latest master (`zephyr-v2.5.0-2093-ga074b335db`):
>
> ```
> [74920.149870] usb 1-1: USB disconnect, [...]

@erwango did you use MCUboot? Because this is the issue here.

erwango, 2021-04-14:
> @erwango did you use MCUboot? Because this is the issue here.

@carlescuffi This was an observation vs following message:

> @jfischer-no I have test the samples/subsys/usb/cdc_acm application without mcuboot. It's do not works.

jfischer-no, 2021-05-18:
Close it since there is no answer and issue is not reproducible.

closed 2021-05-18 by jfischer-no

Cherish-forever, 2021-10-15:
@wysman 
I have the same issue, and I found the where the bug:
usb clock in stm32 max is 48MHZ. Change PLL /Q to limit clock to 48MHZ