zephyrproject-rtos/zephyr #34145: Convert NXP kinetis boards to have pindata in devicetree

galak, 2021-04-08:
- [ ] frdm_k22f
- [ ] frdm_k64f
- [ ] frdm_k82f
- [ ] frdm_kl25z
- [ ] frdm_kw41z
- [ ] hexiwear_k64
- [ ] hexiwear_kw40z
- [ ] ip_k66f
- [ ] twr_ke18f
- [ ] twr_kv58f220m
- [ ] usb_kw24d512

Kludentwo, 2021-04-10:
regarding frdm_kw41z: In the pinmux.c file it says kPORT_MuxAlt4 but in the dts file it says < 0 5 >. Is it muxed wrong?

galak, 2021-04-11:
@Kludentwo take a look at https://github.com/zephyrproject-rtos/zephyr/pull/34146

henrikbrixandersen, 2021-04-19:
@galak I had a look at the usb_kw24d512. I am not sure what you mean by "Need to figure out how to handle internal signals connected to modem"?

galak, 2021-04-19:
@henrikbrixandersen so I believe if you look at the manual for the SoC.  There is this section:

4.3 Internal Functional Interconnects and System Reset

The pins define there relate to how we setup SPI1 in boards/arm/usb_kw24d512/pinmux.c.  However those pins don't show up in the DTS pinmux data file as I dont believe they are in the xml data files from NXP.

So thus my comment on how to handle them.

closed 2021-05-03 by galak