zephyrproject-rtos/zephyr #34152: intel_adsp_cavs15: run tests/kernel/smp/ failed on ADSP

KangJianX, 2021-04-09:
**Describe the bug**
run this testcase failed on ADSP

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/smp/ --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/zephyrproject/modules/audio/sof/zephyr/ext/rimage/config,/home/ztest/work/zephyrproject/modules/audio/sof/zephyr/ext/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history"

**Logs and console output**

> Running test suite smp
===================================================================
START - test_smp_coop_threads
PASS - test_smp_coop_threads in 0.563 seconds
===================================================================
START - test_cpu_id_threads
PASS - test_cpu_id_threads in 1.1 seconds
===================================================================
START - test_coop_resched_threads
PASS - test_coop_resched_threads in 0.301 seconds
===================================================================
START - test_preempt_resched_threads
PASS - test_preempt_resched_threads in 0.512 seconds
===================================================================
START - test_yield_threads
PASS - test_yield_threads in 0.301 seconds
===================================================================
START - test_sleep_threads
PASS - test_sleep_threads in 1.1 seconds
===================================================================
START - test_wakeup_threads
PASS - test_wakeup_threads in 0.201 seconds
===================================================================
START - test_smp_ipi
cpu num=2 PASS - test_smp_ipi in 0.301 seconds
===================================================================
START - test_get_cpu
PASS - test_get_cpu in 0.51 seconds
===================================================================
START - test_fatal_on_smp
@ WEST_TOPDIR/zephyr/tests/kernel/smp/src/main.c:654
PASS - test_fatal_on_smp in 0.1 seconds
===================================================================
START - test_workq_on_smp
wrong error reason
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/smp/src/main.c:711: test_workq_on_smp: k_work_busy_get(&work) not equal to 0
FAIL - test_workq_on_smp in 0.51 seconds
===================================================================
Test suite smp failed.
===================================================================
PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
OS: Fedora28
Toolchain: Zephyr-sdk-0.12.3
Commit id: 5f42a3d4

yingmingx, 2021-04-12:
It seems that the testcase is OK, in passed on up_squared platform.  It is very likely that there is a problem with the ADSP platform. We need an expert in this field to have a look.

nashif, 2021-05-11:
failure confirmed

KangJianX, 2021-05-12:
@andyross @nashif 
This issue still can be reproduced on latest commit 8e69dafccf25bc5b4015a607cdba114216c4ea0e.
DEBUG   - DEVICE: START - **test_workq_on_smp**
DEBUG   - DEVICE: Caught system error -- reason 0 0
DEBUG   - DEVICE: Fatal error was unexpected, aborting...
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/smp/src/main.c:716: test_workq_on_smp: k_work_busy_get(&work) not equal to 0
DEBUG   - DEVICE:
DEBUG   - DEVICE: FAIL - test_workq_on_smp in 0.51 seconds

PR #35239 (2021-05-14): tests/kernel/smp: Misc synchronization fixups
A few mistakes in recent changes to this test:

The test_fatal_on_smp code was calling k_oops() with an irq_lock()
held, which on SMP will result in the global lock counter being wrong
and cause deadlocks on future calls to irq_lock().  Tt technically
"works" on UP, but only because the kernel can't detect the mistake.
This is another face of the "You can't context switch with interrupts
masked" rule.

There was a "LOCK_NO" (i.e. no locking!) case being exercised in
test_inc_concurrency, where three threads would race against each
other incrementin [... cut]
commit 6941c8fd (2021-05-14): tests/kernel/smp: Misc synchronization fixups

A few mistakes in recent changes to this test:

There was a "LOCK_NO" (i.e. no locking!) case being exercised in
test_inc_concurrency, where three threads would race against each
other incrementing and decrementing a single count without
synchronization.  And... it failed on cAVS.  Because there was no
synchronization.  Just remove.

The LOCK_IRQ (irq_un/lock()) case of the same test was was casting
taking a pointer to an integer (that stored the irq_lock() result) and
casting the pointer value to an integer instead of dereferencing it.

Also the  [... cut]
closed 2021-05-14 by nashif