zephyrproject-rtos/zephyr #34154: AArch64 PR reviews and merges are lagging behind

carlocaione, 2021-04-09:
This is a know problem but again I'm having a hard time convincing someone to review or merge AArch64-related PRs. For example I have a trivial PR #33181 opened since 9th of March with only one review but many others are sitting there praying someone would notice, ping doesn't seem to be effective either. 

Now, the problem is that only me and @npitre are actively working on AArch64 with a lot of people coming and go so it's hard to convince people to care. I know people are busy and overwhelmed so I understand this is perfectly normal.

I have a couple of ideas on how to improve this situation:
- Zephyrbot can increase priority or set a special label to PRs waiting for review since long time at least to raise awareness
- Someone allowed to merge should be more involved with this work
- Give me or @npitre the supreme power to merge into master

carlocaione, 2021-04-09:
Tagging the usual suspects @nashif @galak @ioannisg

carlescufi, 2021-04-09:
@carlocaione While I appreciate the situation of AArch64, this is a general problem in this project and many others, especially in areas where there is only one party contributing to at a particular moment in time. I have added the TSC label to see if we can make progress on your proposals.

closed 2021-04-29 by carlocaione