zephyrproject-rtos/zephyr #34167: RFC: memory regions from devicetree partitions

JordanYates, 2021-04-09:
## Introduction

The core problem that this issue is attempting to resolve is that the flash space of devices is partitioned in devicetree, but that partitioning is not used by the only tool that matters for memory layout, the linker. This results in problems such as #27471.

A secondary problem which can be resolved at the same time is the placement of variables at specific memory addresses. Placing variables at specific memory addresses requires the use of linker memory regions. The current setup creates memory regions from hardcoded devicetree nodes, but doesn’t allow users to create their own memory regions for their own purposes (see #31073).

### Proposed change

The proposed changes take 4 parts:
1. Force boards to be explicit about code offset/size for standalone applications
2. Provide a command line mechanism for specifying the partitions relevant for the build (standalone vs chain-loaded vs bootloader)
3. Automatically generate linker memory regions from devicetree partitions
4. Always control the location of output code via `zephyr,code-partition`

## Solution Background

Currently, the `FLASH` partition size and location is controlled in two ways. By default, the region is defined as the entire flash space, minus `CONFIG_FLASH_LOAD_OFFSET`. If `CONFIG_USE_DT_CODE_PARTITION` is enabled, `CONFIG_FLASH_LOAD_OFFSET` and `CONFIG_FLASH_LOAD_SIZE` have default values pulled from the devicetree partition. `CONFIG_USE_DT_CODE_PARTITION` is selected when applications enable a bootloader via `CONFIG_MCUBOOT` or similar.

Devicetree partitions segment flash spaces into discrete chunks, with the implication that partitions are reserved for some specific purpose, and should therefore not be overwritten by data from other partitions. Reserving memory at known addresses is the requirement of #31073, as long as the reservation is turned into a memory region.

There are currently 3 types of applications that are built using Zephyr:
1. Standalone applications
2. Applications that are chain-loaded by a bootloader
3. Bootloaders that chain-load other applications

Each type has different requirements for the space that the application can occupy and its location. Ideally, each application type results in a memory region covering the following:

1.  .
a. Occupies the entire flash space
b. Excluding any reserved memory regions (settings, etc)
2. Occupies a small memory region at the start of flash space
3. Occupies some fixed subset of the flash space at an arbitrary offset

To solve the requirements of #31073, we also need:
4. A generic way to specify an arbitrary number of memory regions.

If we reserve memory for arbitrary memory regions using devicetree partitions (as is already done in most in-tree boards for `settings`), the devicetree partitions are already encoding all the values we need for the above list except 1.a.

## Proposal

### 1. Force boards to be explicit about code offset/size for standalone applications 

Board files should contain a devicetree partition which specifies where code should be placed by default. This will be used in step 4 to provide a valid default target for `zephyr,code-partition`. The expectation is that any application which currently control image location via manually setting `CONFIG_FLASH_LOAD_OFFSET` can obtain the same functionality with a devicetree overlay.

This step can be implemented and merged independently, but logically belongs in the same PR as the layout files from step 2.

### 2. Provide a command line mechanism for specifying the partitions relevant for the build

Step 2 introduces the problem that we have memory regions that we don’t want in our build output (bootloader regions in standalone application and vice-versa). Step 4 will require a mechanism to change the value of `zephyr,code-partition` from the command line.

Because of the flow of the build system (Devicetree -> Kconfig -> Compilation -> Linking), the solution must be purely in devicetree. The canonical way to dynamically change the contents of the devicetree file is through overlays. As we want different overlays for the same application and board combination (same application standalone/chain-loaded), the overlay selection must be driven from the command line.

In order to include only the desired memory regions in the final output, the code region partition definitions should be moved into the overlay files together with the `zephyr,code-partition` selection. To keep the different code layouts distinct from other devicetree files, a new folder in the board directory is proposed `layouts`.

The simplest solution to select the correct overlay is a standard CMake variable with a default value. An optional Kconfig file could also be supported for convenience.
boilerplate.cmake:
```
if(NOT DEFINED CODE_LAYOUT)
  set(CODE_LAYOUT "default")
endif()
```
dts.cmake:
```
list(APPEND DTS_SOURCE ${BOARD_DIR}/layouts/${CODE_LAYOUT}.overlay)
```
Kconfig.cmake:
```
if((EXISTS ${BOARD_DIR}/layouts/${CODE_LAYOUT}.conf)
  list(INSERT CONF_FILE_AS_LIST 0 ${BOARD_DIR}/layouts/${CODE_LAYOUT}.conf)
endif()
```
Building would then be done by:
```
west build -b board zephyr/samples/basic/minimal #default layout
west build -b board zephyr/samples/basic/minimal -- -DCODE_LAYOUT=mcuboot #chain-loaded
```
West shortcuts could be supplied for user friendliness `west build --layout mcuboot`.
Alternatively the layout could be merged into the board specification `west build -b board%mcuboot` and extracted in CMake the same way board revisions are handled.

Step 1 is a prerequisite.

### 3. Automatically generate linker memory regions from devicetree partitions

Use the partitions under the `zephyr,flash` chosen node to automatically generate memory regions. Memory regions are named from the `label` property of the partition, with offsets and sizes from the `reg` property. All values are exposed via the devicetree macros, and all partitions can be easily enumerated via `DT_FOREACH_CHILD`.

This step will automatically solve #31073, and can be implemented and merged independent of other steps. If merged before step 2 it will result in overlapping memory regions being present in the final build output, as seen below:
```
[139/139] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:        5600 B         1 MB      2.14%
            SRAM:        5600 B       256 KB      2.14%
         storage:          0 GB        32 KB      0.00%
         mcuboot:          0 GB        48 KB      0.00%
         image-0:          0 GB       412 KB      0.00%
         image-1:          0 GB       412 KB      0.00%
   image-scratch:          0 GB       120 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
```
This does not cause problems beyond the sub-optimal display.

### 4. Always control the location of output code via `zephyr,code-partition`

Remove the `FLASH` memory region, and when `CONFIG_XIP` is set, define `ROMABLE_REGION` to the label value of `zephyr,code-partition`. This will result in all values currently placed in the `FLASH` region to be placed in the selected partition.

This will allow for a substantial simplification of the image generation options, namely the removal of `USE_DT_CODE_PARTITION`, `HAS_FLASH_LOAD_OFFSET`, `FLASH_LOAD_OFFSET` and `FLASH_LOAD_SIZE` and their associated linker script conditionals.

Steps 1,2 and 3 are prerequisites.
This will be the breaking change for all out-of-tree boards, as the Zephyr pattern is to set `zephyr,code-partition` to the first application slot for use with bootloaders. This can be made to cause immediate compilation failures with appropriate error messages by validating the existence of the layout files in step 3.

## Summary

After applying all 4 steps, code artifacts are always constrained to a memory range explicitly set by the board dts file/overlays, solving #27471. Users can generate/reserve arbitrary memory regions in their flash space, solving #31073. Minor quality of life improvements can be made (specifying the mcuboot partition could automatically enable `CONFIG_MCUBOOT` for example). Linker script conditionals/options are simplified.

mrrosen, 2021-04-23:
I recently ran into the same thing a few weeks ago where the linker script I was using (in this case one based off of one for Xtensa) was not structured to use the information provided in devicetree to place the output sections. This problem extends beyond the partitions in flash but in fact to the entire way the linker scripts are structured to try to figure out where to put both code and data (which is very inconsistent between architectures despite trying to solve the same problem); Zephyr also has a chosen node for the SRAM as well that isnt always used directly.

The way I thought to solve it was to create a new compatibility for all memory devices, whether RAM/FLASH/ROM/etc called `"linkable-memory"` which allowed the linker to fill in the memory command for all devices that use this compatibility:

dts:

    sys_ram: memory@60000000 {
		device_type = "memory";
		compatible = "mmio-sram", "linkable-memory";
		reg = <0x60000000 DT_SIZE_M(64)>;
	};

linker.ld:

    #define DT_DRV_COMPAT linkable_memory
    
    #define MEMORY_REGION(n) \
      DT_DRV_INST(n) : org = DT_INST_REG_ADDR(n), len = DT_INST_REG_SIZE(n)
    
    MEMORY
    {
    DT_INST_FOREACH_STATUS_OKAY(MEMORY_REGION)
    #ifdef CONFIG_GEN_ISR_TABLES
      IDT_LIST : org = 0x80000000, len = 0x2000
    #endif
    }
    
    #if DT_HAS_CHOSEN(zephyr_code_partition) && DT_NODE_HAS_COMPAT(DT_CHOSEN(zephyr_code_partition), linkable_memory)
    #define ROMABLE_REGION DT_CHOSEN(zephyr_code_partition)
    #else
    #error "A chosen code partition must be specified and be a linkable memory"
    #endif
    
    #if DT_HAS_CHOSEN(zephyr_sram) && DT_NODE_HAS_COMPAT(DT_CHOSEN(zephyr_sram), linkable_memory)
    #define RAMABLE_REGION DT_CHOSEN(zephyr_sram)
    #else
    #error "A chosen sram must be specified and be a linkable memory"
    #endif

I also added dedicated sections in the linker script to allow any other linkable memory to be accessed from code using the node id as an attribute (not as nice as your suggestion to use the label but it does avoid the requirement to have a label as many memory elements in device tree doesnt have it):

    #define CREATE_OUTPUT_SECTION_FOR_REGION(n) \
      SECTION_PROLOGUE(.DT_DRV_INST(n),, ALIGN(4)) \
      { \
        KEEP(*(.DT_DRV_INST(n).rodata)) \
        KEEP(*(.DT_DRV_INST(n).text)) \
        KEEP(*(.DT_DRV_INST(n).*)) \
        KEEP(*(.DT_DRV_INST(n))) \
      } >DT_DRV_INST(n)
    
    DT_INST_FOREACH_STATUS_OKAY(CREATE_OUTPUT_SECTION_FOR_REGION)

I dont necessarily think this is the best approach, but I do want to bring up there might be more to do rather than just enable the flash partitions to be brought out in the linker as memory regions, but if this RFC is intended to be limited to flash partitions for XIP targets, this more expansive discussion might fit better in a different one. Another few things to mention:

1. Not all flash partitions are memory mapped (ie, partitions in a SPI flash), so it might be that the chosen flash isnt actually in the memory space of the processor and so shouldnt be added to the linker script
2. RAMs can be partitioned as well (see boards/arm/nrf5340dk_nrf5340/nrf5340dk_nrf5340_common.dts, reserved-memory), so it would be good to use this information in the linker script as well
3. Due to how I use a new compatibility to instruct the linker to include the memory/region/partition, it can be a bit ugly to override in overlays, so there is that to consider
4. XIP and non-XIP can both be supported in my setup, but I dont currently have a clean implementation. My thoughts on it were that memories can be split into groups: memory that requires no initialization (either doesnt need it at all like noinit or is preloaded before the code is run like code in flash or data loaded by a bootloader), memory that requires initialization to 0 (bss), memory that requires initialization from another section (like data in XIP targets, copy data initial values from flash to ram). Ideally, the memories/partitions type can be labelled in devicetree or inferred somehow and then the linker script and initialization code can handle it accordingly.

Again, these are just some thoughts/solutions I had solving this problem for our own project and how I thought it might be generalized across Zephyr.

JordanYates, 2021-04-24:
Thanks for the input @mrrosen. I quite like the idea of the `linkable-memory` compatible, as it elegantly solves the issue of which devicetree nodes should result in memory regions, without needing to add additional properties to nodes. I think the case where a node with the `linkable-memory` has children could be special cased to generate regions for the children instead, allowing the following style to work nicely:
```
&flash0 {
        compatible = "soc-nv-flash", "linkable-memory";
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x000000000 0x0000C000>;
		};
		slot0_partition: partition@c000 {
			label = "image-0";
			reg = <0x0000C000 0x000067000>;
		};
        };
};
```
Regarding the use of `label` to name memory regions, there could be a fallback to `DT_NODE_FULL_NAME()` when no label is defined. 

1. Can the chosen flash validly point to a non memory-mapped device? What does that even signify?
2. This was not intended to be purely for XIP flash, the pattern chosen to generate regions for ROM should be extendable to RAM as well. I am currently looking at M33 devices as well.
3. I don't think this is a problem. If the application doesn't want the region, just don't place anything in it. I don't see a use case for adding the compatible to existing nodes, as if they are addressable they should already have the compatible.
4. If my suggestion to always use `zephyr,code-partition` is adopted, that can be used to infer XIP or not.
5. Using the special case above solves that issue, if the logic recursively applies to children.

I have put in a proposal for a mini-conference on this topic at the Zephyr Developer Summit, if that is approved it would be good to have your input there as well.

mrrosen, 2021-04-26:
> Regarding the use of `label` to name memory regions, there could be a fallback to `DT_NODE_FULL_NAME()` when no label is defined.

While I like that idea, can memory regions have `@` characters in their name? From what I can tell I guess you could stringify it, but I havent tried it.

> 1. Can the chosen flash validly point to a non memory-mapped device? What does that even signify?

For some reason, I remembered seeing the chosen `zephyr,flash` to be an external SPI flash (not memory mapped) for a board for some function. But since I cant find it, I guess its a non-issue. I just wanted to mention that Im not sure chosen `zephyr,flash` should have an impact on the linker, only chosen `zephyr,code-partition` and chosen `zephyr,sram`.

> 3. I don't think this is a problem. If the application doesn't want the region, just don't place anything in it. I don't see a use case for adding the compatible to existing nodes, as if they are addressable they should already have the compatible.

So long as partitioning of a memory/flash is a child of that memory/flash in the device tree, I agree this is not a problem. RAM partitioning done in `reserved-memory` would have to change though, as if I define both the entire SRAM as `"linkable-memory"` and a subset of it as `"linkable-memory"` in `reserved-memory`, it will cause a conflict in the linker script due to overlapping memory regions. What Ive had to do it override the compatibility of the larger memory if it is being partitioned, which is what I was referring to here and is not very pretty.

> 4. If my suggestion to always use `zephyr,code-partition` is adopted, that can be used to infer XIP or not.

Agreed, its just that these types of memory I mentioned have to be addressed. Even in non-XIP memory, its possible there are RAMs in the system used by the application that are not initialized by a bootloader; requiring the application to copy over data to them. (Im thinking something like the RTC SRAMs in ESP32; it would be nice if they could support data segments with initialized data loaded at boot-time). Currently, only the data code is loaded for XIP (and has separate load and virtual address), but Im mentioning this concept might need to be generalized for Zephyr's architecture initialization for all architectures.

> 5. Using the special case above solves that issue, if the logic recursively applies to children.

Having all partitioning of memory of children does help solve problems. Im just mentioning this will make it difficult to support overlapping partitioning of memory which is currently easy to do; though Im not coming up for a reason why this might be useful, its possible others have a need for it.

> I have put in a proposal for a mini-conference on this topic at the Zephyr Developer Summit, if that is approved it would be good to have your input there as well.

Ill keep a look out; thanks!

de-nordic, 2021-05-19:
I think that this issue https://github.com/zephyrproject-rtos/zephyr/pull/34945 is also relevant here.

nandojve, 2021-05-26:
I think any improvement in this field is welcome. I would like comment this RFC could impact other bootloader and/or tools like `bossac runner`.

besmarsh, 2023-01-03:
Has there been any progress/discussion in this area since a year and a half ago? I'm interested in this improvement but I can't see any recent activity. @JordanYates @mrrosen

JordanYates, 2023-01-03:
Progress on snippets stalled, and without an easy way to modify partitions from the command line build tools this proposal is not really feasible.