zephyrproject-rtos/zephyr #34178: apds9960 sensor sample does not build on STM32

GunZi200, 2021-04-09:
**Describe the bug**
Example for sensor apds9960 doesn't build for board Nucleo L412RB-P, I tried also Nucleo F401RE with same results.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p -b nucleo_l412rb_p samples/sensor/apds9960`
2. See various DT errors

**Expected behavior**
The example should finish building.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```
PS C:\Users\Gudni\OneDrive\Desktop\zephyrbase\zephyr> west build -p -b nucleo_l412rb_p samples/sensor/apds9960       
-- west build: making build dir C:\Users\Gudni\OneDrive\Desktop\zephyrbase\zephyr\build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/samples/sensor/apds9960
-- Zephyr version: 2.5.99 (C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr)
-- Found Python3: C:/Python39/python.exe (found suitable exact version "3.9.2") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: nucleo_l412rb_p
-- Cache files will be written to: C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/.cache
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (C:/toolchains/gnu_arm_embedded/10-2020-q4-major)
-- Found BOARD.dts: C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/boards/arm/nucleo_l412rb_p/nucleo_l412rb_p.dts
-- Generated zephyr.dts: C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/build/zephyr/include/generated/device_extern.h
Parsing C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/Kconfig
Loaded configuration 
'C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/boards/arm/nucleo_l412rb_p/nucleo_l412rb_p_defconfig'
Merged configuration 'C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/samples/sensor/apds9960/prj.conf'
Configuration saved to 'C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/toolchains/gnu_arm_embedded/10-2020-q4-major/bin/arm-none-eabi-gcc.exe
CMake Deprecation Warning at C:/Users/Gudni/OneDrive/Desktop/zephyrbase/modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/build
-- west build: building application
[64/146] Building C object zephyr/drivers/sensor/apds9960/CMakeFiles/drivers__sensor__apds9960.dir/apds9960.c.obj    
FAILED: zephyr/drivers/sensor/apds9960/CMakeFiles/drivers__sensor__apds9960.dir/apds9960.c.obj 
C:\toolchains\gnu_arm_embedded\10-2020-q4-major\bin\arm-none-eabi-gcc.exe -DBUILD_VERSION=zephyr-v2.5.0-2114-gd307bd2fddc0 -DCORE_CM4 -DKERNEL -DSTM32L412xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/arm/st_stm32/stm32l4 -I../drivers -I../soc/arm/st_stm32/common -IC:/Users/Gudni/OneDrive/Desktop/zephyrbase/modules/hal/cmsis/CMSIS/Core/Include -IC:/Users/Gudni/OneDrive/Desktop/zephyrbase/modules/hal/stm32/stm32cube/stm32l4xx/soc -IC:/Users/Gudni/OneDrive/Desktop/zephyrbase/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include -IC:/Users/Gudni/OneDrive/Desktop/zephyrbase/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include/Legacy -IC:/Users/Gudni/OneDrive/Desktop/zephyrbase/modules/hal/stm32/stm32cube/common_ll/include -isystem ../lib/libc/minimal/include -isystem c:/toolchains/gnu_arm_embedded/10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include -isystem c:/toolchains/gnu_arm_embedded/10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include-fixed -Os -imacros C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/samples/sensor/apds9960=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/Users/Gudni/OneDrive/Desktop/zephyrbase=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/drivers/sensor/apds9960/CMakeFiles/drivers__sensor__apds9960.dir/apds9960.c.obj -MF zephyr\drivers\sensor\apds9960\CMakeFiles\drivers__sensor__apds9960.dir\apds9960.c.obj.d -o zephyr/drivers/sensor/apds9960/CMakeFiles/drivers__sensor__apds9960.dir/apds9960.c.obj -c C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/drivers/sensor/apds9960/apds9960.c
In file included from ../include/arch/arm/aarch32/arch.h:20,
                 from ../include/arch/cpu.h:19,
                 from ../include/kernel_includes.h:33,
                 from ../include/kernel.h:17,
                 from ../include/init.h:11,
                 from ../include/device.h:29,
                 from C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/drivers/sensor/apds9960/apds9960.c:14:       
../include/devicetree.h:300:40: error: 'DT_N_INST_0_avago_apds9960_BUS_P_label' undeclared here (not in a function)  
  300 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                                        ^~~~~~~~~
../include/devicetree.h:2251:24: note: in definition of macro 'DT_CAT'
 2251 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
../include/devicetree.h:1701:31: note: in expansion of macro 'DT_PROP'
[... 161 lines cut ...]
      |  ^~~~~~~~~~~~~~~~
../include/sys/util_internal.h:84:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
   84 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
../include/devicetree.h:300:31: note: in expansion of macro 'UTIL_CAT'
  300 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                               ^~~~~~~~
../include/devicetree.h:1745:27: note: in expansion of macro 'DT_INST'
 1745 | #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
      |                           ^~~~~~~
../include/device.h:238:19: note: in expansion of macro 'DT_DRV_INST'
  238 |  DEVICE_DT_DEFINE(DT_DRV_INST(inst), __VA_ARGS__)
      |                   ^~~~~~~~~~~
C:/Users/Gudni/OneDrive/Desktop/zephyrbase/zephyr/drivers/sensor/apds9960/apds9960.c:546:1: note: in expansion of macro 'DEVICE_DT_INST_DEFINE'
  546 | DEVICE_DT_INST_DEFINE(0, apds9960_init,
      | ^~~~~~~~~~~~~~~~~~~~~
[81/146] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\Gudni\OneDrive\Desktop\zephyrbase\zephyr\build'
PS C:\Users\Gudni\OneDrive\Desktop\zephyrbase\zephyr> 
```

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: ARM GNU 10-2020-q4-major (GCC 10.2.1)

galak, 2021-04-09:
You need a devicetree node for the sensor, similar to what is in boards/arm/reel_board/dts/reel_board.dtsi.

This is not a bug, but expected behavior as the board does not support the sensor by default.

closed 2021-04-09 by galak

GunZi200, 2021-04-09:
Thanks @galak I will take a better look.