zephyrproject-rtos/zephyr #34184: video samples fail to build

galak, 2021-04-10:
The following samples fail to build:
```
samples/subsys/video/capture/sample.video.capture in mimxrt1064_evk:samples/subsys/video/capture/sample.video
samples/subsys/video/tcpserversink/sample.video.tcpserversink in mimxrt1064_evk:samples/subsys/video/tcpserversink/sample.video
```

Failure is of form:

```
/workdir/zephyr/drivers/video/video_sw_generator.c: In function 'video_sw_generator_stream_stop':
/workdir/zephyr/drivers/video/video_sw_generator.c:66:2: error: too few arguments to function 'k_work_cancel_delayable_sync'
   66 |  k_work_cancel_delayable_sync(&data->buf_work);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../../../../../../../include/zephyr.h:18,
                 from /workdir/zephyr/drivers/video/video_sw_generator.c:6:
../../../../../../../include/kernel.h:3363:6: note: declared here
 3363 | bool k_work_cancel_delayable_sync(struct k_work_delayable *dwork,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

galak, 2021-04-10:
Possible fix:

```
diff --git a/drivers/video/video_sw_generator.c b/drivers/video/video_sw_generator.c
index 2b78d44e84..5514b6b9e2 100644
--- a/drivers/video/video_sw_generator.c
+++ b/drivers/video/video_sw_generator.c
@@ -63,7 +63,7 @@ static int video_sw_generator_stream_stop(const struct device *dev)
 {
        struct video_sw_generator_data *data = dev->data;
 
-       k_work_cancel_delayable_sync(&data->buf_work);
+       k_work_cancel_delayable_sync(&data->buf_work, NULL);
 
        return 0;
 }
```

loicpoulain, 2021-04-10:
cf #34186

PR #34186 (2021-04-10): drivers: video: sw_generator: Fix missing parameter
Fix #34184
closed 2021-04-10 by nashif