zephyrproject-rtos/zephyr #34204: nvs_write: Bad documented return value.

Vladimirek, 2021-04-11:
Docs say:
https://github.com/zephyrproject-rtos/zephyr/blob/1342d12500867d3de09025037e65e756d5486eba/include/fs/nvs.h#L105-L106

But when the key is already in flash with the same value zero is returned:
https://github.com/zephyrproject-rtos/zephyr/blob/6acc4a25213e70ad47524a176bcf47431b0547e5/subsys/fs/nvs/nvs.c#L842-L846

Need update code or doc header.

Laczen, 2021-04-11:
@Vladimirek, thanks for reporting this. I am unsure what the error is, when a rewrite of the same data is done, nothing is written to flash, so nvs reports 0 (as documented). 

Wether this operation is succesful or not, this can be discussed.

Vladimirek, 2021-04-11:
> so nvs reports 0 (as documented). 
That is exactly what I missing. Simple note about 0 will be describe the situation exactly.
_Number of bytes written. On success, it will be equal to the number of bytes requested to be written. **Zero if value already present.** On error returns -ERRNO code._ 
Yes I know that there is documented on other place, but who have a time read it all :-)

nvlsianpu, 2021-04-12:
I think from user perspective API should return the number of bytes requested to be written for this case as well.

nvlsianpu, 2021-04-15:
Will response to this in a patch.

Laczen, 2021-04-15:
@nvlsianpu, I don't think a patch is needed. It should be possible for a user to distinguish between a real write to flash and a trial to rewrite. If a documentation update is required i leave up to you.

closed 2021-08-29 by github-actions[bot]

jefffhaynes, 2022-04-01:
> @nvlsianpu, I don't think a patch is needed. It should be possible for a user to distinguish between a real write to flash and a trial to rewrite. If a documentation update is required i leave up to you.

Decisions like this are why zephyr isn't as successful as it could be. The user doesn't care _how_ the data was written, only that it was written. This just makes the API needlessly confusing.