zephyrproject-rtos/zephyr #34206: Question: Is zephyrproject actively maintaining the windows-curses sub-project?

pjfarleyiii, 2021-04-11:
There have been several issues filed in the zephyrproject-rtos/windows-curses sub-project since June 2020 with no responses.  I am asking here whether zephyrproject-rtos intends to actively maintain the windows-curses sub-project or whether issues in windows-curses must be resolved via fork and independent debugging and corrections.

Peter

carlescufi, 2021-04-12:
@pjfarleyiii thanks for the interest. The windows curses package was originally developed to enable menuconfig on Windows, and a new release has been made with the latest (3.9) Python repository. While the original maintainer is no longer involved with Zephyr, I do believe that maintaining the respository's location as the reference one is within the intent of the project. So I would encourage anyone to send Pull Requests and try to help with the issues, and we will make releases of it as needed.

pjfarleyiii, 2021-04-13:
I guess my question then becomes:  What can a "just plain python user" do when an issue s/he has found in windows-curses is blocking their own work, and they do not have the skills or experience to tackle debugging code as deep and complex as curses seems to be (especially the underlying PDCurses library)?

I would personally be more than happy to make a generous donation to anyone who would do that deep debugging work for windows-curses users and make windows-curses better for everyone who needs it.

Regards,

Peter

closed 2021-05-18 by MaureenHelm