zephyrproject-rtos/zephyr #34214: codes reference weak variable are optimized out

mengxianglinx, 2021-04-12:
**Describe the bug**
In subsys/power/device.c, there is a variable defined as
__weak const char *const z_pm_core_devices[] = {...
When define a strong version of this variable in other file, the code in this file
153                 while (z_pm_core_devices[cdi]) {
154                         if (strcmp(dev->name, z_pm_core_devices[cdi]) == 0) {
155                                 pm_devices[cdi] = pmi;
156                                 break;
157                         }
158                         ++cdi;
159                 }
can't be compiled correctly, they are optimized out.

**To Reproduce**
Steps to reproduce the behavior:
1. cherry pick patch https://github.com/mengxianglinx/zephyr/commit/60de96996c0dd8b3e8bb2c837b95afd08d032318
2. run "twister -s tests/subsys/power/power_mgmt/subsys.power.interface -p qemu_x86", test case failed
3. run "twister -s tests/subsys/power/power_mgmt/subsys.power.interface -p qemu_x86 --coverage", test case pass
4. compare zephyr.lst generated in step 2 and step 3, assembly codes for the codes above are different

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK 0.12.2, 0.12.3, 0.12.4)

**Additional context**
Add any other context about the problem here.

mengxianglinx, 2021-04-12:
before root cause, I have tried four work around:
1. change "__weak const char *const z_pm_core_devices[]" to "__weak const char *z_pm_core_devices[]"
2. define a local variable like pm_core_devices inside function pm_create_device_list() to reference z_pm_core_devices and use that variable in the function
3. add "__atrribute__(optimize(O0)) to function pm_create_device_list()
4. add configuration to test case just like "--coverage" is enbled in command line.

tejlmand, 2021-04-21:
Did some investigation on this, and looking at both the map file, and the elf file, it does look correct:
Map file, `z_pm_core_devices` symbol is chosen from correct object file:
```
 .rodata.z_pm_core_devices                                                                          
            0x000000000010916c        0x8 app/libapp.a(pm_interfaces.c.obj)                     
            0x000000000010916c                z_pm_core_devices        
```

and in the elf:
```
 109160 78745f73 74617465 00000000 b1931000  xt_state........
```
so let's look at address: `0x001093b1`:
```
1093b0 0064756d 6d795f64 72697665 72005745  .dummy_driver.WE
```
So content is correct.

Let's look at the code:
devices.c
```
__weak const char *const z_pm_core_devices[] = { 
    NULL
};
```
and:
pm_interfaces.c
```
const char * const z_pm_core_devices[] = {"dummy_driver", (char *)0};
```

so the weak array is of size 1, and the strong array is size 2.

Let's align the sizes, by adding the following to devices.c:
```diff
diff --git a/subsys/power/device.c b/subsys/power/device.c
index 1fb1c35d04..fd8ce82a6a 100644
--- a/subsys/power/device.c
+++ b/subsys/power/device.c
@@ -37,7 +37,7 @@ __weak const char *const z_pm_core_devices[] = {
 #elif defined(CONFIG_SOC_SERIES_STM32L4X) || defined(CONFIG_SOC_SERIES_STM32WBX)
        "sys_clock",
 #endif
-       NULL
+       NULL, NULL
 };
 
 /* Ordinal of sufficient size to index available devices. */
```

and then run:
```
$ ./scripts/twister -s tests/subsys/power/power_mgmt/subsys.power.interface -p qemu_x86  -N  --inline-logs
...
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 skipped with 0 warnings in 8.03 seconds
INFO    - In total 7 test cases were executed, 0 skipped on 1 out of total 342 platforms (0.29%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
```
and the log:
```
2021-04-21 23:01:40,523 - twister - DEBUG - QEMU (35245): Running test suite power_management_test
2021-04-21 23:01:40,524 - twister - DEBUG - QEMU (35245): ===================================================================
2021-04-21 23:01:40,524 - twister - DEBUG - QEMU (35245): START - test_suspend_device
2021-04-21 23:01:40,526 - twister - DEBUG - QEMU (35245): Async wakeup request queued
2021-04-21 23:01:40,526 - twister - DEBUG - QEMU (35245): Dummy device resumed
2021-04-21 23:01:40,528 - twister - DEBUG - QEMU (35245): PASS - test_suspend_device in 0.1 seconds
2021-04-21 23:01:40,528 - twister - DEBUG - QEMU (35245): ===================================================================
2021-04-21 23:01:40,529 - twister - DEBUG - QEMU (35245): START - test_device_cannot_suspend
2021-04-21 23:01:40,529 - twister - DEBUG - QEMU (35245): Async wakeup request queued
2021-04-21 23:01:40,529 - twister - DEBUG - QEMU (35245): Dummy device resumed
2021-04-21 23:01:40,532 - twister - DEBUG - QEMU (35245): PASS - test_device_cannot_suspend in 0.328 seconds
2021-04-21 23:01:40,532 - twister - DEBUG - QEMU (35245): ===================================================================
2021-04-21 23:01:40,532 - twister - DEBUG - QEMU (35245): START - test_device_pm_disable
2021-04-21 23:01:40,532 - twister - DEBUG - QEMU (35245): PASS - test_device_pm_disable in 0.1 seconds
2021-04-21 23:01:40,532 - twister - DEBUG - QEMU (35245): ===================================================================
2021-04-21 23:01:40,532 - twister - DEBUG - QEMU (35245): START - test_device_pm_state_str
2021-04-21 23:01:40,533 - twister - DEBUG - QEMU (35245): PASS - test_device_pm_state_str in 0.1 seconds
2021-04-21 23:01:40,533 - twister - DEBUG - QEMU (35245): ===================================================================
2021-04-21 23:01:40,533 - twister - DEBUG - QEMU (35245): Test suite power_management_test succeeded
2021-04-21 23:01:40,533 - twister - DEBUG - QEMU (35245): ===================================================================
2021-04-21 23:01:40,533 - twister - DEBUG - QEMU (35245): PROJECT EXECUTION SUCCESSFUL
```

So both compiler and linker behaves correct, but the sizes of the  weak and the strong arrays must be aligned.

carlescufi, 2021-04-22:
@gmarull FYI ultimately we probably want to get the array length from Devicetree here.

gmarull, 2021-04-22:
@carlescufi agreed. I'm currently looking at all these things.

ceolin, 2021-04-28:
I got a similar problem when trying to generate the device list at build time using the address from the pre-built image. The length of the weak / strong have to be the same otherwise we get into problems.

carlescufi, 2021-05-19:
@galak this is not going to be fixed for 2.6.0, since it requires a substantial change, as per conversations with @gmarull. That said, the actual description is not exactly a bug, it's a compiler feature where differently sized arrays will not override the original weak array.

commit 3dad4a06 (2021-05-27): pm: device: Get rid of z_pm_core_devices

z_pm_core_devices was hack to set some priority between devices. It
was doing it hardcoding some devices that were the first to bring up
and the last to power down. Remove it and use the same list used to
initialize devices.

Fixes #34214
closed 2021-05-27 by nashif