zephyrproject-rtos/zephyr #34218: Can't connect to civetweb webserver via browser

gebauerf, 2021-04-12:
**Describe the bug**
I tried to run the civetweb sample on the recommended target (Atmel SAM-E70 Xplained). Building and flashing seems to work fine, but i can't connect to the webserver via browser @ IP: 192.0.2.1:8080. I'm using the Revision "B" of the Atmel-Chip, that i already figured out. Before this, i flashed a bunch of basic samples without any issue.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b sam_e70b_xplained samples\net\civetweb\http_server
2. west flash

**Expected behavior**
I expected to connect to the webserver's basic environment via browser (chrome), described in zephyr's sample documentation https://docs.zephyrproject.org/2.5.0/samples/net/civetweb/http_server/README.html

**Logs and console output**
\`\`\`
[00:00:00.000,000] <inf> i2c_sam_twihs: Device I2C_2 initialized
[00:00:00.000,000] <inf> i2c_sam_twihs: Device I2C_0 initialized
[00:00:00.001,000] <inf> eth_sam: MAC: fc:c2:3d:0c:2e:12
[00:00:00.001,000] <inf> eth_sam: Queue 0 activated
[00:00:00.001,000] <inf> eth_sam: Queue 1 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 2 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 3 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 4 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 5 set to idle
[00:00:00.001,000] <inf> eth_sam_phy: Soft Reset of ETH PHY
*** Booting Zephyr OS build zephyr-v2.5.0  ***
[00:00:00.091,000] <inf> eth_sam_phy: PHYID: 0x221561 at addr: 0
[00:00:00.095,000] <inf> net_config: Initializing network
[00:00:00.095,000] <inf> net_config: Waiting interface 1 (0x2040483c) to be up...
[00:00:03.122,000] <inf> eth_sam: Link up
[00:00:06.027,000] <inf> eth_sam_phy: common abilities: speed 100 Mb, full duplex
[00:00:06.027,000] <inf> net_config: Interface 1 (0x2040483c) coming up
[00:00:06.027,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:06.028,000] <dbg> lib_extensions.sscanf: [IMPLEMENTATION MISSING : sscanf]
\`\`\`

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: gnu arm embedded
 - Zephyr Version: 2.5.0

Nukersson, 2021-04-13:
Hello @gebauerf and thank you for your bug description here. 

Can you test any other simpler sample, which is using ethernet driver? I would first try to determine if ethernet driver works properly.

mglettig, 2021-04-13:
Hi @Nukersson & @gebauerf 

The same error can be reproduced when running the sample with qemu_x86. However on this setup it didn't crash when testing with Firefox browser. Also the sample does not crash when running a GET request with CURL. For me it looks like the problem is that strftime is not implemented (just stubed). I guess that civerweb expects to get an initialized char array when calling strftime.

Nukersson, 2021-04-13:
> Hi @Nukersson & @gebauerf
>
> The same error can be reproduced when running the sample with qemu_x86. However on this setup it didn't crash when testing with Firefox browser. Also the sample does not [...]

Could you test it with your suggested fix?

gebauerf, 2021-04-13:
Hi @Nukersson and thank you for your response.

I just tried the "dumb_http_server"-sample and this one worked just fine. I even connected multiple times to the webserver to ensure proper functionality.

```
[00:00:00.000,000] <inf> i2c_sam_twihs: Device I2C_2 initialized
[00:00:00.000,000] <inf> i2c_sam_twihs: Device I2C_0 initialized
[00:00:00.001,000] <inf> eth_sam: MAC: fc:c2:3d:0c:2e:12
[00:00:00.001,000] <inf> eth_sam: Queue 0 activated
[00:00:00.001,000] <inf> eth_sam: Queue 1 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 2 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 3 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 4 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 5 set to idle
[00:00:00.001,000] <inf> eth_sam_phy: Soft Reset of ETH PHY
*** Booting Zephyr OS build zephyr-v2.5.0  ***
[00:00:00.091,000] <inf> eth_sam_phy: PHYID: 0x221561 at addr: 0
[00:00:00.095,000] <inf> net_config: Initializing network
[00:00:00.095,000] <inf> net_config: Waiting interface 1 (0x2040062c) to be up...
[00:00:02.112,000] <inf> eth_sam: Link up
Single-threaded dumb HTTP server waits for a connection on port 8080...
[00:00:03.915,000] <inf> eth_sam_phy: common abilities: speed 100 Mb, full duplex
[00:00:03.915,000] <inf> net_config: Interface 1 (0x2040062c) coming up
[00:00:03.915,000] <inf> net_config: IPv4 address: 192.168.1.199
Connection #0 from 192.168.1.2
Connection from 192.168.1.2 closed
Connection #1 from 192.168.1.2
Connection from 192.168.1.2 closed
Connection #2 from 192.168.1.2
Connection from 192.168.1.2 closed
Connection #3 from 192.168.1.2
Connection from 192.168.1.2 closed
```

[image]

Nukersson, 2021-04-13:
> Hi @Nukersson and thank you for your response.
>
> I just tried the "dumb_http_server"-sample and this one worked just fine. I even connected multiple times to the webserver to ensure proper [...]

Okay. This is very good. I'll take a look at civetweb later.

mglettig, 2021-04-13:
> Could you test it with your suggested fix?

To  check if the problem is actually related to strftime I added a dummy implementation: strncpy(dst,"dummy",dst_size); in the function strftime in zephyr/samples/net/civetweb/common/src/libc_extensions.c. However the error was still there.

gebauerf, 2021-04-14:
Update:
I tried to debug the civetweb-sample to evaluate the point zephyr crashes. Maybe the call-stack helps to find the issue.

[image]

After trying to connect to the webserver with several browsers, its interesting that zephyr does not crash when connecting to it with curl. After the webserver is up... (I'm using DHCP now)

```
[00:00:06.138,000] <inf> net_config: Running dhcpv4 client...
[00:00:06.138,000] <dbg> lib_extensions.sscanf: [IMPLEMENTATION MISSING : sscanf]

[00:00:10.243,000] <inf> net_dhcpv4: Received: 192.168.0.129
[00:00:10.243,000] <inf> net_config: IPv4 address: 192.168.0.129
[00:00:10.243,000] <inf> net_config: Lease time: 3600 seconds
[00:00:10.243,000] <inf> net_config: Subnet: 255.255.255.0
[00:00:10.243,000] <inf> net_config: Router: 192.168.0.1
```

...using the curl command multiple times...

`curl 192.168.0.129:8080`

...the webserver responds without crashing...

```
[00:04:01.381,000] <dbg> mg_cry_internal_impl.log_access: 192.168.0.143 - "GET /?null HTTP/1.1" 200
[00:04:07.670,000] <dbg> mg_cry_internal_impl.log_access: 192.168.0.143 - "GET /?null HTTP/1.1" 200
[00:04:16.725,000] <dbg> mg_cry_internal_impl.log_access: 192.168.0.143 - "GET /?null HTTP/1.1" 200
[00:04:28.214,000] <dbg> mg_cry_internal_impl.log_access: 192.168.0.143 - "GET /?null HTTP/1.1" 200
[00:20:00.023,000] <dbg> mg_cry_internal_impl.log_access: 192.168.0.143 - "GET /?null HTTP/1.1" 200
[00:20:03.044,000] <dbg> mg_cry_internal_impl.log_access: 192.168.0.143 - "GET /?null HTTP/1.1" 200
```

I hope this helps.

gebauerf, 2021-04-15:
Another update:
Thanks to @mglettig, i partially solved the issue. When connecting to the webserver via browser, there's a GET request for an unimplemented resource called "favicon.ico". When this request is blocked in the developer-options of the browser, zephyr does not crash!

[image]

The Output-Log seems fine now.

```
...
[00:00:14.256,000] <inf> net_dhcpv4: Received: 192.168.0.129
[00:00:14.256,000] <inf> net_config: IPv4 address: 192.168.0.129
[00:00:14.256,000] <inf> net_config: Lease time: 3600 seconds
[00:00:14.256,000] <inf> net_config: Subnet: 255.255.255.0
[00:00:14.256,000] <inf> net_config: Router: 192.168.0.1
[00:00:24.006,000] <dbg> mg_cry_internal_impl.log_access: 192.168.0.143 - "GET /?null HTTP/1.1" 200

[00:00:31.255,000] <dbg> mg_cry_internal_impl.log_access: 192.168.0.143 - "GET /history?null HTTP/1.1" 200

[00:00:41.990,000] <dbg> mg_cry_internal_impl.log_access: 192.168.0.143 - "GET /info?null HTTP/1.1" 200

[00:00:57.027,000] <dbg> mg_cry_internal_impl.log_access: 192.168.0.143 - "GET /history?null HTTP/1.1" 200

[00:01:02.727,000] <dbg> mg_cry_internal_impl.log_access: 192.168.0.143 - "GET /info?null HTTP/1.1" 200
```

The zephyr-documentation for the civetweb-sample states, that a regular 404 error-code is returned then trying to access any unknown URL. It looks like this is not the case and should be fixed. Another approach would be to implement the corresponding resource. Please let me know if i misunderstood the issue.

mglettig, 2021-04-15:
Hi @Nukersson & @gebauerf 

I tested with the latest version of [zephyrproject-rtos/civetweb](https://github.com/zephyrproject-rtos/civetweb) (revision 73fd993ec8e4f922c04a313904f51099aa1cf539 ) and with master of [civetweb/civetweb](https://github.com/civetweb/civetweb) (revision 4aee4ea66135f000f3f7c11f35192328d4466005). All versions behaved the same.

If I return early (before the call to send_no_cache_header) in the function mg_send_http_error_iml the problem does not occur anymore. Will anybody investigate this bug? I think webserver functionality is very essential for Zephyr.

mglettig, 2021-05-03:
Hi @Nukersson & @gebauerf

I noticed that the webserver behaves correctly when compiling with CONFIG_DEBUG=y. In that case an error 404 is returned when I try to access a non existing resource which is what is expected. Therefore I suspect some issues with optimizations.

Regards,
Michael

Nukersson, 2021-05-03:
Thank you all very much for your very big effort in finding the bugs. I think with this knowledge the investigation of the bug can be done much faster. Unfortunately, I have no time for fix now. Sorry. I will take a look as soon as I can. If you are kind, you can always create your own fix. I can provide a review. I hope it is okay for you.

mglettig, 2021-06-02:
@Nukersson Do you think you will find time to investigate this issue in the near future? It would help us a lot to get a more stable version of civetweb.

Nukersson, 2021-06-02:
> @Nukersson Do you think you will find time to investigate this issue in the near future? It would help us a lot to get a more stable version of civetweb.

I ma currently on it. But first need to update Civetweb to latest one.

Nukersson, 2021-06-02:
@mglettig @gebauerf What happens if your explicitly try to access `http://192.0.2.1:8080/favicon.ico`. Cause in my case it works from latest Chrome and Mozilla. And nothing crushes.

Nukersson, 2021-06-02:
> @mglettig @gebauerf What happens if your explicitly try to access `http://192.0.2.1:8080/favicon.ico` and/or setting `CONFIG_NET_LOG=n`? Cause in my case it works from latest Chrome and Mozilla. And nothing crushes. Dou you have possibility to test it on other platform than e70?

Can you please put here your `prj.conf` or is it the same as in `http_server` sample?

closed 2021-08-18 by github-actions[bot]

lucasgcbkhomp, 2022-05-13:
Still relevant, unfortunately. 

I'm not sure how I could help fix it. 
@Nukersson do you have any guidance on how could someone implement the fix? 

Would it be updating civetweb for Zephyr, and then what else?

Thanks

Nukersson, 2022-05-13:
Hey Lucas,

Thank you for your message. See my answers below.

> Still relevant, unfortunately. Acessing through the Browser occurs in the following error due to favicon:
> 
> ```
> [01:45:06.872,558] �[1;31m<err> os: ***** USAGE FAULT *****�[0m
> [01:45:06.872,558] �[1;31m<err> os:   Illegal load of EXC_RETURN into PC�[0m
> [01:45:06.872,589] �[1;31m<err> os: r0/a1:  0xfc23ac00  r1/a2:  0x00000006  r2/a3:  0x00026ac5�[0m
> [01:45:06.872,589] �[1;31m<err> os: r3/a4:  0x00000000 r12/ip:  0x73030080 r14/lr:  0x2000a098�[0m
> [01:45:06.872,589] �[1;31m<err> os:  xpsr:  0x2000a000�[0m
> [01:45:06.872,589] �[1;31m<err> os: Faulting instruction address (r15/pc): 0x00000000�[0m
> [01:45:06.872,589] �[1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0�[0m
> [01:45:06.872,619] �[1;31m<err> os: Current thread: 0x20005bb0 (<log_strdup alloc failed>)�[0m
> [01:45:07.253,570] �[1;31m<err> fatal_error: Resetting system�[0m
> [00:00:00.257,507] �[0m<inf> eth_w5500: W5500 Initialized�[0m
> [00:00:00.259,399] �[0m<inf> net_config: Initializing network�[0m
> [00:00:00.259,460] �[0m<inf> net_config: IPv4 address: 192.168.0.77�[0m
> [00:00:00.259,582] �[0m<inf> main: Set up button at GPIO_0 pin 4
> ```
> 
> I'm not sure how I could help fix it. @Nukersson do you have any guidance on how could someone implement the fix?
> 

Well here is important to know, if ```DEBUG```option is enabled, cause as I mentioned previously not all ```libc```functionality is implemented in Zephyr today and civetweb expects some more libc functions, which will be compiled as stubs with -O3 e.g. DEBUG option enabled. Please check it. IF ```DEBUG```option is disabled Civetweb shouldn't lead to MPU fault actually. Try to compile  [`samples/net/civetweb/websocket_server`](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/civetweb/websocket_server#id4). sample.

> Would it be updating to latest civetweb for Zephyr, and then what else?
> 

I think, we need to test it. I've already proposed updating Civetweb to the latest one (see it [here](https://github.com/zephyrproject-rtos/civetweb/pull/6)). But I am not able to merge it. :) Please try to escalate it. I believe the more people interested in it, the mor chances we have to merge it.

> Thanks

I hope I could help you.