zephyrproject-rtos/zephyr #34222: Commit related to null pointer exception detection causing UART issues

torsteingrindvik, 2021-04-12:
**Describe the bug**

I have been seeing issues related to UART lately.
The current master (`8bee027ec4`) has the following problem:

* UART works right after flashing
* If the board is reset (`nrfjprog -r`), no output is shown

I have only found this to be true in `tests`, not in `samples`.
I found that the bug happens when ZTEST is used.

I have bisected the current master (bad) vs. the `v2.5.0`/`fe7c2ef` tag/commit (good), and found the likely causal commit to be: `1b22f6b8c8`.
This commit ("enable null-pointer exception detection in the tests") introduces `CONFIG_CORTEX_M_DEBUG_NULL_POINTER_EXCEPTION`.
If the below steps to reproduce are repeated but with `CONFIG_CORTEX_M_DEBUG_NULL_POINTER_EXCEPTION_DETECTION_NONE=y` (since we are not allowed to set the former manually), the problem goes away (UART appears even when restarting).

**To Reproduce**

I found a simple test which includes printing stuff to UART.
(This test uses `debug.conf` instead of the implicit `prj.conf`, so it must be explicitly passed)

1. `cd tests/misc/test_build`
2. `west build -b nrf52840dk_nrf52840 --pristine -- -DOVERLAY_CONFIG=debug.conf`
3. Notice the continous UART output
4. `nrfjprog -r`
5. Notice the lack of any UART output

**Expected behavior**
UART output also when board is reset.

**Impact**
Is likely a cause for some of the CI issues in `sdk-nrf` seen the last several weeks (for tests relying on UART).

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gnuarmemb, 9 q4

jfischer-no, 2021-04-12:
related to #33381

galak, 2021-04-13:
I think we should disable this be default.  Users keep running into this and I think their expectation of being able to debug should take precedence.

closed 2021-08-16 by github-actions[bot]