zephyrproject-rtos/zephyr #34225: BBC micro:bit v1.5 LSM303AGR-ACCEL

Liviu112, 2021-04-12:
**Describe the bug**
Hi, I was trying to use the lis2dh example code in combination with BBC micro:bit v1.5 . 
The code compiles without issues and can be flashed, when the board is reseted the following message is received: 

[image]

**What have you tried to diagnose or workaround this issue?**
I have found that, in the definition of the board, the status is set to disable for some reason: 
[image]
(zephyr/boards/arm/bbc_microbit/bbc_microbit.dts)

In order to get the lis2dh to work, it is required to change the status to "okay". After reflashing the lis2dh example the terminal showed the following: 
[image]

***Fix:*** 
Change line 81 and 89 to : ```status = "okay";``` in file ``` zephyr/boards/arm/bbc_microbit/bbc_microbit.dts```

**To Reproduce**
Steps to reproduce the behavior:
1.  ```west build -p auto -b bbc_microbit samples/sensor/lis2dh```
2. ``` west flash```
3. See error

**Impact**
Inability to use the LSM303AGR-ACCEL sensor.

**Environment:**
OS: Linux 20.04
Toolchain: Zephyr version: 2.5.0

FRASTM, 2021-04-12:
LSM303AGR sensor (MAGN part) is mentioned in the issue https://github.com/zephyrproject-rtos/zephyr/issues/34196

Liviu112, 2021-04-12:
That does not yet explain why is disabled in bbc_microbit.dts file :)

jfischer-no, 2021-04-12:
> That does not yet explain why is disabled in bbc_microbit.dts file :)

Because it is for [v1.5 variant of bbc:microbit](https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/bbc_microbit/bbc_microbit.dts#L78), the "default" variant v1.3 has MMA8653FC enabled.
"_See https://tech.microbit.org/hardware/i2c/ for board variants_ "
This is all in the dts file you screenshotted instead of just posting a link to the line.

Liviu112, 2021-04-12:
It didn't was clear to me at the start what was the problem. Now looking back that makes sense, also you do not have to feel offended I tried just to help.

closed 2021-04-12 by Liviu112