zephyrproject-rtos/zephyr #34226: Compile error when building civetweb http_server sample for posix_native

mglettig, 2021-04-12:
**Describe the bug**
Using Zephyr 2.5.0 the build of civetweb http_server sample fails for the board posix_native. From the build logs it looks like an issue with conflicting types (useconds_t, pthread_t, pthread_mutexattr_t, pthread_condattr_t, ...)

**To Reproduce**
`west build -b native_posix samples/net/civetweb/http_server`

**Expected behavior**
The build should succeed.

**Impact**
For testing purposes it would be very convenient to run the webserver on x86. An alternative approach would be to use QEMU. But this doesn't work to to limited SRAM of qemu_cortex_m3.

**Logs and console output**
`root@fce79450bfae:/opt/zephyrproject/zephyr# west build -b native_posix samples/net/civetweb/http_server
-- west build: generating a build system
Including boilerplate (Zephyr base): /opt/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /opt/zephyrproject/zephyr/samples/net/civetweb/http_server
-- Zephyr version: 2.5.0 (/opt/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: native_posix
-- Cache files will be written to: /root/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /opt/zephyrproject/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: /opt/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /opt/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /opt/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h

warning: MINIMAL_LIBC_MALLOC_ARENA_SIZE (defined at lib/libc/Kconfig:102) was assigned the value
'16384' but got the value ''. Check these unsatisfied dependencies: MINIMAL_LIBC_MALLOC (=n),
MINIMAL_LIBC (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE.html
and/or look up MINIMAL_LIBC_MALLOC_ARENA_SIZE in the menuconfig/guiconfig interface. The Application
Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of
the manual might be helpful too.

warning: POSIX_API (defined at lib/posix/Kconfig:12) was assigned the value 'y' but got the value
'n'. Check these unsatisfied dependencies: (!ARCH_POSIX) (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_POSIX_API.html and/or look up
POSIX_API in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

Parsing /opt/zephyrproject/zephyr/Kconfig
Loaded configuration '/opt/zephyrproject/zephyr/boards/posix/native_posix/native_posix_defconfig'
Merged configuration '/opt/zephyrproject/zephyr/samples/net/civetweb/http_server/prj.conf'
Configuration saved to '/opt/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/opt/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
^Croot@fce79450bfae:/opt/zephyrproject/zephyr# west build -b native_posix samples/net/civetweb/my_http_server
-- west build: generating a build system
Including boilerplate (Zephyr base): /opt/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /opt/zephyrproject/zephyr/samples/net/civetweb/my_http_server
-- Zephyr version: 2.5.0 (/opt/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: native_posix
-- Cache files will be written to: /root/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /opt/zephyrproject/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: /opt/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /opt/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /opt/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /opt/zephyrproject/zephyr/Kconfig
Loaded configuration '/opt/zephyrproject/zephyr/build/zephyr/.config'
No change to configuration in '/opt/zephyrproject/zephyr/build/zephyr/.config'
No change to Kconfig header in '/opt/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
[... 234 lines cut ...]
      |                ^~~~
      |                fread
/opt/zephyrproject/modules/lib/civetweb/src/civetweb.c: In function ‘connect_socket’:
/opt/zephyrproject/modules/lib/civetweb/src/civetweb.c:853:25: warning: implicit declaration of function ‘close’; did you mean ‘pclose’? [-Wimplicit-function-declaration]
  853 | #define closesocket(a) (close(a))
      |                         ^~~~~
/opt/zephyrproject/modules/lib/civetweb/src/civetweb.c:8942:3: note: in expansion of macro ‘closesocket’
 8942 |   closesocket(*sock);
      |   ^~~~~~~~~~~
[106/132] Building C object zephyr/kernel/CMakeFiles/kernel.dir/idle.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /opt/zephyrproject/zephyr/build`

If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 (running in Docker)
 - Toolchain (e.g Zephyr SDK, gcc-arm-none-eabi-8-2019-q3-update-linux)
 - Zephyr 2.5.0

**Additional context**
Add any other context about the problem here.

mglettig, 2021-04-13:
Could this be related to the following topic #13054? Is it currently not possible to use POSIX subsytem for ARCH_POSIX?

Nukersson, 2021-04-13:
Hello and thank you for your bug description.  I think, that posix support for civetweb need to be rethinked. 
The first guys, who ported civetweb, were cool dudes from Anrmicro. @PiotrZierhoffer can you provide more info on posix support within civetweb?

PiotrZierhoffer, 2021-04-14:
I must say it's been a while, and the civetweb module got pulled in during a very heated discussion on the POSIX APIs. I'm not very familiar with the current state of affairs in this topic.

If what @mglettig mentions still holds, then it is definitely a relevant problem. And it does seem so from your log:

```
warning: POSIX_API (defined at lib/posix/Kconfig:12) was assigned the value 'y' but got the value
'n'. Check these unsatisfied dependencies: (!ARCH_POSIX) (=n).
```

So Civetweb relies on the POSIX_API, and I'm not aware if there is any "switch" that you can use to turn to native POSIX. If there's a "standard way of handling this", then it should not be very difficult to apply to Civetweb, the port is not very complex. I would assume that we'd need to adjust `ifdef`s here and there: from `#if ZEPHYR` to `#if ZEPHYR && !ARCH_POSIX`, maybe?

galak, 2021-05-11:
@Nukersson any updates on this?

Nukersson, 2021-05-11:
> @Nukersson any updates on this?

Some progress can be seen here: #34438

But I am not actively involved onto it. Just monitoring and trying to help, where I can. At the end I can do the tests on my HW of course, before the merge can be done.
It looks currently, that multiple posix functions needed by Civetweb must be implemented on Zephyr posix api. But not all the functionality is possible, since there are some constraints from HW side.

stephanosio, 2022-02-18:
Re-opening since this is still a valid issue.

cfriedt, 2022-04-11:
This is probably fixed by #43987

mbolivar-nordic, 2022-06-22:
"Fixed" by the removal of civetweb. See  #46746 for more details. Thanks for taking the time to file an issue and sorry for the inconvenience.

PR #46746 (2022-06-22): civetweb: remove obsolete code
Retrying https://github.com/zephyrproject-rtos/zephyr/pull/45988 now that we have released v3.1 and nobody has stepped up to maintain civetweb. (I can't reopen the old PR due to GitHub restrictions.)

This code has gone unmaintained and bugs continue to be reported
against it. We do not have the resources as a project to maintain this
in "odd fixes" mode, and nobody has stepped up to maintain it [1], so
sadly this must be removed for now.

If anyone would like to see civetweb supported in upstream Zephyr
again, they are welcome to add it back, as long as they pro [... cut]
commit f49d11d3 (2022-06-22): civetweb: remove obsolete code

This code has gone unmaintained and bugs continue to be reported
against it. We do not have the resources as a project to maintain this
in "odd fixes" mode, and nobody has stepped up to maintain it [1], so
sadly this must be removed for now.

If anyone would like to see civetweb supported in upstream Zephyr
again, they are welcome to add it back, as long as they promise to
maintain it going forward.

Many thanks to everyone who has contributed to civetweb support in
Zephyr while it was here. So long and thanks for all the fish.

Fixes: #45807
Fixes: #43910
Fixes [... cut]