zephyrproject-rtos/zephyr #34227: Use compile time resoved device bindings in flash map, when possible

de-nordic, 2021-04-12:
**Is your enhancement proposal related to a problem? Please describe.**
Currently the flash map API uses `device_get_binding` at run-time to obtain pointer to the `struct device` type object for the flash device it is going to operate on.
This means, that with current source code each flash operation requires obtaining binding prior to any work done.

**Describe the solution you'd like**
With new DT macros that allow to resolve pointer to device object at compile time (since https://github.com/zephyrproject-rtos/zephyr/pull/30196) the pointers can be obtained during build-time but require changes to `flash_area` structure: new fields have to be added to hold the pointer.

**Describe alternatives you've considered**
Not doing the change.

**Additional context**
The new structure element should be added but the possibility of using the old `device_get_binding` should be left as an option, as we currently support custom flash maps that may be generated outside of our build system and linked into the compiled binary by user with custom linker scripts. For that reason the user should be allowed to revert to using `device_get_binding` till ready to move to the compile time resolved objects.
The new field that will hold the pointer for  `device` object should be left optional to not increase size of structure if the device pointer is not used.
The usage of `device_get_binding` within flash map should be considered for deprecation.

de-nordic, 2021-04-12:
Attempting to resolve here: https://github.com/zephyrproject-rtos/zephyr/pull/33346

mbolivar-nordic, 2021-04-12:
Here is a summary of my concerns as discussed in #33346.

- The `struct flash_area` members were copy/pasted straight from the mynewt operating system's HAL when they were introduced in 25269fb0ede953f70d0e4e22d220638269291abd. Every [port of MCUboot to a specific RTOS must define this structure](https://github.com/mcu-tools/mcuboot/blob/master/docs/PORTING.md#flash-map) in this way for purely historical reasons, because that bootloader was originally written for mynewt and nobody has changed the design. However, multiple `flash_area` fields have no logical place in Zephyr's flash map API.
- This is simply bad design. `struct flash_area` should have been left in the zephyr port of mcuboot. It does not belong in zephyr/include/storage/flash_map.h.
- The PR in its current state adds Zephyr-specific members onto a structure that is really specific to the MCUboot application. This bloats the structure unnecessarily and makes a bad API even worse.

The PR is being presented as a small change that we should just merge in order to get compile-time device pointers into the flash map API. However, flash map is a [stable API](https://docs.zephyrproject.org/latest/reference/overview.html) and we can't back out of changes to it easily, so I don't agree with adding device pointers in a hacky way like this.

1. take the opportunity to go through the stable API change process to define a flash map API that really works for zephyr, while we still have time before 2.6
2. put `struct flash_area` back into the mcuboot/boot/zephyr where it belongs

Laczen, 2021-04-12:
@mbolivar-nordic, correct me if I'm wrong, but doesn't flash_map api provide 2 functionalities:

a. A flash MAP that describes the flash partitions in a run-time structure,
b. Write, read and erase instructions for each of the areas in the flash MAP

Functionality a. seems only needed for a bootloader, and should be part of bootloader code,
Functionality b. is what can be found in other toolchains as "flash_partition" access, and is very relevant/missing in zephyr. The present name flash_map is not fit for "flash_partition" access.

pabigot, 2021-04-12:
> Functionality b. is what can be found in other toolchains as "flash_partition" access, and is very relevant/missing in zephyr. The present name flash_map is not fit for "flash_partition" access.

It is that functionality that made it the API of choice for littlefs.

de-nordic, 2021-04-13:
The advantage of the flash map API is that, using only the partition identifier, it allows to gain access to some partition on some device, hiding the details from the user and taking care of proper offsets and sizes.
The user defines everything in DTS and then just uses the identifier to collect object that and use API on that object.

> * The `struct flash_area` members were copy/pasted straight from the mynewt operating system's HAL when they were introduced in [25269fb](https://github.com/zephyrproject-rtos/zephyr/commit/25269fb0ede953f70d0e4e22d220638269291abd). Every [port of MCUboot to a specific RTOS must define this structure](https://github.com/mcu-tools/mcuboot/blob/master/docs/PORTING.md#flash-map) in this way for purely historical reasons, because that bootloader was originally written for mynewt and nobody has changed the design. However, multiple `flash_area` fields have no logical place in Zephyr's flash map API.
> 
> * This is simply bad design. `struct flash_area` should have been left in the zephyr port of mcuboot. It does not belong in zephyr/include/storage/flash_map.h.

Yes, this is bad design on part of mcuboot although we have agreed to use it when we have accepted the mcuboot, probably no one cared at that time.
We have been using it for quite some time and gathered experience, now we should probably sit down and work with mcuboot to improve the interface between both entities.

Per the design: my opinion is that the mcuboot should only use a partition identifier to obtain the flash area object from the system and then use the system API to operate on that object without knowing the implementation details.
We should work with the mcuboot project to define (improve) the common API for the `flash_area` object access; the Zephyr, and other systems, should export such API leaving internal details of `flash_area` object to itself, allowing manipulation of such object only via the API.
This way both, the mcuboot and Zephyr (and the others), would be free from consequences of changes to `flash_area` structure as long as the interface and its operation are not affected.

Finally: the problem with reworking the `flash_area` is that we allow custom flash map definitions that are defined by users, with knowledge of, and access to, internal structure of `flash_area`. That is why I am reluctant to just drop the current definition.

mcuboot and other Zephyr applications does not have to share the same definition of the `flash_area` structure so maybe that is the solution: to use the old one for mcuboot compilation (and users that will take time to adapt), till we have common API agreed with  mcuboot, and start working on the new (experimental) one in Zephyr.
The new, experimental,  one could be following the idea of exporting API without internal details, as mentioned above, in which case we would be able to prototype our proposal as the parallel `flash_area` API and when we got into agreement with mcuboot, we would have the implementation ready and could easily drop the current solution in favor prototyped one . Such approach also requires careful consideration as I would like to avoid another `bad design` rework too soon. 

So, for some time the current solution would be available anyway.

carlescufi, 2021-04-13:
API meeting 2021.04.13

- Agree to remove MCUboot-specific structure members from `flash_area`
- Investigate whether it is possible to entirely deprecate the first 32-bits of `flash_area` without compromising mcuboot/mcumgr
- Depending on the outcome, revisit in order to decide

de-nordic, 2021-04-19:
> 
>     * Investigate the use of the structure by mcuboot/mcumgr
> 
OK. With mcumgr it should be easier as it does not seem to use `area_id` only to identify the current transfer, so it keeps it in global state variable; there is only once condition regarding that variable which is that `-1` is "bad area_id" or rather no area id selected.
I guess that we could just define that field to be `const struct flash_area *` for Zephyr and the original `int` for non-zephyr systems.
Fortunately the internals of `flash_area` seems to be accessed only within backend code, so we should be able to figure that out without affecting the library common code.

In case of mcuboot it is more complicated as there is often pair of calls:
```
fa_id = flash_area_id_from_multi_image_slot(image_index, slot);
rc = flash_area_open(fa_id, &BOOT_IMG_AREA(state, slot));
```
So the flash area id for the `flash_are_open` is not directly generated from DTS (although the `flash_area_id_from_multi_image_slot` uses DTS info for translation.
There are some direct access to `fa_id` field of `flash_area`, for example in `swap_erase_trailer_sectors`, but this can be replaced with `flash_area_id_from_multi_image_slot` and the other instance is in log message which can be substituted too.
The problem is with some functions (like `boot_enc_load`, `boot_enc_valid` , etc) access the `fa_id` field with call to `flash_area_id_from_multi_image_slot` to figure out whether the given area is OK, and this is in common code so this part requires rework.

Regarding the `fa_device_id` field: it is used to figure out whether the area is placed on the device we can boot from; 0 in our case means boot from internal/soc flash, luckily this is default value for initialization as we somehow forgotten to initialize it in flash map...
There is one instance where it is used in Zephyr specific code directly, `boot_go` of single_loader.c, where it is rewritten to `boot_rsp.br_flash_dev_id`. In common code it is twice accessed, as `flash_area->fa_device_id`, in `context_boot_go` and rewritten to `boot_rsp`.

In execution path of boot there is one of the above functions called, depending on compilation options, and then the  `boot_go` will be called and will use the `boot_rsp->br_flash_dev_id` copy of the `fa_device_id` with Zephyr specific  implementation of `flash_device_base`; the `flash_device_base` only checks if the device id matches the one defined via `SOC_FLASH_0_ID` (which is, luckily, 0) and then returns `CONFIG_FLASH_BASE_ADDRESS`.

The Zephyr specific implementations can be probably changed to work without the `fa_device_id`, with `flash_device_base`, probably discarded completely. The problem is with `context_boot_go`, from common code, Zephyr execution of boot process does not seem to need it.

If we would hide implementation details of the `flash_area` from the mcuboot and just provide API to access fields, that would allow as to remove all the dependencies on any internal fields. 

> 
>     * Investigate whether it is possible to entirely deprecate the first 32-bits of `flash_area` without compromising mcuboot/mcumgr
Yes, easier with mcumgr. With mcuboot we should work to hide `flash_area` implementation details with mcuboot.

Laczen, 2021-04-19:
The proposed method seems like a very complex solution to the problem. I am at the moment working on a flash partition api that does (IMO) everything zephyr needs, mainly a partition write, read and erase, but also a partition get sectors, ... 
The definition of a flash partition would be a pointer to a const struct **cfg** (only containing the device) and a pointer to a struct **state** (empty at the moment, but prepared for future use, e.g. To temporary disable access to the partition), the offset and size. The offset and size would be non-const to allow for resizing partitions.
On top of the flash partition API the flash area API could be easily rewritten, without breaking anything. In the long run it would only be mcuboot that uses the flash area API.

de-nordic, 2021-04-19:
> 
> The proposed method seems like a very complex solution to the problem. I am at the moment working on a flash partition api that does (IMO) everything zephyr needs, mainly a partition write, read and erase, but also a partition get sectors, ...

It is not. It is intended to allow having one `flash_area` definition and avoid translations and additional API over/under the one we have already. The mcuboot will have to be changed and mcumgr will have to be altered.

> The definition of a flash partition would be a pointer to a const struct **cfg** (only containing the device) and a pointer to a struct **state** (empty at the moment, but prepared for future use, [...]

So basically it would be what @mbolivar has proposed, and has been so far accepted, that we should have `const struct device *` + offset + size and that is all that is needed to serve flash areas defined (as partitions) within DTS.

>
> On top of the flash partition API the flash area API could be easily rewritten, without breaking anything. In the long run it would only be mcuboot that uses the flash area API.

I think we should avoid unnecessary layers. The flash area is the flash partition API, I doubt that another layer is necessary to fix problems with current implementation. This might be the right moment to figure out what mcuboot really needs, what mcumgr really needs and what is internally needed, and do the rework accordingly.

Laczen, 2021-04-19:
> It is not. It is intended to allow having one `flash_area` definition and avoid translations and additional API over/under the one we have already. The mcuboot will have to be changed and mcumgr [...]

Your future `flash_area` definition would be the same as my proposed `flash_partition`, it is just a name and by making it different the name `flash_area` could be maintained in mcuboot.
 
> So basically it would be what @mbolivar has proposed, and has been so far accepted, that we should have `const struct device *` + offset + size and that is all that is needed to serve flash areas [...]

I would take the opportunity to think about this a little more and to create something that could be easily extended. At the moment there is nothing more than `const struct device *`, `offset` (is this const or not?), `size` (is this const or not?), but how can we add other things without changing the structure ? Why do devices have a cfg part and a data part, would something similar be good ?

> I think we should avoid unnecessary layers. The flash area is the flash partition API, I doubt that another layer is necessary to fix problems with current implementation. This might be the right [...]

The proposal is not an additional layer, it is "extracting" the `flash_area` API and renaming it `flash_partition`, and to keep the `flash_map/flash_area` as a glue API for mcuboot/mcumgr. The renaming makes the transition easy and smooth.

Laczen, 2021-04-20:
@de-nordic, something that might also be usefull: allow the dts to specify a `erase_block_size` for the partition. This has to be at least equal to the largest flash page size of the partition, but could also be a multiple of this. Creating this would allow sector size in nvs/fcb to be specified from the dts, and also the block size in littlefs (@pabigot). A lot of code in trying to find the largest page size could be removed by this compile time definition.

JordanYates, 2022-07-22:
Is this resolved by #47193?

de-nordic, 2022-07-26:
> Is this resolved by #47193?

Yes.

closed 2022-07-26 by de-nordic