zephyrproject-rtos/zephyr #34229: C++ Exception Support in qemu_riscv32 emulation

bob2oneil, 2021-04-12:
I have a test application written in C++ using the qemu_riscv32 emulation.  It simply tosses and exception with a handler, which never gets called.   The real target is a RISC-V processor.  My KConfig file includes what I believe are the necessary dependencies as follows:

CONFIG_LIB_CPLUSPLUS=y
CONFIG_NEWLIB_LIBC=y
CONFIG_CPLUSPLUS=y
CONFIG_EXCEPTIONS=y
CONFIG_STD_CPP2A=y

The build script is as follows:  "west build -p auto -b qemu_riscv32 samples/exception"

To reproduce, run the standard "west build -t run" command for QEMU emulation.

This produces the following output (shows no exception caught):

-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: riscv32
*** Booting Zephyr OS build zephyr-v2.5.0-2076-g930893fc5678  ***
Exception test! qemu_riscv32
Throwing integer exception 20.
qemu-system-riscv32: terminating on signal 2
ninja: build stopped: interrupted by user.

The source code to the simple tester is as follows as main.cpp, the project's only file.

------------

/**
 * @brief main
 */
void main(void)
{
    // NOTE: may require additional OS module

    printk("Exception test! %s\n", CONFIG_BOARD);

    try
    {
        printk("Throwing integer exception 20.\n");
        throw 20;
    }
    catch (int e)
    {
        printk("An integer exception occurred and was handled.\n");
    }
    catch (bool e)
    {
        printk("A bool exception occurred and was handled.\n");
    }
    catch (void* e)
    {
        printk("A C++ void exception occurred and was handled.\n");
    }

    printk("Program exit.\n");
}

[CMakeLists.txt](https://github.com/zephyrproject-rtos/zephyr/files/6298964/CMakeLists.txt)

galak, 2021-04-13:
This looks to be a duplicate or related to https://github.com/zephyrproject-rtos/zephyr/issues/32448

galak, 2021-04-14:
Going to close this and lets use #32448 for discussion.

closed 2021-04-14 by galak