zephyrproject-rtos/zephyr #34231: uzlib (decompression library)

Delsian, 2021-04-12:
## Origin

https://github.com/pfalcon/uzlib

Radically unbloated DEFLATE/zlib/gzip compression/decompression library. Can decompress any gzip/zlib data, and offers simplified compressor which produces gzip-compatible output, while requiring much less resources (and providing less compression ratio of course).

## Mode of integration

@pfalcon Please, can you add your library as additional module or subsystem? We urgently need any decompression library to reduce size of data bundles sent from host to Zephyr-powered device.
Thanks in advance!

carlescufi, 2021-04-14:
@Delsian could you please fill out the whole template instead of just a small part of it?

Delsian, 2021-04-14:
I hope @pfalcon, as one of Zephyr code owners, can make better decisions about his library integration?

pfalcon, 2021-04-15:
@Delsian: Glad that you find uzlib useful! uzlib is a simple, minimalist, os-, hardware-, and vendor- independent zlib compression/decompression library, which can easily integrate with Zephyr or any other system. uzlib needs more work in the areas of testsuite and finishing the API. I definitely would be interested to work in those areas (otherwise the work on uzlib was done in my own free time, which I of course have too little, so many areas, like the mentioned above, need more work.)

Otherwise, I'm just a Zephyr contributor (as part of my day job), and don't decide what to work on. What it means to integrate uzlib with Zephyr would also need to be more precisely defined.

Delsian, 2021-04-24:
I'll close this request as it requires some changes in UZLIB.
Currently I've done including this library in my project, I can share implementation details by request.
Thank you!

closed 2021-04-24 by Delsian

mikevoyt, 2021-07-15:
Hi @delsian, would you mind sharing your implementation details?  I'm trying to include uzlib in one of my Zephyr projects, and I'm struggling with the CMakeLists.txt implementation.  I suspect I'll need to change the uzlib `makefile` as well but I'm not exactly sure how, any pointers would be appreciated!

Delsian, 2021-07-20:
@mikevoyt you can check my implementation here https://github.com/Delsian/uzlib/tree/zephyr
My code contains two additional functions:
```
void uzlib_dest_putc(uint8_t c) {
    if(d_context.dest >= uct->flash_write_ptr+FLASH_BLOCK_SIZE) {
        //LOG_DBG("Block ready: %p", uct->flash_write_ptr);
        flash_area_write(uct->upload_area, (off_t) uct->flash_write_ptr,
                    uct->unpack_buf, FLASH_BLOCK_SIZE);		
        uct->flash_write_ptr += FLASH_BLOCK_SIZE;
    }

    uct->unpack_buf[(uint32_t)(d_context.dest - uct->flash_write_ptr)] = c;
}

uint8_t uzlib_dest_getc(const uint8_t *addr) {
    if(addr < uct->flash_write_ptr) {
        uint8_t rv;
        flash_area_read(uct->upload_area, (off_t) addr, &rv, 1);
        return rv;
    }
    return uct->unpack_buf[(uint32_t)(addr - uct->flash_write_ptr)];
}
```

mikevoyt, 2021-07-20:
Cool thanks for the pointer @Delsian !

Delsian, 2021-07-21:
@mikevoyt On another side, the current Zephyr version contains module lz4. Probably, better to use the standard variant.

mikevoyt, 2021-07-21:
> @mikevoyt On another side, the current Zephyr version contains module lz4. Probably, better to use the standard variant.

Ah cool, I didn't realize that - thanks for the tip!

krishnasubramani, 2023-10-31:
@Delsian Is the following page available?

https://github.com/Delsian/uzlib/tree/zephyr

Delsian, 2023-11-22:
> @Delsian Is the following page available?
> 
> https://github.com/Delsian/uzlib/tree/zephyr

No, I've completely moved to lz4