zephyrproject-rtos/zephyr #34233: OpenThread build issues

lodup29, 2021-04-12:
As previously reported [here](https://github.com/zephyrproject-rtos/zephyr/issues/29567#issuecomment-717832182), a clean build appears to required for projects integrating OpenThread. Moreover, some parameters (ex. : _CONFIG_OPENTHREAD_NETWORK_NAME_) just never make it into the final build, clean or not.

carlescufi, 2021-05-11:
@rlubos ping on this one.

rlubos, 2021-05-12:
From what I've seen, the OpenThread project does not catch up with the Kconfig changes done after the project is generated. @tejlmand Is there any Kconfig dependency we could add to the OpenThread build stage?

rlubos, 2021-05-13:
@lodup29 I've submitted a fix for the OpenThread build system behaviour(https://github.com/zephyrproject-rtos/zephyr/pull/35252), it should keep up with the Kconfig changes now.

> Moreover, some parameters (ex. : CONFIG_OPENTHREAD_NETWORK_NAME) just never make it into the final build, clean or not.

Parameters like `CONFIG_OPENTHREAD_NETWORK_NAME` are not intended for OpenThread build, they're processed by the Zephyr part of the OpenThread integration. Generall, options that affect the OpenThread build are kept in `openthread/Kconfig.features` (or at least that's what we want to achieve).

> Also, the parameter CONFIG_OPENTHREAD_CUSTOM_PARAMETERS must be defined on top of already equivalent parameters such as CONFIG_OPENTHREAD_JOINER. For example:

This one I'm not able to reproduce. Even w/o this fix, if you set `CONFIG_OPENTHREAD_JOINER` in the prj.conf file, it will be recognized by OpenThread and will enable the joiner feature in the OpenThread stack. No need to use `CONFIG_OPENTHREAD_CUSTOM_PARAMETERS ` (this should only be used for special cases, OpenThread options that are not covered with Kconfig. Actually I'm not sure if there're any left).

tejlmand, 2021-05-18:
> @tejlmand Is there any Kconfig dependency we could add to the OpenThread build stage?

Not sure that would be needed. There is a global CMake dependency on `.config` so if any Kconfig option is changed, for example through `menuconfig` then CMake will rerun on next rebuild, and this in itself should trigger a rerun of the OpenThread CMake code, which hopefully inside OpenThread triggers correct rebuild of OpenThread source files in case OpenThread build parameters has changed.

PR #35252 (2021-05-25): manifest: Update OpenThread revision
Update OpenThread revision, to introduce fixes for the OpenThread
configuration cache variables behaviour, making them up-to-date
with the actual Kconfig configuration.

Fixes #34233
commit 33043dca (2021-05-25): manifest: Update OpenThread revision

Update OpenThread revision, to introduce fixes for the OpenThread
configuration cache variables behaviour, making them up-to-date
with the actual Kconfig configuration.

Fixes #34233
closed 2021-05-25 by galak