zephyrproject-rtos/zephyr #34234: UART NS16550 Driver Issue

lipengit, 2021-04-13:
**Describe the bug**
When the CPU reads the RDR register during initialization, the UART reports a buffer under flow error. 

**To Reproduce**
This is a new board that is still under development. But for any boards that uses this device driver, it should be reproduced. 

**Expected behavior**
During the UART initialization, when the CPU reads the RDR register to clear the port, the UART should NOT reports a buffer under flow error. 

**Impact**
We have fixed this issue with the following minor changes to "uart_ns16550.c". Before clear the port, we just need to set DLAB to 1. After clear the port, we reset DLAB to 0:
`
diff --git a/drivers/serial/uart_ns16550.c b/drivers/serial/uart_ns16550.c
index 19b4dd6430..5777f08095 100644
--- a/drivers/serial/uart_ns16550.c
+++ b/drivers/serial/uart_ns16550.c
@@ -339,7 +339,7 @@ static int uart_ns16550_configure(const struct device *dev,
 {
 	struct uart_ns16550_dev_data_t * const dev_data = DEV_DATA(dev);
 	const struct uart_ns16550_device_config * const dev_cfg = DEV_CFG(dev);
-
+    uint8_t lcr_cache;
 	uint8_t mdc = 0U;
 
 	/* temp for return value if error occurs in this locked region */
@@ -466,7 +466,10 @@ static int uart_ns16550_configure(const struct device *dev,
 		);
 
 	/* clear the port */
+	lcr_cache = INBYTE(LCR(dev));
+	OUTBYTE(LCR(dev), LCR_DLAB | lcr_cache);
 	INBYTE(RDR(dev));
+	OUTBYTE(LCR(dev), lcr_cache);	
 
 	/* disable interrupts  */
 	OUTBYTE(IER(dev), 0x00);
`

**Logs and console output**
UART_0 Error[Buffer Underflow]   RBR doesn't contains the valid data.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Xtensa XCC
 - Commit SHA or Version used: v2.5.0, fe7c2efca800a0cf1bccf23aefe08b3c4beb88bf 

**Additional context**
We have fixed this issue. Please help merge the above patch to the main branch. Thanks.

galak, 2021-04-13:
Fix in #34235

jenmwms, 2021-04-14:
Can the title be slightly modified to describe the issue? Maybe add 'underflow issue' or 'does not fully clear the port'?

PR #34235 (2021-04-15): Fix the uart_ns16550.c buffer underflow issue. 
driver: serial: uart_ns16550: fixed the buffer underflow issue when clearing the port. 

To clear the port by reading the RDR register, the DLAB bit should be set, otherwise, the Synopsys UART IP reports a buffer underflow error.

Fixes #34234
closed 2021-04-15 by ioannisg