zephyrproject-rtos/zephyr #34249: Unable to initialize on STM32F103RE + Quectel EC21 using BG9x driver

ycsin, 2021-04-13:
**Describe the bug**
I'm trying the Quectel BG9x modem driver on my STM32F1 custom board with Quectel EC21 as their initialization should be the same and the offloaded TCP AT commands are the same.

I compiled the modem on `basic/blinky` example with these

`prj.conf`

```
CONFIG_GPIO=y
CONFIG_LOG=y
CONFIG_KERNEL_LOG_LEVEL_DBG=y
CONFIG_KERNEL_LOG_LEVEL_DBG=y
CONFIG_MODEM_LOG_LEVEL_DBG=y
```

`def_config`
```
# Modem
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_NETWORKING=y
CONFIG_NET_SOCKETS=y
CONFIG_MODEM=y
CONFIG_MODEM_QUECTEL_BG9X=y
```

`.dts`:
```
    /* QUECTEL EC21 */
    quectel_ec21: quectel_ec21 {
        compatible = "quectel,bg9x";
        label = "QUECTEL EC21";
        mdm-power-gpios = <&gpiob 1 GPIO_ACTIVE_LOW>;
        mdm-reset-gpios = <&gpiob 0 GPIO_ACTIVE_LOW>;
        mdm-dtr-gpios = <&gpioa 15 GPIO_ACTIVE_LOW>;
        status = "okay";
    };
```

The driver initialized fine, however it seems to be stucked in a bootloop:

Log:
```
[00:00:00.000,000] <dbg> os.setup_thread_stack: stack 0x20003d60 for thread 0x20001ba8: obj_size=1024 buf_start=0x20003d60  buf_size 1024 stack_ptr=0x20004160
[00:00:00.000,000] <dbg> os.setup_thread_stack: stack 0x20004160 for thread 0x20001b30: obj_size=256 buf_start=0x20004160  buf_size 256 stack_ptr=0x20004260
[00:00:00.000,000] <dbg> os.setup_thread_stack: stack 0x200023c8 for thread 0x20000af0: obj_size=768 buf_start=0x200023c8  buf_size 768 stack_ptr=0x200026c8
[00:00:00.000,000] <dbg> os.setup_thread_stack: stack 0x20004a60 for thread 0x20001c20: obj_size=1024 buf_start=0x20004a60  buf_size 1024 stack_ptr=0x20004e60
[00:00:02.000,000] <dbg> os.setup_thread_stack: stack 0x20002ad0 for thread 0x20001808: obj_size=2048 buf_start=0x20002ad0  buf_size 2048 stack_ptr=0x200032d0
[00:00:02.000,000] <dbg> os.setup_thread_stack: stack 0x200026c8 for thread 0x20001790: obj_size=1032 buf_start=0x200026c8  buf_size 1032 stack_ptr=0x20002ad0
[00:00:02.000,000] <inf> modem_quectel_bg9x: Setting Modem Pins
[00:00:04.750,000] <inf> modem_quectel_bg9x: ... Done!
[00:00:04.750,000] <inf> modem_quectel_bg9x: Waiting for modem to respond
[00:00:13.729,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [RDY] (len:3)
[00:00:13.747,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:13.800,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:13.883,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:13.953,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [] (len:7)
[00:00:13.953,000] <inf> modem_quectel_bg9x: Manufacturer: Quectel
[00:00:13.954,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:14.006,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [] (len:4)
[00:00:14.006,000] <inf> modem_quectel_bg9x: Model: EC21
[00:00:14.007,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:14.060,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [] (len:16)
[00:00:14.060,000] <inf> modem_quectel_bg9x: Revision: EC21EFAR06A01M4G
[00:00:14.060,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:14.131,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [] (len:15)
[00:00:14.131,000] <inf> modem_quectel_bg9x: IMEI: 867962045083204
[00:00:14.132,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:14.205,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:14.207,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [+CSQ: ] (len:11)
[00:00:14.207,000] <inf> modem_quectel_bg9x: RSSI: -1000
[00:00:14.208,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:16.213,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [+CSQ: ] (len:11)
[00:00:16.213,000] <inf> modem_quectel_bg9x: RSSI: -1000
[00:00:16.214,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:18.217,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [+CSQ: ] (len:11)
[00:00:18.217,000] <inf> modem_quectel_bg9x: RSSI: -1000
[00:00:18.217,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:20.220,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [+CSQ: ] (len:11)
[00:00:20.220,000] <inf> modem_quectel_bg9x: RSSI: -57
[00:00:20.220,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:00.000,000] <dbg> os.setup_thread_stack: stack 0x20003d60 for thread 0x20001ba8: obj_size=1024 buf_start=0x20003d60  buf_size 1024 stack_ptr=0x20004160
[00:00:00.000,000] <dbg> os.setup_thread_stack: stack 0x20004160 for thread 0x20001b30: obj_size=256 buf_start=0x20004160  buf_size 256 stack_ptr=0x20004260
[00:00:00.000,000] <dbg> os.setup_thread_stack: stack 0x200023c8 for thread 0x20000af0: obj_size=768 buf_start=0x200023c8  buf_size 768 stack_ptr=0x200026c8
[00:00:00.000,000] <dbg> os.setup_thread_stack: stack 0x20004a60 for thread 0x20001c20: obj_size=1024 buf_start=0x20004a60  buf_size 1024 stack_ptr=0x20004e60
[00:00:02.000,000] <dbg> os.setup_thread_stack: stack 0x20002ad0 for thread 0x20001808: obj_size=2048 buf_start=0x20002ad0  buf_size 2048 stack_ptr=0x200032d0
[00:00:02.000,000] <dbg> os.setup_thread_stack: stack 0x200026c8 for thread 0x20001790: obj_size=1032 buf_start=0x200026c8  buf_size 1032 stack_ptr=0x20002ad0
```

After this it will repeat again at the 
```
[00:00:02.000,000] <inf> modem_quectel_bg9x: Setting Modem Pins
```
continuously. If I add a 5 seconds delay after [bg9x:L1070](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/modem/quectel-bg9x.c#L1070), then it will be able to print an additional line:
```
[00:00:17.080,000] <inf> modem_quectel_bg9x: Network is ready.
```

I then debugged this and found that it resets at around

[image]

or

[image]

But then I don't know what to do next to further trace this issue down. No errors printed in the log.

**To Reproduce**
Steps to reproduce the behavior:
1. Build `samples/blinky` example with the `dts`, `def_config` and `prj.conf` on a STM32F103RE board with Quectel EC21
2. See error

**Expected behavior**
I'm expecting that LED to be blinking on `samples/blinky`

**Impact**
This is a showstopper, I'm working on a IoT device that depends on modem to send data.

**Logs and console output**
See above

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: gnuarmemb
 - Just did `west init zephyr_project` and `west update` today, should be the latest version

**Additional context**
N/A

galak, 2021-04-13:
@bwasim can you help with this.

bwasim, 2021-04-14:
The stack trace in your screenshot shows Zephyr Native Networking function calls, which isn't needed when you're using an offloaded driver like quectel-bg9x. So as a first pass, can you please update your configuration to include these options:
```
# Quectel BG9x modem
CONFIG_MODEM=y
#CONFIG_MODEM_SHELL=y
CONFIG_MODEM_QUECTEL_BG9X=y

# Networking (required for modem)
CONFIG_NETWORKING=y
CONFIG_NET_NATIVE=n
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_OFFLOAD=y
```
This doesn't directly fix whatever problem you're seeing, but tries to bypass it as you don't need native networking stack. Should help resolve the problem.

ycsin, 2021-04-14:
> # Quectel BG9x modem
> CONFIG_MODEM=y
> #CONFIG_MODEM_SHELL=y
> CONFIG_MODEM_QUECTEL_BG9X=y
> 
> # Networking (required for modem)
> CONFIG_NETWORKING=y
> CONFIG_NET_NATIVE=n
> CONFIG_NET_IPV4=y
> CONFIG_NET_IPV6=n
> CONFIG_NET_SOCKETS=y
> CONFIG_NET_SOCKETS_OFFLOAD=y

I'm building this now

closed 2021-04-14 by ycsin

bwasim, 2021-04-14:
Great, I would like to get some feedback on how well this driver works with the EC21 modem. When I initially wrote it, I primarily looked only at the BG95 guide, but found that the BG96 TCP commands are the same. Hence named the driver, "quectel-bg9x". 

I haven't looked at the EC21 guides, but if it works well (even with minor modifications), we may want to incorporate those changes in this driver and give it a more generic name..

ycsin, 2021-04-14:
Hey @bwasim , I'm still working on the driver, been stuck for quite a few hours.

I've checked the TCP/IP AT commands of BG96 & EC21 and they are identical. Then I try to run the `mqtt_publisher` example, but it doesn't work.

[image]

I traced a bit and found that it connected with my MQTT broker, but disconnected immediately after that. This is the reading from of the AT commands going into the modem

[image]

```
AT+QIOPEN=1,0,"TCP","server_ip_addr",1883,0,0
AT+QISEND=0,30
MQTT<zephyr_publisher
AT+QICLOSE=0
```

I can see that the modem received something from the server, likely the CONNACK from the MQTT broker:
```
+QIURC: "recv",0
```

But somehow `offload_read` and `offload_recvfrom` never get called, so the application didn't progress further from connecting the broker.

I noticed that you tested this driver with MQTT, have you tried the `mqtt_publisher`?